espressif/esp-idf #3111: Sending BLE notifications as fast as possible causes random stack assertion errors

jeremyherbert, 2019-03-01:
## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  ESP32-WROVER-B
- IDF version: v3.3-beta1-328-gabea9e4c0
- Build System: CMake
- Compiler version: crosstool-ng-1.22.0-80-g6c4433a
- Operating System: Linux
- Power Supply: USB

## Problem Description

I have a simple BLE GATT server which is sending a notification with 506 bytes as fast as possible in a while (1) loop and an MTU of 509 (set by the client). Of course, this will cause congestion events, and the sending halts while the congestion is occurring (as per `ESP_GATTS_CONGEST_EVT`). I am seeing random BT stack crashes as below. Sometimes during a congestion event and sometimes not.

It's a bit tricky for me to extract some sample code for this, but I will try to do it when I get time soon.

## Debug Logs

```
assertion "length > 0" failed: file "/home/jeremy/esp/esp-idf/components/bt/bluedroid/hci/hci_hal_h4.c", line 140, function: transmit_data
abort() was called at PC 0x400d7583 on core 0
0x400d7583: __assert_func at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

Backtrace: 0x4008ecc8:0x3ffd5410 0x4008eef9:0x3d7583:0x3ffd5450 0x40121923:0x3ffd5480 0x40114a61:0x3ffd54b0 0x401158f6:0x3ffd54d0 0x40114d37:0x3ffd5500 0x40115e5b:0x3ffd5520 0x40114bc3:0x3ffd5540 0x4008f131:0x3ffd5570
0x4008ecc8: invoke_abort at /home/jeremy/esp/esp-idf/components/esp32/panic.c:707

0x4008eef9: abort at /home/jeremy/esp/esp-idf/components/esp32/panic.c:707

0x40121923: transmit_data at /home/jeremy/esp/esp-idf/components/bt/bluedroid/hci/hci_hal_h4.c:140 (discriminator 1)

0x40114a61: transmit_fragment at /home/jeremy/esp/esp-idf/components/bt/bluedroid/hci/hci_layer.c:352

0x401158f6: fragment_and_dispatch at /home/jeremy/esp/esp-idf/components/bt/bluedroid/hci/packet_fragmenter.c:125

0x40114d37: event_packet_ready at /home/jeremy/esp/esp-idf/components/bt/bluedroid/hci/hci_layer.c:343

0x40115e5b: fixed_queue_process at /home/jeremy/esp/esp-idf/components/bt/bluedroid/osi/fixed_queue.c:289

0x40114bc3: hci_host_thread_handler at /home/jeremy/esp/esp-idf/components/bt/bluedroid/hci/hci_layer.c:238

0x4008f131: vPortTaskWrapper at /home/jeremy/esp/esp-idf/components/freertos/port.c:435
```

Weijian-Espressif, 2019-03-06:
@jeremyherbert, 
please test our ble_throughput code.

jeremyherbert, 2019-03-10:
Ok, I will try to give this a go soon

jeremyherbert, 2019-03-29:
I haven't been able to reproduce this since, so I am going to close this. Will reopen if I find an issue.

closed 2019-03-29 by jeremyherbert