espressif/esp-idf #3112: HTTP OTA problem

Andy001847, 2019-03-01:
My project base on BLE and WiFi coexistence, the ble is scanning device when ota process, also, the project used mqtt.

_Originally posted by @Andy001847 in https://github.com/espressif/esp-idf/issues/3107#issuecomment-468568946_

TianHao-Espressif, 2019-03-01:
@Andy001847 , there's 2 temporary solution that:
1. stop ble scanning while wifi is in OTA.
2.  Run wifi task and other High task different from bluedroid.

This problem is due to bluedroid(bluetooth host) have little chance to run while device is in OTA, especially WiFi task is run the same CPU core as bluedroid. Then you know ble scanning have no flow control mechanism, so it will receive packets from the air continuously, but the host cannot handle it, then the memory pool of bluetooth controller is not enough.

We will do something to protect it to avoid assert.

homeit-hq, 2019-03-01:
@TianHao-Espressif can you clarify option 2? Are you asking to run the wifi on CPU0 and bluedroid on CPU1?

TianHao-Espressif, 2019-03-04:
Hi, @homeit-hq , exactly, you can make bluedroid and bluetooth controller task run on CPU1, while wifi on CPU0.

jimparis, 2019-03-06:
Hi @TianHao-Espressif,
I also see this problem while Bluetooth scanning is enabled and I'm writing to ESP flash, presumably because the CPU is blocked during flash writes.

jimparis, 2019-03-15:
I think this may be fixed by 75287eb56989d1e3ec6bd3ea7cfc8fd32bb4df61

jack0c, 2022-01-05:
Also version after v3.3 have better coexistence performance.

closed 2022-01-05 by jack0c