espressif/esp-idf #3113: NVS Partition Generator Utility: ESP_ERR_NVS_NOT_FOUND

asweeney99, 2019-03-01:
## Environment
- Development Kit: ESP32-Wrover-Kit
- Module or chip used:  ESP32-WROVER
- IDF version: master: v3.3-beta1-482-gf746b0d76
- Build System: Make
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     Linux

## Problem Description
There seems to be a key length restriction when calling nvs_get_blob().

This line throws ESP_ERR_NVS_NOT_FOUND:
nvs_get_blob(my_handle, "certificate_pem_crt", NULL, &required_size);

But if I change the name of the key to "cert_pem_crt" and recreate the bin file with the shorter key name then this works:
nvs_get_blob(my_handle, "cert_pem_crt", NULL, &required_size)

### Expected Behavior
Test Program should not print "Nothing saved yet!"

### Actual Behavior
Program throws ESP_ERR_NVS_NOT_FOUND and aborts.

### Steps to repropduce
1. Create file certs.csv with NVS partition keys
```
key,type,encoding,value
certs_namespace,namespace,,
certificate_pem_crt,file,binary,/home/andrew/esp/esp-idf/examples/protocols/aws_iot/certs/certificate_pem_crt
private_pem_key,file,binary,/home/andrew/esp/esp-idf/examples/protocols/aws_iot/certs/private_pem_key
```
2. Generate bin file with nvs_partition_gen.py
3. Flash bin file 
4. Run app with make flash

### Code to reproduce this issue
```
#define STORAGE_NAMESPACE "certs_namespace"
#define PARTITION_LABEL "certs"

void app_main()
{
	printf("Hello this is a test program!\n");

	// Initialize NVS.
	esp_err_t err = nvs_flash_init_partition(PARTITION_LABEL);
	if (err == ESP_ERR_NVS_NO_FREE_PAGES) {
		printf("Error: ESP_ERR_NVS_NO_FREE_PAGES\n");
		ESP_ERROR_CHECK(nvs_flash_erase());
		err = nvs_flash_init_partition(PARTITION_LABEL);
	}
	ESP_ERROR_CHECK( err );

	//Get a handle to the NVS partition
	nvs_handle my_handle;

	// Open
	err = nvs_open_from_partition(PARTITION_LABEL, STORAGE_NAMESPACE, NVS_READONLY, &my_handle);
	if (err != ESP_OK) return err;

	// Read the certificate
	size_t required_size = 0;  // value will default to 0, if not set yet in NVS
    	// obtain required memory space to store certificate being read from NVS
	err = nvs_get_blob(my_handle, "certificate_pem_crt", NULL, &required_size); //error here
	
	ESP_ERROR_CHECK(err);
	
	//Print the certificate
    	printf("AWS cetificate.pem.crt:\n");
    	if (required_size == 0) {
        	printf("Nothing saved yet!\n");
    	}
        ...
}
```

## Debug Logs
```
Hello this is a test program!
ESP_ERROR_CHECK failed: esp_err_t 0x1102 (ESP_ERR_NVS_NOT_FOUND) at 0x40084e70
0x40084e70: _esp_error_check_failed at /home/andrew/esp/esp-idf/components/esp32/panic.c:720

file: "/home/andrew/esp/esp-idf/examples/get-started/test/main/test.c" line 40
func: app_main
expression: err

ELF file SHA256: 6fde17d233eed0b01dffeb8429d93680adc7d2c9ad746e6b840435d965cd6f7e
Backtrace: 0x40084a0c:0x3ffb4c70 0x40084e73:0x3ffb4c90 0x400d2347:0x3ffb4cb0 0x400d0bd6:0x3ffb4ce0 0x40086919:0x3ffb4d00
0x40084a0c: invoke_abort at /home/andrew/esp/esp-idf/components/esp32/panic.c:715
0x40084e73: _esp_error_check_failed at /home/andrew/esp/esp-idf/components/esp32/panic.c:721
0x400d2347: app_main at /home/andrew/esp/esp-idf/examples/get-started/test/main/test.c:40 (discriminator 1)
0x400d0bd6: main_task at /home/andrew/esp/esp-idf/components/esp32/cpu_start.c:526
0x40086919: vPortTaskWrapper at /home/andrew/esp/esp-idf/components/freertos/port.c:403
```

igrr, 2019-03-01:
The limit is mentioned here: https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/storage/nvs_flash.html#keys-and-values

Partition generator should check the limit, though.

asweeney99, 2019-03-01:
Good to know, thank you. I was following the specification set out in the NVS Partition Generator Documentation.
[image]

closed 2019-03-01 by asweeney99

william-ferguson-au, 2019-06-04:
Thanks @igrr  I had a similar issue, but it was because the key was longer than 15 chars.
The partition generator doesn't complain and nvs_get_u8() just returns ESP_ERR_NVS_NOT_FOUND.

It would be be much better if both of those the an error that indicated the key was too long.