espressif/esp-idf #3114: The git tag v3.1.3 is not an annotated tag

chrta, 2019-03-01:
In order to make `git describe` work correctly, it would me good if all your tags would be annotated tags (created with `git tag -a`).

Most of your tags are annotated tags, but not all (tag is an annotated tag, commit is not):
```
$ git for-each-ref refs/tags
...
4d83cae037051ecff050cd87161643310fac51c6 tag	refs/tags/v3.1.1
4eb20f4d9955a311a8f7ae66b0c77dfa188c53d4 commit	refs/tags/v3.1.1-rc2
7b58d83835187af26aef070c29a4e1767f454f67 tag	refs/tags/v3.1.2
cf5dbadf4f25b395887238a7d4d8251c279afa8c commit	refs/tags/v3.1.3
...
```

If I now check out v3.1.3 i get this:
```
$ git describe
v3.1.2-99-gcf5dbad
```

For the correctly annotated tag v3.1.2 I get:
```
$ git describe 
v3.1.2
```

Please create all your tags ass annotated tags.

projectgus, 2019-03-05:
Hi @chrta ,

Sorry, this one is on me. I had aimed to make all final release tags as annotated and prereleases as un-annotated. I can't remember why I thought not annotating prerelease tags was a good idea. But my bash history shows I goofed when tagging v3.1.3 and didn't make an annotated tag.

We're going to add a CI check that prevents our internal repo from pushing any un-annotated tags to github.

In the meantime, as you probably know, `git describe --all` or `git describe --tags` will give you the correct version output.

chrta, 2019-03-05:
Thank you, no worries. I just wanted to prevent this happening for future tags.

closed 2020-10-14 by espressif-bot