espressif/esp-idf #3115: PROJECT_VER variable is not defined as Build documentation states

nameofuser1, 2019-03-01:
Build system documentation says that:

- If PROJECT_VER variable set in project Makefile file, its value will be used.
- Else, if the $PROJECT_PATH/version.txt exists, its contents will be used as PROJECT_VER.
- Else, if the project is located inside a Git repository, the output of git describe will be used.
- Otherwise, PROJECT_VER will be "1".

While in project makefile it is set to empty string if is not defined before:
https://github.com/espressif/esp-idf/blob/ebdcbe8c60a3b5a228d9e80335d869cc01c527ab/make/project.mk#L306-L309

igrr, 2019-03-01:
Sorry, the documentation has not been updated to match the final implementation (with regards to setting CPPFLAGS). The logic related to PROJECT_VER is here: https://github.com/espressif/esp-idf/blob/58b53942661bd40567c988eaafe86ceb519bde04/components/app_update/component.mk

The flag is only passed to one source file. However you can retrieve the version from esp_app_desc_t structure here: https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/ota.html#_CPPv227esp_ota_get_app_descriptionv

nameofuser1, 2019-03-01:
Oh, I see it. Thanks!

krzychb, 2019-03-02:
@nameofuser1 I will keep it open to update documentation and then close

closed 2019-03-08 by igrr

leoribg, 2023-09-14:
OMG, this is still happening on ESP-IDF V5.1 :(