espressif/esp-idf #3116: building esp-idf with ccache on windows

tbitowt, 2019-03-01:
Hello,

I would like to use ccache with esp-idf on my Windows machine, but seems that it's not being used automatically.

I'm using idf.py as a build trigger, and seems that build is not triggered with ccache. (I have checked with --verbose, and files are compiled without ccache). Ccache exists in my PATH of course.

This problem happens on Windows 10.

renzbagaporo, 2019-05-06:
Thanks for reporting this issue @tbitowt! It seems that the issue encompasses not just Windows, as the build system code previously handling allowing use of ccache when present has been deleted by a past commit. Will fix ASAP.

closed 2019-05-10 by igrr

saloid, 2020-05-01:
Is it works now? I've added ccache to path but it still not used. Need to something else?

igrr, 2020-05-01:
It does, you can pass a flag to idf.py to make it use ccache, or set IDF_CCACHE_ENABLE=1 environment variable.

(When installing IDF command prompt on Windows, IDF_CCACHE_ENABLE=1 gets set automatically for recent IDF versions)