espressif/esp-idf #3117: xRingbufferGetCurFreeSize is not correct

nopnop2002, 2019-03-01:
/* The example of ESP-IDF
	 *
	 * This sample code is in the public domain.
	 */
	#include <string.h>
	#include "freertos/FreeRTOS.h"
	#include "freertos/ringbuf.h"
	#include "freertos/task.h"
	#include "freertos/semphr.h"
	#include "esp_log.h"

	#define No_Split
	//#define Allow_Split

	TaskHandle_t xTask1 = NULL, xTask2 = NULL;
	SemaphoreHandle_t xSemaphore;
	RingbufHandle_t xRingbuffer;

	void task1(void *pvParameter)
	{
	    TickType_t nowTick;
	    UBaseType_t res;
	    char tx_item[64];
	    size_t freeSize;
	    UBaseType_t prio = uxTaskPriorityGet(NULL);
	    nowTick = xTaskGetTickCount();
	    printf("[%s:%d] Start Priority=%d\n",pcTaskGetTaskName(NULL),nowTick,prio);

	    //Send an item
	    xSemaphoreTake(xSemaphore, portMAX_DELAY);
	    xRingbufferPrintInfo(xRingbuffer);
	    memset(tx_item,'A',sizeof(tx_item));
	    res =  xRingbufferSend(xRingbuffer, tx_item, 32, pdMS_TO_TICKS(1000));
	    if (res != pdTRUE) {
	        printf("Failed to send item\n");
	    }
	    nowTick = xTaskGetTickCount();
	    freeSize = xRingbufferGetCurFreeSize(xRingbuffer);
	    printf("[%s:%d] freeSize=%d\n",pcTaskGetTaskName(NULL),nowTick,freeSize);
	    xSemaphoreGive(xSemaphore);
	    vTaskDelay(1);

	    xSemaphoreTake(xSemaphore, portMAX_DELAY);
	    xRingbufferPrintInfo(xRingbuffer);
	    memset(tx_item,'B',sizeof(tx_item));
	    res =  xRingbufferSend(xRingbuffer, tx_item, 16, pdMS_TO_TICKS(1000));
	    if (res != pdTRUE) {
	        printf("Failed to send item\n");
	    }
	    nowTick = xTaskGetTickCount();
	    freeSize = xRingbufferGetCurFreeSize(xRingbuffer);
	    printf("[%s:%d] freeSize=%d\n",pcTaskGetTaskName(NULL),nowTick,freeSize);
	    xSemaphoreGive(xSemaphore);
	    vTaskDelay(1);

	    xSemaphoreTake(xSemaphore, portMAX_DELAY);
	    xRingbufferPrintInfo(xRingbuffer);
	    memset(tx_item,'C',sizeof(tx_item));
	    res =  xRingbufferSend(xRingbuffer, tx_item, 40, pdMS_TO_TICKS(1000));
	    if (res != pdTRUE) {
	        printf("Failed to send item\n");
	    }
	    nowTick = xTaskGetTickCount();
	    freeSize = xRingbufferGetCurFreeSize(xRingbuffer);
	    printf("[%s:%d] freeSize=%d\n",pcTaskGetTaskName(NULL),nowTick,freeSize);
	    xSemaphoreGive(xSemaphore);
	    vTaskDelay(1);

	    xSemaphoreTake(xSemaphore, portMAX_DELAY);
	    xRingbufferPrintInfo(xRingbuffer);
	    memset(tx_item,'D',sizeof(tx_item));
	    res =  xRingbufferSend(xRingbuffer, tx_item, 28, pdMS_TO_TICKS(1000));
	    if (res != pdTRUE) {
	        printf("Failed to send item\n");
	    }
	    nowTick = xTaskGetTickCount();
	    freeSize = xRingbufferGetCurFreeSize(xRingbuffer);
	    printf("[%s:%d] freeSize=%d\n",pcTaskGetTaskName(NULL),nowTick,freeSize);
	    xSemaphoreGive(xSemaphore);
	    vTaskDelay(1);

	    while(1) {
	      vTaskDelay(100);
	    }
	}

	void task2(void *pvParameter)
	{
	    TickType_t nowTick;
	    UBaseType_t prio = uxTaskPriorityGet(NULL);
	    nowTick = xTaskGetTickCount();
	    printf("[%s:%d] Start Priority=%d\n",pcTaskGetTaskName(NULL),nowTick,prio);

	#ifdef No_Split
	    size_t item_size;
	    while (1) {
	      xSemaphoreTake(xSemaphore, portMAX_DELAY);
	      //Receive an item from no-split ring buffer
	      char *item = (char *)xRingbufferReceive(xRingbuffer, &item_size, pdMS_TO_TICKS(1000));

	      //Check received item
	      if (item != NULL) {
	        //Print item
	        nowTick = xTaskGetTickCount();
	        printf("[%s:%d] item_size=%d:",pcTaskGetTaskName(0),nowTick,item_size);
	        for (int i = 0; i < item_size; i++) {
	            printf("%c", item[i]);
	        }
	        printf("\n");
	        //Return Item
	        vRingbufferReturnItem(xRingbuffer, (void *)item);
	#if 0
	      } else {
	        //Failed to receive item
	        printf("Failed to receive item\n");
	#endif
	      }
	      xSemaphoreGive(xSemaphore);
	//      nowTick = xTaskGetTickCount();
	//      printf("[%s:%d] Give\n",pcTaskGetTaskName(NULL),nowTick);
	    }
	#endif

	#ifdef Allow_Split
	    //Receive an item from allow-split ring buffer
	    size_t item_size1, item_size2;
	    char *item1, *item2;
	    while (1) {
	      xSemaphoreTake(xSemaphore, portMAX_DELAY);
	      //Receive an item from allow-split ring buffer
	      BaseType_t ret = xRingbufferReceiveSplit(xRingbuffer, (void **)&item1, (void **)&item2, &item_size1, &item_size2, pdMS_TO_TICKS(1000));

	    //Check received item
	      if (ret == pdTRUE && item1 != NULL) {
	        nowTick = xTaskGetTickCount();
	        printf("[%s:%d] item_size1=%d:",pcTaskGetTaskName(0),nowTick,item_size1);
	        for (int i = 0; i < item_size1; i++) {
	            printf("%c", item1[i]);
	        }
	        printf("\n");
	        vRingbufferReturnItem(xRingbuffer, (void *)item1);
	        //Check if item was split
	        if (item2 != NULL) {
	            printf("[%s:%d] item_size2=%d:",pcTaskGetTaskName(0),nowTick,item_size2);
	            for (int i = 0; i < item_size2; i++) {
	                printf("%c", item2[i]);
	            }
	            printf("\n");
	            vRingbufferReturnItem(xRingbuffer, (void *)item2);
	        }
	#if 0
	      } else {
	        //Failed to receive item
	        printf("Failed to receive item\n");
	#endif
	      }
	      xSemaphoreGive(xSemaphore);
	    }
	#endif

	}

	void app_main()
	{
	    TickType_t nowTick;
	    vTaskDelay(2000/portTICK_PERIOD_MS);
	    UBaseType_t prio = uxTaskPriorityGet(NULL);
	    nowTick = xTaskGetTickCount();
	    printf("[%s:%d] Start Priority=%d\n",pcTaskGetTaskName(NULL),nowTick,prio);
	    printf("[%s:%d] portTICK_PERIOD_MS=%d\n",pcTaskGetTaskName(NULL),nowTick,portTICK_PERIOD_MS);

	    /* Create Mutex */
	    xSemaphore = xSemaphoreCreateMutex();

	    /* Create Ring Buffer */
	#ifdef No_Split
	    xRingbuffer = xRingbufferCreate(128, RINGBUF_TYPE_NOSPLIT);
	#endif
	#ifdef Allow_Split
	    xRingbuffer = xRingbufferCreate(128, RINGBUF_TYPE_ALLOWSPLIT);
	#endif

	    /* Check everything was created. */
	    configASSERT( xSemaphore );
	    configASSERT( xRingbuffer );

	    /* Create task */
	    xTaskCreatePinnedToCore(task1, "Task1", 4096, NULL, 1, &xTask1, tskNO_AFFINITY);
	    xTaskCreatePinnedToCore(task2, "Task2", 4096, NULL, 1, &xTask2, tskNO_AFFINITY);
	}

Result

	[main:200] Start Priority=1
	[main:200] portTICK_PERIOD_MS=10
	[Task2:200] Start Priority=1
	[Task1:200] Start Priority=1
	Rb size:128     free: 128       rptr: 0 freeptr: 0      wptr: 0
	[Task1:200] freeSize=56 -----> ??
	[Task2:200] item_size=32:AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
	Rb size:128     free: 128       rptr: 40        freeptr: 40     wptr: 40
	[Task1:301] freeSize=56 -----> ??
	[Task2:301] item_size=16:BBBBBBBBBBBBBBBB
	Rb size:128     free: 128       rptr: 64        freeptr: 64     wptr: 64
	[Task1:401] freeSize=56 -----> ??
	[Task2:401] item_size=40:CCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCC
	Rb size:128     free: 128       rptr: 112       freeptr: 112    wptr: 112
	[Task1:501] freeSize=56 -----> ??
	[Task2:501] item_size=28:DDDDDDDDDDDDDDDDDDDDDDDDDDDD

Dazza0, 2019-03-01:
`xRingbufferGetCurFreeSize()` returns the maximum size an item/data can have if it was currently sent to the ring buffer. The maximum size of item in No-Split buffers are limited to `(pxRingbuffer->xSize / 2) - 8`. Therefore, in the log you provided above, an empty No-Split buffer of 128 bytes will have a max current item size of `(128/2)-8 = 56`.

nopnop2002, 2019-03-02:
OK
Thank you.
Please write the contents above on the API reference.

Please close this issue.

nopnop2002, 2019-03-02:
I have one more question about ring buffer.
Do i need Exclusive access control using Mutex ??

Dazza0, 2019-03-05:
Nope, the function `xRingbufferGetCurFreeSize()` itself is thread safe. The note about thread safety in the API reference is about having multiple tasks sending and receiving to the same ring buffer simultaneously. 

Take the case of two tasks A and B simultaneously sending to the same ring buffer as an example. Task A calls `xRingbufferGetCurFreeSize()` to get an `item_size`, then attempts to call `xRingbufferSend(item_size)`. In between the calls to `xRingbufferGetCurFreeSize()` and  `xRingbufferSend()` by Task A, Task B may have also sent something to the ring buffer making the `item_size` used by Task A no longer valid. 

Therefore, in order to ensure that `item_size` is valid, it is the applications responsibility to ensure that in between the call of `xRingbufferGetCurFreeSize()` to `xRingbufferSend()` by task A, no other task or ISR sends to the ring buffer.

nopnop2002, 2019-03-11:
Thank you for answer.
Plase close this issues.

commit 8a597b86 (2019-06-30): esp_ringbuf: Add static allocation, refactor, and update API reference

This commit updates does the following:
    - xRingbufferCreateStatic() added to allow ringbuffers via to be statically
      allocated. Docs and unit tests update accordingly. Closes #3064

    - Fix doc typos, closes #3248. Updated API reference regarding to mention
      item size limits of no-split buffers when calling
      xRingbufferGetCurFreeSize() or xRingbufferGetMaxItemSize(), closes #3117.

    - Remove the following deprecated types/functions for v4.0
        - xRingbufferIsNextItemWrapped()
        - xRingbuf [... cut]
closed 2019-03-15 by projectgus