espressif/esp-idf #3120: mqtt event disconnected

AkhileshThorat, 2019-03-02:
## Environment

- Development Kit:   ESP32-DevKitC
- Kit version : DevKitC: v2
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it): 
    3.3-dev-301-gadd7c49a2
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // -ng-1.22.0-80-g6c4433a5)
- Operating System:     [Windows]
- Power Supply:         USB

## Problem Description

Used openssl utility to generate client.key and client.crt for the example code.
On opening client.crt it shows following message"the integrity of the certificate cannot be guaranteed.The certificate maybe corrupted or altered"
Is there any additional step i need to do to install certificate or something is missing?
How do i validate or digitally sign the certificate?

I get following logs on monitor

## Debug Logs

I (4724) MQTTS_EXAMPLE: [APP] Free memory: 235796 bytes
I (4724) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
E (5934) esp-tls: mbedtls_ssl_handshake returned -0x4080
E (5934) esp-tls: Failed to open new connection
E (5934) TRANS_SSL: Failed to open a new connection
E (5934) MQTT_CLIENT: Error transport connect
I (5944) MQTT_CLIENT: Reconnect after 10000 ms
I (5944) MQTTS_EXAMPLE: MQTT_EVENT_DISCONNECTED

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

david-cermak, 2019-03-04:
Hi @AkhileshThorat 

I assume you are talking about `ssl_mutual_auth` example where not only client verifies the server, but also server has to verify the client (and therefore you need client certificate)?
You have to generate client key and client *CSR* (not crt) with openssl, then send your csr to CA, which would generate a certificate based on your request.

The example code uses `test.mosquitto.org` when you can get the certificate generated online. Please follow the steps described in https://github.com/espressif/esp-idf/blob/master/examples/protocols/mqtt/ssl_mutual_auth/README.md

AkhileshThorat, 2019-03-05:
Hi @david-cermak 

I performed the following activities after reading the instructions given in Readme file
1=Generated a client key and CSR (not crt) using open ssl utility
2=copied the content of .csr file and pasted it to certificate generating utility at test.mosquitto.org
3=saved the .crt file and pasted it to main directory along with client key
4=I'm still doubtful about the whole process since I'm stuck 

OUTPUT:MQTT event disconnected

logs: E (343992) esp-tls: mbedtls_ssl_handshake returned -0x4080(I tried searching the error code for it's explanation,but found nothing)
E (273892) esp-tls: Failed to open new connection
E (273892) TRANS_SSL: Failed to open a new connection
E (273892) MQTT_CLIENT: Error transport connect
I (273902) MQTTS_EXAMPLE: MQTT_EVENT_DISCONNECTED

chegewara, 2019-03-06:
@AkhileshThorat 
mbedTLS -0x4080
https://tls.mbed.org/api/rsa_8h.html#abf2997e01281b898fd4594af0b7bfbb4

AkhileshThorat, 2019-03-07:
@chegewara 
I managed to find following function corresponding to error code 0x4080 at rsa.h file located inside mbedtls 

int mbedtls_rsa_import_raw( mbedtls_rsa_context *ctx,
                            unsigned char const *N, size_t N_len,
                            unsigned char const *P, size_t P_len,
                            unsigned char const *Q, size_t Q_len,
                            unsigned char const *D, size_t D_len,
                            unsigned char const *E, size_t E_len );

/**
 * \brief          This function completes an RSA context from
 *                 a set of imported core parameters.
 *
 *                 To setup an RSA public key, precisely \p N and \p E
 *                 must have been imported.
 *
 *                 To setup an RSA private key, sufficient information must
 *                 be present for the other parameters to be derivable.
 *
 *                 The default implementation supports the following:
 *                 <ul><li>Derive \p P, \p Q from \p N, \p D, \p E.</li>
 *                 <li>Derive \p N, \p D from \p P, \p Q, \p E.</li></ul>
 *                 Alternative implementations need not support these.
 *
 *                 If this function runs successfully, it guarantees that
 *                 the RSA context can be used for RSA operations without
 *                 the risk of failure or crash.
 *
 * \warning        This function need not perform consistency checks
 *                 for the imported parameters. In particular, parameters that
 *                 are not needed by the implementation might be silently
 *                 discarded and left unchecked. To check the consistency
 *                 of the key material, see mbedtls_rsa_check_privkey().
 *
 * \param ctx      The initialized RSA context holding imported parameters.
 *
 * \return         \c 0 on success.
 * \return         #MBEDTLS_ERR_RSA_BAD_INPUT_DATA if the attempted derivations
 *                 failed.

How do I proceed further with this information?

david-cermak, 2019-03-12:
Hi @AkhileshThorat ,

I've just checked the test.mosquitto.org client cert generator and it looks like it returns a zero file. Does your client.crt look like a certificate?
```
-----BEGIN CERTIFICATE-----
...base64 data...
-----END CERTIFICATE-----
```
My guess is that it does not and that would also explain the above error of `MBEDTLS_ERR_RSA_BAD_INPUT_DATA`.

We have already received some other similar issues related to the test.mosquitto.org broker, which runs experimental come and might not be always good as a stable example for using esp-mqtt client.

Have just checked with local mosquitto and mutual authentication works as expected on my side. I think it would be good to update the README with some config hint of a local server, so that anyone can configure a stable broker to test the client with.

I would recommend to setup a local mosquitto with the following configuration:
```
port 8883
# "openssl rehash <path to capath>" each time you add/remove a certificate.
cafile /home/david/certs/ca.crt

# Path to the PEM encoded server certificate.
certfile /home/david/ssl/certs/server.crt

# Path to the PEM encoded keyfile.
keyfile /home/david/certs/server.key

require_certificate true
use_identity_as_username true
```
- You can follow this very good guide to generate server keys and self signed certificate: http://www.steves-internet-guide.com/mosquitto-tls/
- You would need a client key and certificate request -> just follow the guide in README (or use those already generated)
- Then sign this with the same CA as server keys (same command as in step 5 in Steve's guide)
- And paste client crt and key to the text files -> you should now connect without any issues

david-cermak, 2019-05-06:
Hi @AkhileshThorat 

Any update? Were you able to resolve the issue by connecting to a different public broker or a local one?

Alvin1Zhang, 2019-05-20:
@AkhileshThorat Thanks for reporting the issue. Would you help share if any updates for the issue? Thanks.

AkhileshThorat, 2019-05-20:
Hi @david-cermak  and @Alvin1Zhang 
Lately I'm running into basic issues like not able to generate a .CSR file. I'm looking into this matter and will update you soon