espressif/esp-idf #3121: mDNS responding to requests unreliably

swanav, 2019-03-04:
## Environment

- Development Kit:      [NodeMCU | none]
- Kit version (for WroverKit/PicoKit/DevKitC): []
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): v3.3-beta1-506-gebdcbe8c6
- Build System:         [Make]
- Compiler version: 5.2.0
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description
**mDNS responding to requests unreliably** 

When working with the mDNS library, I experienced that the mDNS server will respond intermittently to the requests from devices on the network. It will almost always respond near controller boot-up, but will stop responding soon. 

### Expected Behavior
mDNS responds immediately to requests on the network.

### Actual Behavior
mDNS response intermittent.

### Steps to repropduce

Experienced in the example skech at [mDNS](https://github.com/espressif/esp-idf/tree/ebdcbe8c60a3b5a228d9e80335d869cc01c527ab/examples/protocols/mdns)

negativekelvin, 2019-03-04:
Did you try `esp_wifi_set_ps(WIFI_PS_NONE)`

swanav, 2019-03-05:
The `esp_wifi_set_ps(WIFI_PS_NONE)` does indeed fix the issue. But, it was my understanding that the WIFI_PS_NONE is the default behaviour of the chipset. Has something changed?

negativekelvin, 2019-03-05:
https://github.com/espressif/esp-idf/blob/6c49f19247cc95d364b54718e4c29b7d345267af/components/esp32/include/esp_wifi.h#L458

swanav, 2019-03-05:
This clears everything. Thanks @negativekelvin !

closed 2019-03-05 by swanav