espressif/esp-idf #3123: Question: Reduce TX-power for BT-Classic

i-franzki, 2019-03-04:
## Environment

- Development Kit:       ESP32-PICO-Kit
- Module or chip used:  ESP32-PICO-D4
- IDF version: latest
- Build System:         Make
- Compiler version: 1.22.0-80-g6c4433a5
- Operating System:     Windows
- Power Supply:         USB or external 5V

## Problem Description
This is more a question than a problem:

For Wifi there exists a way to reduce the TX-Power using function `esp_err_t esp_wifi_set_max_tx_power(int8_t power)` see [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/network/esp_wifi.html#_CPPv225esp_wifi_set_max_tx_power6int8_t). With this function I can reduce the Wifi TX power to for example -14dBm. 

Is there a similar function for BT-Classic ? I can't use BLE because I need to use A2DP, which is only available with BT-Classic.

chegewara, 2019-03-04:
https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/bluetooth/controller_vhci.html#_CPPv222esp_bredr_tx_power_set17esp_power_level_t17esp_power_level_t

Alvin1Zhang, 2019-03-04:
@i-franzki Hi, thanks for raising the issue, and also advise you could move to our forum https://esp32.com and raise topics there, thanks.

closed 2019-03-04 by Alvin1Zhang