espressif/esp-idf #3124: CAN driver missing from PlatformIO

giordano-cm, 2019-03-04:
It looks like the CAN driver files are missing from the esp-idf v3.1.3. Hence, i cannot compile any program in PlatformIO when utilizing the CAN driver. Why is that? Why those files are missing?

https://community.platformio.org/t/having-trouble-using-esp-idf-can-driver-in-platformio/6840

giordano-cm, 2019-03-04:
Has something to do with this? -> https://github.com/espressif/esp-idf/issues/2485

negativekelvin, 2019-03-05:
Can driver is a recent feature.  New features are not usually backported to previous releases.  When platform updates their idf base it will be available.

giordano-cm, 2019-03-05:
When is it going to happen?

negativekelvin, 2019-03-05:
You have to ask platformio but probably when 3.3 stable is released.  Please close this as it is outside the scope of esp-idf.

giordano-cm, 2019-03-05:
Actually, i have been asking for PlatformIO and they said i should ask ESP-IDF why this issue was happening. See it here --> https://community.platformio.org/t/having-trouble-using-esp-idf-can-driver-in-platformio/6840/6?u=giordano-cm

maxgerhardt, 2019-03-05:
This is all correct. PlatformIO clones the latest release tags (currently 3.1.3). As soon as ESP-IDF releases the next version with CAN support, PlatformIO will pull it and all is good.

giordano-cm, 2019-03-05:
Surely. Thanks in advance!

Is it possible to keep me (us) warned about this CAN driver release? Maybe, through this issue..

negativekelvin, 2019-03-05:
Just click on watch repository and select releases only

Dazza0, 2019-03-05:
@giordano-cm FYI it's not too difficult for you branch off release 3.1.3 and add the CAN driver related changes yourself. You'll need to add the following changes the following:

- Add `can.c` to the [components/driver directory](https://github.com/espressif/esp-idf/tree/v3.1.3/components/driver)
- Add `can.h` to the [components/driver/include/driver directory](https://github.com/espressif/esp-idf/tree/v3.1.3/components/driver/include/driver)
- Add `can_struct.h` to the [components/soc/esp32/include/soc directory](https://github.com/espressif/esp-idf/tree/v3.1.3/components/soc/esp32/include/soc)
- Add `#define DR_REG_CAN_BASE                         0x3ff6B000` to [`soc.h`](https://github.com/espressif/esp-idf/blob/cf5dbadf4f25b395887238a7d4d8251c279afa8c/components/soc/esp32/include/soc/soc.h#L122)
- Add `PROVIDE ( CAN = 0x3ff6B000 );` to [`esp32.peripherals.ld`](https://github.com/espressif/esp-idf/blob/cf5dbadf4f25b395887238a7d4d8251c279afa8c/components/esp32/ld/esp32.peripherals.ld#L29)

giordano-cm, 2019-03-05:
I'll try out this and ASAP i leave a response here ;)

Thanks in advance!

giordano-cm, 2019-03-05:
@Dazza0. The problem with PlatformIO is the one we have talked here [](https://community.platformio.org/t/having-trouble-using-esp-idf-can-driver-in-platformio/6840). When i open up the IDE, PlatformIO overwrites all the files and erase the "can.c", "can.h" and "can_struct.h", because they don't appear in esp-idf-v3.1.3, the stable version that PlatformIO copies from Github.

projectgus, 2019-03-06:
FWIW you won't have to wait for v3.3 for this, v3.2 has the CAN driver (maybe there is some additions in v3.3, @Dazza0 can confirm).

v3.2 is currently in beta (v3.2-beta3), final release should be in the next couple of weeks.

To make a "frankenCAN" version, you could probably copy all the files @Dazza0 mentioned into a totally new component and add this to your project - just need to get the paths correct for all the headers. I don't know how to do this in platform.io, sorry.

giordano-cm, 2019-03-06:
Thanks, @projectgus. I'll try doing this along the day and i'll keep you guys in touch.

Thanks for now.

giordano-cm, 2019-03-06:
@projectgus. It Works with v3.2, and probably Works with the master branch release. The problem is that the v3.1.3 doesn't support CAN driver, and this is the latest stable version from ESP-IDF repositories, the one that PlatformIO supports. I'll have to wait for now until they upload a new release with CAN driver included.

projectgus, 2019-05-15:
v3.2 final release with CAN driver is available.

closed 2019-05-15 by projectgus