espressif/esp-idf #3125: HID : Gatt server table auto connection

Marc-Aurele, 2019-03-05:
Hello,

In the same way as the HID project example i would like to be able to auto connect the device to the phone after a reset. With the HID project example, it works as a charm but as soon as i remove the HID service and i keep only the battery service, the phone no longer auto connects to the device. Is it possible to auto connect a phone to a previous paired device with a custom services table ?

Regards,

Aurélien

chegewara, 2019-03-05:
Only with custom android app. It is controlled by client app.

Marc-Aurele, 2019-03-05:
Hi,
Thank you very much for the reply. So, for HID, i guess this is because the phone recognizes an HID profile that it autoconnects to it but for any other profiles over BLE, i need a client app ?
Regards,

chegewara, 2019-03-05:
Yes. For custom UUIDs you need to write android client app anyway.

closed 2019-03-05 by Marc-Aurele