espressif/esp-idf #3126: wifi task: abort in ppRxProtoProc()

tsichevski, 2019-03-05:
I've got the abort() in the 'wifi' task periodically. Since the library is closed-sourced, I can't find any fixes or workarounds.
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32]
- IDF version: v3.3-beta1-328-gabea9e4c0
- Build System:         [Make]
- Compiler version: crosstool-NG crosstool-ng-1.22.0-80-g6c4433a
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description
abort() was called from a closed-source IDF code

### Expected Behavior
System libraries must never call abort()

### Actual Behavior

Thread #6 1073513100 (wifi : Running @CPU0) (Suspended : Breakpoint)	
	abort() at panic.c:160 0x4008ce4c	
	ppRxProtoProc() at 0x4013749c	
	ppRxPkt() at 0x40137b88	
	ppTask() at 0x40088121	
	vPortTaskWrapper() at port.c:143 0x4008f3b8	

### Steps to repropduce

1. init wifi
2. spend some time in debugger (stop program execution at breakpoints, step program etc.
3. abort() occurred in the 'wifi' task.

The rest of the message is the log. The last log record seems being related to the problem.
-----------------------------------------------------------------------------------------------------------------------------
I (551) phy: phy_version: 4100, 6fa5e27, Jan 25 2019, 17:02:06, 0, 0
I (551) wifi: mode : sta (d8:a0:1d:62:d1:68)
D (551) event: SYSTEM_EVENT_STA_START
I (1761) wifi: new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (2751) wifi: state: init -> auth (b0)
I (2751) wifi: state: auth -> assoc (0)
I (2761) wifi: state: assoc -> run (10)
I (2781) wifi: connected with monster1, channel 11, bssid = 38:4c:4f:12:28:84
I (2781) wifi: pm start, type: 1

D (2781) event: SYSTEM_EVENT_STA_CONNECTED, ssid:monster1, ssid_len:8, bssid:38:4c:4f:12:28:84, channel:11, authmode:3
D (2791) tcpip_adapter: dhcp client init ip/mask/gw to all-0
D (2791) tcpip_adapter: if0 start ip lost tmr: enter
D (2801) tcpip_adapter: if0 start ip lost tmr: no need start because netif=0x3ffcd0e0 interval=120 ip=0
D (2811) tcpip_adapter: dhcp client start successfully
D (9271) tcpip_adapter: if0 dhcpc cb
D (9271) tcpip_adapter: if0 ip changed=1
D (9271) event: SYSTEM_EVENT_STA_GOT_IP, ip:192.168.0.53, mask:255.255.255.0, gw:192.168.0.1
I (9271) event: sta ip: 192.168.0.53, mask: 255.255.255.0, gw: 192.168.0.1
D (9281) sdspi_host: sdspi_host_init_slot: SPI2 miso=18 mosi=19 sck=21 cs=22 cd=23 wp=-1, dma_ch=2
D (9291) spi: SPI2 use gpio matrix.
D (9291) intr_alloc: Connected src 30 to int 13 (cpu 0)
D (9301) spi_master: eff: 400, limit: 26666k(/2), 0 dummy, -1 delay
D (9301) spi_master: SPI2: New device added to CS0, effective clock: 400kHz
I (9311) gpio: GPIO[22]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (9321) gpio: GPIO[23]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
D (9331) spi_master: SPI device changed from 3 to 0
D (9331) sdspi_host: start_command_default: poll_cmd_response returned 0x107
D (9341) sdspi_host: sdspi_host_start_command: cmd=52 error=0x107
D (9351) sdmmc_cmd: cmd=52, sdmmc_req_run returned 0x107
D (9391) sdmmc_sd: SDHC/SDXC card
D (9391) sdspi_transaction: cmd=5, R1 response: command not supported
D (9391) sdmmc_cmd: cmd=5, sdmmc_req_run returned 0x106
D (9391) sdmmc_io: sdmmc_init_io: io_send_op_cond (1) returned 0x106; not IO card
D (9401) sdspi_host: data CRC set=1
D (9571) sdmmc_common: host_ocr=0x40ff8000 card_ocr=0xc0ff8000
D (9571) sdmmc_common: sdmmc_card_init: host_ocr=40ff8000, card_ocr=c0ff8000
D (9571) sdmmc_init: sdmmc_card_init: card type is SD
D (9581) sdmmc_common: sdmmc_init_host_frequency: using 20000 kHz bus frequency
D (9581) sdspi_host: Setting card clock to 20000 kHz
D (9591) spi_master: eff: 20000, limit: 26666k(/2), 0 dummy, 0 delay
D (9591) spi_master: SPI2: New device added to CS0, effective clock: 20000kHz
D (9601) spi_master: SPI device changed from 3 to 0
I (9611) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (26541) wifi: bcn_timout,ap_probe_send_start
I (45451) wifi: bcn_timout,ap_probe_send_start
I (54231) wifi: bcn_timout,ap_probe_send_start
E (98211) wifi: l:209138774, o:411, n:219374678, now:286578322

igrr, 2019-03-05:
> spend some time in debugger (stop program execution at breakpoints, step program etc.

How long, approximately, do you keep the program halted in the debugger?

The thing is, Wi-Fi hardware can keep working for a short while without interrupts being serviced, but not indefinitely. At some point, because the CPU is halted and can not process interrupts, Wi-Fi hardware will come to a state from which the driver fails to recover, indicating it with an abort() call.

If you are debugging something that is not Wi-Fi related, it might make sense to build the application in a configuration where it will not initialize Wi-Fi, or mock that initialization somehow.

tsichevski, 2019-03-05:
Hi Ivan,

05.03.2019 18:12, Ivan Grokhotkov пишет:

>
>     spend some time in debugger (stop program execution at
>     breakpoints, step program etc.
>
> How long, approximately, do you keep the program halted in the debugger?
It depends. For about a minute, totally.
> The thing is, Wi-Fi hardware can keep working for a short while
> without interrupts being serviced, but not indefinitely. At some
> point, because the CPU is halted and can not process interrupts, [...]
the thing is I develop a Wi-Fi-based application. It would be nice, if I 
could set some timeout value in the library for the period while I'm 
debugging.
Regards,

Vladimir

PS: when do you plans to open-source the library code?

liuzfesp, 2019-03-06:
@tsichevski  we have no plan to fully open-source WiFi library code in short-time.

liuzfesp, 2019-03-06:
Hi @tsichevski, the abort() happens in following scenario:

- The WiFi power save is enabled (it's enabled by default)
- The STA receives a beacon0 in time T0 (T0 here means the local time is T0), the timestamp in beacon0 is T0 (or about T0), the CPU is halted before the power save module processes beacon0.
- The CPU is stopped for about 1 minutes, about 600 beacons are lost (the beacon interval of most of the AP is 100ms).
- The CPU is resume, power save module processes beacon0. It detects that the next TBTT time is T0 + 100*3 = T0 + 300 (let's assume that the DTIM counter is 3),  but the current time is T0 + 600*100 = T0 + 60000ms. Power module thinks it's abnormal and abort the system.

Briefly speaking, the WiFi power save module will abort() the system if the CPU halts for more than about 30000ms. 

How to avoid abort():

- Stop power save by calling esp_wifi_set_ps(WIFI_PS_NONE) or
- Don't halt CPU too long, e.g. just halts CPU for 10 seconds etc.

-

TimXia, 2019-03-06:
@tsichevski, we can remove this abort(). Before that, please do not halt CPU too long when WiFi is working.

tsichevski, 2019-03-06:
Thank you!

> How to avoid abort():
> 
> * Stop power save by calling esp_wifi_set_ps(WIFI_PS_NONE) or
I will try this approach and report the results.

> * Don't halt CPU too long, e.g. just halts CPU for 10 seconds etc.
It is hard to achieve while I am debugging :(

Regards,
Vladimir

tsichevski, 2019-03-06:
> @tsichevski, we can remove this abort().

Please, do. The abort() is for detecting programming errors, and shall never be called in the production environment.

tsichevski, 2019-03-06:
> Stop power save by calling esp_wifi_set_ps(WIFI_PS_NONE)

1. Should I call this before or after the call to esp_wifi_set_config()?
2. Should I call this before or after the call to esp_wifi_start()?
3. Do you have any examples of using any esp_wifi_set_XXX() functions?
4. Can I use the examples/wifi/power_save/main/power_save.c as the esp_wifi_set_ps() usage example?

Regards,
Vladimir

tsichevski, 2019-03-12:
Hi,
> * Stop power save by calling esp_wifi_set_ps(WIFI_PS_NONE) 
Unfortunately, this does not help :( I still got abort() call from time to time.

liuzfesp, 2019-11-26:
Hi @tsichevski this issue has been fixed, please check.

Alvin1Zhang, 2020-03-12:
@tsichevski Thanks for reporting. Feel free to reopen if the issue still happens. Thanks.

closed 2020-03-12 by Alvin1Zhang