espressif/esp-idf #3127: Interrupt wdt timeout during esp_ota_set_boot_partition()

bearsh, 2019-03-05:
## Environment

- Development Kit: none
- Module or chip used: NINA-W1
- IDF version : v3.3-beta2-18-g0f927791b
- Build System: Make
- Compiler version: 1.22.0-80-g6c4433a
- Operating System: Linux
- Power Supply: external 3.3V

## Problem Description

In ``esp_ota_set_boot_partition()`` a interrupt wdt timeout occurs.

### Expected Behavior

esp_ota_set_boot_partition() should return, no wdt timeout should occur.

### Actual Behavior / Debug Logs

```
I (93550) esp_image: segment 0: paddr=0x00100020 vaddr=0x3f400020 size=0x1f38c (127884) map
I (93760) esp_image: segment 1: paddr=0x0011f3b4 vaddr=0x3ffb0000 size=0x00c5c (  3164) 
I (93810) esp_image: segment 2: paddr=0x00120018 vaddr=0x400d0018 size=0x73318 (471832) map
0x400d0018: _stext at ??:?

I (94490) esp_image: segment 3: paddr=0x00193338 vaddr=0x3ffb0c5c size=0x021cc (  8652) 
I (94560) esp_image: segment 4: paddr=0x0019550c vaddr=0x40080000 size=0x00400 (  1024) 
0x40080000: _WindowOverflow4 at /home/bearsh/esp/esp-idf/components/freertos/xtensa_vectors.S:1779

I (94600) esp_image: segment 5: paddr=0x00195914 vaddr=0x40080400 size=0x11510 ( 70928) 
I (94750) esp_image: Verifying image signature...
I (95230) esp_image: segment 0: paddr=0x00100020 vaddr=0x3f400020 size=0x1f38c (127884) map
I (95440) esp_image: segment 1: paddr=0x0011f3b4 vaddr=0x3ffb0000 size=0x00c5c (  3164) 
I (95490) esp_image: segment 2: paddr=0x00120018 vaddr=0x400d0018 size=0x73318 (471832) map
0x400d0018: _stext at ??:?

I (96170) esp_image: segment 3: paddr=0x00193338 vaddr=0x3ffb0c5c size=0x021cc (  8652) 
I (96240) esp_image: segment 4: paddr=0x0019550c vaddr=0x40080000 size=0x00400 (  1024) 
0x40080000: _WindowOverflow4 at /home/bearsh/esp/esp-idf/components/freertos/xtensa_vectors.S:1779

I (96280) esp_image: segment 5: paddr=0x00195914 vaddr=0x40080400 size=0x11510 ( 70928) 
I (96430) esp_image: Verifying image signature...
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x4008f043  PS      : 0x00060534  A0      : 0x80081df1  A1      : 0x3ffd2860  
0x4008f043: vTaskEnterCritical at /home/bearsh/esp/esp-idf/components/freertos/tasks.c:5029

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00060523  A5      : 0x00000020  
A6      : 0xffffffe0  A7      : 0x72c2f62d  A8      : 0x3ffb3abc  A9      : 0x00000001  
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x3ffb1334  A13     : 0x00000001  
A14     : 0x00060523  A15     : 0x00000003  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

ELF file SHA256: 46bcae19475b660d15a7cb9004aff7da7d84bdc298e11da30dac158177385993

Backtrace: 0x4008f043:0x3ffd2860 0x40081dee:0x3ffd2880 0x400f118f:0x3ffd28a0 0x400ebad4:0x3ffd28e0 0x400eb4e1:0x3ffd2930 0x400eb98f:0x3ffd2a60 0x400d6dc7:0x3ffd2a80 0x400d55de:0x3ffd2aa0 0x400d759f:0x3ffd2ac0 0x4008e6a9:0x3ffd2b00
0x4008f043: vTaskEnterCritical at /home/bearsh/esp/esp-idf/components/freertos/tasks.c:5029

0x40081dee: esp_dport_access_stall_other_cpu_start at /home/bearsh/esp/esp-idf/components/esp32/dport_access.c:216

0x400f118f: esp_sha at /home/bearsh/esp/esp-idf/components/esp32/hwcrypto/sha.c:316

0x400ebad4: esp_secure_boot_verify_signature at /home/bearsh/esp/esp-idf/components/bootloader_support/src/secure_boot_signatures.c:61

0x400eb4e1: image_validate at /home/bearsh/esp/esp-idf/components/app_update/esp_ota_ops.c:482

0x400eb98f: esp_ota_set_boot_partition at /home/bearsh/esp/esp-idf/components/app_update/esp_ota_ops.c:482

0x400d6dc7: fwupd_end at /home/bearsh/esp32/cloudbridge/main/fwupd.c:172

0x400d55de: msr_uart_received_cb at /home/bearsh/esp32/cloudbridge/main/cloudconnect.c:1589 (discriminator 9)

0x400d759f: rx_task at /home/bearsh/esp32/cloudbridge/main/msr_uart.c:512

0x4008e6a9: vPortTaskWrapper at /home/bearsh/esp/esp-idf/components/freertos/port.c:403

Core 1 register dump:
PC      : 0x4008f79e  PS      : 0x00060934  A0      : 0x8008f7f3  A1      : 0x3ffcf750  
0x4008f79e: prvGetExpectedIdleTime at /home/bearsh/esp/esp-idf/components/freertos/tasks.c:5029

A2      : 0x3ffb3abc  A3      : 0x00000001  A4      : 0x00000001  A5      : 0x00000001  
A6      : 0x00060021  A7      : 0x00000000  A8      : 0x00000001  A9      : 0x3ffcf730  
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x8008e8b2  A13     : 0x3ffcf660  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x0000001d  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

ELF file SHA256: 46bcae19475b660d15a7cb9004aff7da7d84bdc298e11da30dac158177385993

Backtrace: 0x4008f79e:0x3ffcf750 0x4008f7f0:0x3ffcf770 0x4008e6a9:0x3ffcf790
0x4008f79e: prvGetExpectedIdleTime at /home/bearsh/esp/esp-idf/components/freertos/tasks.c:5029

0x4008f7f0: prvIdleTask at /home/bearsh/esp/esp-idf/components/freertos/tasks.c:5029

0x4008e6a9: vPortTaskWrapper at /home/bearsh/esp/esp-idf/components/freertos/port.c:403
```

## Other items if possible

- [x] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.) 

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/2931658/sdkconfig.txt)

igrr, 2019-03-05:
Thanks for reporting! It seems that the same issue was reported on the forum, along with a possible fix: https://esp32.com/viewtopic.php?f=13&t=9506#p39591.

bearsh, 2019-03-05:
thanks @igrr for the fast response!
I can **confirm** that the mention **fix works**.

Alvin1Zhang, 2019-03-06:
@bearsh Glad to hear that the mention fix works, and feel free to reopen the issue if it still exists. Thanks.

igrr, 2019-03-06:
Reopened as this hasn't been fixed in IDF yet.

closed 2019-03-15 by igrr