espressif/esp-idf #3128: Header file "gatt_api.h" not found during compilation

giordano-cm, 2019-03-05:
## Environment

- Development Kit: **ESP32-DevKitC**
- Kit version (DevKitC): **v4**
- Module or chip used:  **ESP32-WROOM-32**
- IDF version: **v3.1.3**
- Build System: **Make**
- Compiler version: **xtensa-esp32-elf-gcc.exe (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0**
- Operating System: **Windows10**

## Problem Description

I was running the `make` commands in the PlatformIO IDE (with its own configurations) and jumped into the ESP-IDF, using the mingw32 compiler and the shell prompt. In the firts one, everything ok. But in the second one (mingw32), for some reason, when i do `make -j4 flash` the following lines appear to me:

`C:/msys32/home/MANP_Manfroi/esp/projects/PROJECT_NAME/main/main.c:48:28: fatal error: stack/gatt_api.h: No such file or directory
compilation terminated.
make[1]: *** [/home/MANP_Manfroi/esp/esp-idf/make/component_wrapper.mk:286: main.o] Error 1
CC build/libsodium/libsodium/src/libsodium/crypto_box/curve25519xsalsa20poly1305/box_curve25519xsalsa20poly1305.o
make: *** [C:/msys32/home/MANP_Manfroi/esp/esp-idf/make/project.mk:468: component-main-build] Error 2`.

### Expected Behavior

It should find and compile the "gatt_api.h" header, once it exists at components/bt/bluedroid/stack/include/stack directory.

### Actual Behavior

It doesn't compile. This happens:

`C:/msys32/home/MANP_Manfroi/esp/projects/PROJECT_NAME/main/main.c:48:28: fatal error: stack/gatt_api.h: No such file or directory`

igrr, 2019-03-06:
Have you enable Bluetooth in menuconfig, under component config?

giordano-cm, 2019-03-06:
Yes, i did. Otherwise, as long as i know, the compiler would warned something about it, during compilation, wouldn't it?

giordano-cm, 2019-03-06:
It looks like this guy here (https://github.com/espressif/arduino-esp32/issues/1434) had the same problem i did. He was using the Arduino core, but he couldn't compile without taking out the

giordano-cm, 2019-03-06:
It looks like this guy here (https://github.com/espressif/arduino-esp32/issues/1434) had the same problem i did. He was using the Arduino core, but he couldn't compile without taking out the `#include "stack/gatt_api.h"`. That is, at least, very akward.

igrr, 2019-03-14:
It seems that bluedroid/stack/include directory is only added to COMPONENT_PRIV_INCLUDEDIRS, which means it is only accessible to the bt component (https://github.com/espressif/esp-idf/blob/12b2268ee56dac79988b75dbb49a453629ed8022/components/bt/component.mk).
Can you explain why you need to include stack/gatt_api.h rather than esp_gatt_api.h?

giordano-cm, 2019-03-14:
I didn't solve this problem, but i made a way to turn it around.

The only thing i was using from stack/gatt_api.h was the GATT_AUTH_REQ_NONE define, that doens't exist in esp_gatt_api.h. I made a define of it in my code, deleted the include header and it worked. But still, i couldn't compile using the gatt_api.h header.

So, is there no way to access the gatt_api.h? What does the COMPONENT_PRIV_INCLUDEDIRS do in component.mk?

dhrishi, 2019-03-15:
@giordano-cm You can use ESP_GATT_AUTH_REQ_NONE which is mapped to BTA_GATT_AUTH_REQ_NONE (see here: https://github.com/espressif/esp-idf/blob/ce6b73ca5f52ceb49fc4b7a76d6b0e343418caac/components/bt/bluedroid/api/include/api/esp_gatt_defs.h#L256).
BTA_GATT_AUTH_REQ_NONE is set to GATT_AUTH_REQ_NONE (see here: https://github.com/espressif/esp-idf/blob/ce6b73ca5f52ceb49fc4b7a76d6b0e343418caac/components/bt/bluedroid/bta/include/bta/bta_gatt_api.h#L260)

giordano-cm, 2019-03-15:
Thank you, @dhrishi. It worked. I still don't know why i can't include gatt_api.h in my code, but that's for another issue. For now, it is solved.

closed 2019-03-15 by giordano-cm

sagarvegg, 2019-06-15:
@giordano-cm you can change [idf_path]\components\..\component.mk file content COMPONENT_PRIV_INCLUDEDIRS to COMPONENT_ADD_INCLUDEDIRS than you can able to compile your code..