espressif/esp-idf #3131: error: static assertion failed: "IDF_VER is longer than idf_ver field in structure"

jimparis, 2019-03-06:
IDF_VER may be too long and should be truncated, similar to https://github.com/espressif/esp-idf/issues/2826

Since an abbreviated commit hash is included in the version string, this can show up if the user has changed their "core.abbrev" git config option, for example:

    git config --global core.abbrev 20

will break the build.

lsgunth, 2019-03-07:
I also see this with newer versions of git which automatically choose a longer abbreviation for git describe. I specifically set core.abbrev to 7 to work around this. It might be a good idea to increase the size for IDF_VER or use a specific "--abbrev=X"  for the git describe on IDF_VER.

KonstantinKondrashov, 2019-03-13:
@jimparis and @lsgunth Thanks for your report, it will be fixed. Have created MR into our internal repo.

closed 2019-03-14 by igrr