espressif/esp-idf #3132: GPIO Interrupt doesn't occur during duration test

hvishu, 2019-03-06:
Hi,
We are using the latest ESP IDF code with our application.  Below is print message to confirm the same
**I (30) boot: ESP-IDF v3.3-dev-264-g8a69ffc36-dirty 2nd stage bootloade**

We are using two GPIO as interrupts IO33 and IO21.  What we have observed that, the ISR runs fine for more than a day or so, but fail to occur randomly.  Way to recover is reset the board.  This problem is intermittent.
I have checked the https://github.com/espressif/esp-idf/issues/1234, though it is in similar line, but is not exactly the same.  We have some wrapper on top ESP IDF code to configure the interrupt.  I have pasted the relevant code below.  In the problem situation "g_totalTurbineIsrCount" variable stops increment.  We have provided the same interrupt source to RL78 platform and it works fine.  

Please let me know if there is any wrong usage of the IDF api's or give some pointers as to why this could be happening.  Please be noted that , this is a intermittent problem, sometimes takes couple of day's to occur.

typedef struct
{
    gpio_num_t pin;
    void *funPtr;
    gpio_int_type_t intr_type;
}esphal_intrConfig_t;

uint32_t g_totalTurbineIsrCount=0;

static void IRAM_ATTR TurbineIsr(void* arg)
{
    g_totalTurbineIsrCount++;
}

#define BOARD_TURBINE_SENSOR_SIGNAL_IO21_PIN                    GPIO_NUM_21
#define BOARD_POWER_DETECT_IO33_PIN                             GPIO_NUM_33

bool esphal_configGPIOAsInterrupt(esphal_intrConfig_t intrConfig)
{
    if (interupt_counter >=MAX_GPIO_ISR_SUPPORTED)
    {
        ESP_LOGI(DEBUG_MSGS,"max supported GPIO As interrupt is %d",MAX_GPIO_ISR_SUPPORTED);
        ESP_LOGI(DEBUG_MSGS,"Change the definition MAX_GPIO_ISR_SUPPORTED in esp_hal.c");
        return false;
    }
    intrConfigData[interupt_counter].pin = intrConfig.pin;
    intrConfigData[interupt_counter].funPtr = intrConfig.funPtr;
    intrConfigData[interupt_counter].intr_type = intrConfig.intr_type;
    interupt_counter++;
    return true;
}

bool esphal_finalizeGPIOIInterrupt(void)
{
    gpio_config_t io_conf;
    uint8_t loopCount;
    io_conf.intr_type = GPIO_INTR_DISABLE;
    io_conf.pull_down_en = 0;
    io_conf.pull_up_en = 0;
    io_conf.pin_bit_mask = 0;
    for (loopCount =0;loopCount<interupt_counter;loopCount++)
    {
        io_conf.pin_bit_mask |= (1ULL << intrConfigData[loopCount].pin);
    }
    io_conf.mode = GPIO_MODE_INPUT;
    gpio_config(&io_conf);

    for (loopCount =0;loopCount<interupt_counter;loopCount++)
        gpio_set_intr_type(intrConfigData[loopCount].pin, intrConfigData[loopCount].intr_type);

    //install gpio isr service
    gpio_install_isr_service(0);
    //hook isr handler for specific gpio pin
    for (loopCount =0;loopCount<interupt_counter;loopCount++)
        gpio_isr_handler_add(intrConfigData[loopCount].pin, 
                intrConfigData[loopCount].funPtr, (void*) intrConfigData[loopCount].pin);
    return true;
}

void PowerInit(void)
{
    esphal_intrConfig_t intrConfig;
    intrConfig.pin = BOARD_POWER_DETECT_IO33_PIN;
    intrConfig.funPtr = (void *)PowerIsr;
    intrConfig.intr_type = GPIO_PIN_INTR_NEGEDGE;
    esphal_configGPIOAsInterrupt(intrConfig);
}

void TurbineInit(void)
{
    esphal_gpioConfig_t config;
    esphal_intrConfig_t intrConfig;
    intrConfig.pin = BOARD_TURBINE_SENSOR_SIGNAL_IO21_PIN;
    intrConfig.funPtr = (void *)TurbineIsr;
    intrConfig.intr_type = GPIO_PIN_INTR_NEGEDGE;
    esphal_configGPIOAsInterrupt(intrConfig);
}

//extract from our main application
taskA()
{
	while(1)
	{
		printf("g_totalTurbineIsrCount = %d\n",g_totalTurbineIsrCount);
		vTaskDelay(100);//
	}
}
app_main()
{
	.
	.
	.
	PowerInit();
	TurbineInit();
	esphal_finalizeGPIOIInterrupt();
	.
	taskA,taskB, etc...
	
	.
}

koobest, 2019-04-01:
Hi, @hvishu 
I think your api usage is OK, can you provide more information to help us debug this?

1. Is this the simplest code that can reproduce this issue? I want to know if there is any possibility that other programs are causing this problem.
2. Do you mean gpio_isr will never be triggered and the other parts of your code still works when this issue occurs? 
3. Where is your trigger signal coming from?
4. Will this issue still occur when pull-up or pull-down is enabled? 

thanks !!

hvishu, 2019-04-04:
Using gpio_set_intr_type, instead of gpio_intr_disable and gpio_intr_enable solved the problem.  Thanks.

closed 2019-04-04 by hvishu