espressif/esp-idf #3134: [ble mesh] Provisioning issue after the first time

ferremarco, 2019-03-06:
Hello,

I was able to provisioning my "ble_mesh_client" with nRF Mesh Nordic Android app only the first time, now it's no more possible.
I tryed to flash againg the device but I'm not able to performe the provisioning procedure anymore.

Here the log:

I (30) boot: ESP-IDF v3.2-dev-525-g2f5656356 2nd stage bootloader
I (30) boot: compile time 10:47:07
I (30) boot: Enabling RNG early entropy source...
I (30) boot: SPI Speed      : 40MHz
I (31) boot: SPI Mode       : DIO
I (31) boot: SPI Flash Size : 4MB
I (32) boot: Partition Table:
I (32) boot: ## Label            Usage          Type ST Offset   Length
I (33) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (34) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (35) boot:  2 factory          factory app      00 00 00010000 00100000
I (36) boot: End of partition table
I (37) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x2c2b0 (180912) map
I (101) esp_image: segment 1: paddr=0x0003c2d8 vaddr=0x3ffc0000 size=0x03bbc ( 15292) load
I (107) esp_image: segment 2: paddr=0x0003fe9c vaddr=0x3ffc3bbc size=0x00000 (     0) load
I (108) esp_image: segment 3: paddr=0x0003fea4 vaddr=0x40080000 size=0x0016c (   364) load
0x40080000: _WindowOverflow4 at /home/vimar/esp/esp-idf/components/freertos/xtensa_vectors.S:1685

I (109) esp_image: segment 4: paddr=0x00040018 vaddr=0x400d0018 size=0x8ffc8 (589768) map
0x400d0018: _flash_cache_start at ??:?

I (316) esp_image: segment 5: paddr=0x000cffe8 vaddr=0x4008016c size=0x00294 (   660) load
0x4008016c: _WindowUnderflow12 at ??:?

I (317) esp_image: segment 6: paddr=0x000d0284 vaddr=0x40080400 size=0x122dc ( 74460) load
I (348) esp_image: segment 7: paddr=0x000e2568 vaddr=0x400c0000 size=0x00000 (     0) load
I (348) esp_image: segment 8: paddr=0x000e2570 vaddr=0x50000000 size=0x00000 (     0) load
I (360) boot: Loaded app from partition at offset 0x10000
I (360) boot: Disabling RNG early entropy source...
I (360) cpu_start: Pro cpu up.
I (360) cpu_start: Starting app cpu, entry point is 0x40081090
0x40081090: call_start_cpu1 at /home/vimar/esp/esp-idf/components/esp32/cpu_start.c:225

I (0) cpu_start: App cpu up.
I (362) heap_init: Initializing. RAM available for dynamic allocation:
I (363) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (363) heap_init: At 3FFCEBB0 len 00011450 (69 KiB): DRAM
I (364) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (365) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (366) heap_init: At 400926DC len 0000D924 (54 KiB): IRAM
I (366) cpu_start: Pro cpu start user code
I (41) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (42) ble_mesh_client: Initializing...
I (112) BTDM_INIT: BT controller compile version [a9156c1]

I (112) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (182) phy: phy_version: 3960, 5211945, Jul 18 2018, 10:40:07, 0, 0
I (532) ble_mesh_client: esp_ble_mesh_prov_cb, event = ESP_BLE_MESH_PROV_REGISTER_COMP_EVT
I (532) ble_mesh_client: ESP_BLE_MESH_PROV_REGISTER_COMP_EVT, err_code 0
I (542) ble_mesh_client: esp_ble_mesh_prov_cb, event = ESP_BLE_MESH_NODE_PROV_ENABLE_COMP_EVT
I (542) ble_mesh_client: ESP_BLE_MESH_NODE_PROV_ENABLE_COMP_EVT, err_code 0
I (542) ble_mesh_client: BLE Mesh node initialized

Here the log with the SiliconLabs android app:

I (112) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (182) phy: phy_version: 3960, 5211945, Jul 18 2018, 10:40:07, 0, 0
I (532) ble_mesh_client: esp_ble_mesh_prov_cb, event = ESP_BLE_MESH_PROV_REGISTER_COMP_EVT
I (532) ble_mesh_client: ESP_BLE_MESH_PROV_REGISTER_COMP_EVT, err_code 0
I (542) ble_mesh_client: esp_ble_mesh_prov_cb, event = ESP_BLE_MESH_NODE_PROV_ENABLE_COMP_EVT
I (542) ble_mesh_client: ESP_BLE_MESH_NODE_PROV_ENABLE_COMP_EVT, err_code 0
I (542) ble_mesh_client: BLE Mesh node initialized
I (197972) ble_mesh_client: esp_ble_mesh_prov_cb, event = ESP_BLE_MESH_NODE_PROV_LINK_OPEN_EVT
I (197972) ble_mesh_client: ESP_BLE_MESH_NODE_PROV_LINK_OPEN_EVT, bearer PB-GATT
I (205932) ble_mesh_client: esp_ble_mesh_prov_cb, event = ESP_BLE_MESH_NODE_PROV_INPUT_EVT
I (265992) ble_mesh_client: esp_ble_mesh_prov_cb, event = ESP_BLE_MESH_NODE_PROV_LINK_CLOSE_EVT
I (265992) ble_mesh_client: ESP_BLE_MESH_NODE_PROV_LINK_CLOSE_EVT, bearer PB-GATT

WCCWCC, 2019-03-07:
Hi,
       I use nRF Mesh Nordic Android app(V1.10) and ble_mesh_client_model,When I enter the command to restart the device.
      use command: make monitor
           [esp32-ble-mesh/examples/bluetooth/ble_mesh/ble_mesh_client_model$ make monitor]
      The log:

I (382) heap_init: Initializing. RAM available for dynamic allocation:
I (382) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (383) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (384) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (385) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (385) heap_init: At 3FFCB5A8 len 00014A58 (82 KiB): DRAM
I (386) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (387) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (388) heap_init: At 40092290 len 0000DD70 (55 KiB): IRAM
I (389) cpu_start: Pro cpu start user code
I (63) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (65) ble_mesh_client: Initializing...
I (85) BTDM_INIT: BT controller compile version [0b60040]
I (85) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (175) phy: phy_version: 4006, e540b8e, Dec 17 2018, 11:53:06, 0, 0
I (645) ble_mesh_client: esp_ble_mesh_prov_cb, event = ESP_BLE_MESH_PROV_REGISTER_COMP_EVT
I (645) ble_mesh_client: ESP_BLE_MESH_PROV_REGISTER_COMP_EVT, err_code 0
I (645) ble_mesh_client: BLE Mesh node initialized
I (655) ble_mesh_client: esp_ble_mesh_prov_cb, event = ESP_BLE_MESH_NODE_PROV_ENABLE_COMP_EVT
I (655) ble_mesh_client: ESP_BLE_MESH_NODE_PROV_ENABLE_COMP_EVT, err_code 0

I was able to provisioning my "ble_mesh_client" with nRF Mesh Nordic Android app many time.you should check the code in function(ble_mesh_init),this api will trigger event.  （ESP_BLE_MESH_NODE_PROV_ENABLE_COMP_EVT）
   esp_ble_mesh_node_prov_enable(ESP_BLE_MESH_PROV_ADV | ESP_BLE_MESH_PROV_GATT);

copercini, 2019-03-13:
I'm having this issue too, this seems an Android bluetooth cache problem, try just disable the phone bluetooth then re-enable it. Here after do it I'm able to provision again

closed 2019-04-11 by ferremarco