espressif/esp-idf #3139: A serious security bug in the ESP NOW protocol

MacWyznawca, 2019-03-07:
When receiving data sent via ESP NOW, there is no information whether it was sent to a specific recipient or via multicast on MAC FFFFFFFFFFFF. There is also no information about whether they were encrypted. This allows multicast commands to be sent by the recipient. Checking the MAC of the sender in the list of known peers is not enough, because you can easily change the MAC address.

Very useful would be information in the receive callback about the fact that the packet came with multicast and additional or was encrypted.

TimXia, 2019-03-18:
@MacWyznawca Yes, I think the multicast information is important. But why do you need to know whether the packet is encrypted or not?

MacWyznawca, 2019-03-18:
Thanks @TimXia!
It was more about interchangeably with the information about the multicast. If the information is not encrypted and the sender is on the peer list, it means that it is a mutlicast message. But better is know multicast recieve it is.

Now I have to set up a special key during pairing, known as the encryption key only to paired devices. Whit this additional key, I can distinguish whether the sender is known or multicast.

Alvin1Zhang, 2019-05-16:
@MacWyznawca Thanks for reporting the issue. Would you please help share if any updates for the issue? Has your issue been fixed? Thanks.

MacWyznawca, 2019-05-16:
@Alvin1Zhang Now any changes in ESP-NOW API. I use my own workaround. I add „magic-secret” key to any paired transmision as addition to standard ESP encryption.

Alvin1Zhang, 2019-12-04:
@MacWyznawca Thanks for reporting, feel free to reopen if you still have the issue. Thanks.

closed 2019-12-04 by Alvin1Zhang