espressif/esp-idf #3140: BT_APPL: bta_dm_pm_sniff BTM_SetPowerMode() returns ERROR status=3

ChiShaoJun, 2019-03-07:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used: ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it): 
   v3.3-beta1-210-gb173cf281
- Build System:Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
   1.22.0-80-g6c4433a
- Operating System:Linux
- Power Supply:USB

## Problem Description
Program function：a2dp+ble+bt
my ipad is connect to esp32.when I frist play voice，the a2dp connection is over.

### Expected Behavior

### Actual Behavior

### Steps to repropduce

1. step1
2. ...

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs
I (54134) SPP_ACCEPTOR_DEMO: authentication success: “microe”的 iPad
I (54134) SPP_ACCEPTOR_DEMO: 8c 7c 92 5b df af 
I (54504) SPP_ACCEPTOR_DEMO: ESP_BT_GAP_CFM_REQ_EVT Please compare the numeric value: 507404
I (55794) SPP_ACCEPTOR_DEMO: authentication success: “microe”的 iPad
I (55794) SPP_ACCEPTOR_DEMO: 8c 7c 92 5b df af 
E (55884) BT_APPL: rcb[0].lidx=3, lcb.conn_msk=x1
I (55884) BT_AV: AVRC conn_state evt: state 1, [8c:7c:92:5b:df:af]
E (55964) BT_APPL: bta_av_rc_create ACP handle exist for shdl:0
I (55964) BT_AV: A2DP connection state: Connecting, [8c:7c:92:5b:df:af]
I (56194) BT_AV: AVRC remote features 0x24b
I (56394) BT_AV: A2DP audio stream configuration, codec type 0
I (56394) I2S: PLL_D2: Req RATE: 44100, real rate: 44642.000, BITS: 16, CLKM: 14, BCK: 8, MCLK: 11289966.924, SCLK: 1428544.000000, diva: 64, divb: 11
I (56404) BT_AV: Configure audio player 0x22-0x15-0x2-0x35
I (56414) BT_AV: Audio player configured, sample rate=44100
W (56554) BT_APPL: new conn_srvc id:19, app_id:0
I (56554) BT_AV: A2DP connection state: Connected, [8c:7c:92:5b:df:af]
I (56554) BT_AV: ESP_A2D_CONNECTION_STATE_CONNECTED
I (56554) SPP_ACCEPTOR_DEMO: required_size:256

I (56604) BT_AV: AVRC remote features 0x24b
I (59944) BT_AV: AVRC remote features 0x24b
I (82124) BT_LOG: bta_av_link_role_ok hndl:x41 role:1 conn_audio:x1 bits:1 features:x824b

W (82124) BT_APPL: new conn_srvc id:19, app_id:1
I (82134) BT_AV: A2DP audio state: Started
I (84634) BT_AV: Audio packet count 100
I (86554) BT_AV: A2DP audio state: Suspended
E (89754) BT_APPL: bta_dm_pm_sniff BTM_SetPowerMode() returns ERROR status=3
E (91604) BT_APPL: bta_dm_pm_sniff BTM_SetPowerMode() returns ERROR status=3
W (91604) BT_AVCT: avct_lcb_last_ccb
W (91604) BT_AVCT: 0: aloc:1, lcb:0x3ffcd0b8/0x3ffcd0b8, ccb:0x3ffcd120/0x3ffcd120
W (91614) BT_AVCT: 1: aloc:0, lcb:0x0/0x3ffcd0b8, ccb:0x3ffcd138/0x3ffcd120
W (91614) BT_AVCT: 2: aloc:0, lcb:0x0/0x3ffcd0b8, ccb:0x3ffcd150/0x3ffcd120
I (91624) BT_AV: A2DP connection state: Disconnected, [8c:7c:92:5b:df:af]
I (101624) BT_AV: AVRC conn_state evt: state 0, [00:00:00:00:00:00]
E (110614) BT_APPL: bta_dm_pm_btm_status hci_status=12

blueMoodBHD, 2019-03-11:
Please run `make menuconfig` -> `Component config` -> `Bluetooth` -> `Bluedroid Enable`, then increase `BT/BLE MAX ACL CONNECTIONS`, Then test again. Thanks.

blueMoodBHD, 2019-03-12:
As I have not reproduced the problem. Could you please help me do some test?
Run `make menuconfig` -> `Component config` -> `Bluetooth` -> `Bluedroid Enable` -> `BT DEBUG LOG LEVEL`, set `BTM layler` to `DEBUG`. Then reproduce the problem and send the logs to me. 
Thanks.

ChiShaoJun, 2019-03-12:
Toolchain path: /home/kyle/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
Python requirements from /home/kyle/esp/esp-idf/requirements.txt are satisfied.
MONITOR
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6236
load:0x40078000,len:11152
load:0x40080400,len:6632
entry 0x40080764
I (29) boot: ESP-IDF v3.3-beta1-210-gb173cf281-dirty 2nd stage bootloader
I (29) boot: compile time 17:38:16
I (30) boot: Enabling RNG early entropy source...
I (35) boot: SPI Speed      : 40MHz
I (40) boot: SPI Mode       : DIO
I (44) boot: SPI Flash Size : 4MB
I (48) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (74) boot:  2 factory          factory app      00 00 00010000 00100000
I (81) boot: End of partition table
I (85) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x39c64 (236644) map
I (177) esp_image: segment 1: paddr=0x00049c8c vaddr=0x3ff80000 size=0x00000 (     0) load
I (177) esp_image: segment 2: paddr=0x00049c94 vaddr=0x3ff80000 size=0x00000 (     0) load
I (184) esp_image: segment 3: paddr=0x00049c9c vaddr=0x3ffbdb60 size=0x02a50 ( 10832) load
I (197) esp_image: segment 4: paddr=0x0004c6f4 vaddr=0x3ffc05b0 size=0x00000 (     0) load
I (202) esp_image: segment 5: paddr=0x0004c6fc vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/kyle/esp/esp-idf/components/freertos/xtensa_vectors.S:1685

I (211) esp_image: segment 6: paddr=0x0004cb04 vaddr=0x40080400 size=0x0350c ( 13580) load
I (225) esp_image: segment 7: paddr=0x00050018 vaddr=0x400d0018 size=0xaf3c8 (717768) map
0x400d0018: _flash_cache_start at ??:?

I (480) esp_image: segment 8: paddr=0x000ff3e8 vaddr=0x4008390c size=0x0ed48 ( 60744) load
0x4008390c: esp_ptr_internal at /home/kyle/esp/esp-idf/components/spi_flash/flash_mmap.c:339
 (inlined by) spi_flash_mmap_pages at /home/kyle/esp/esp-idf/components/spi_flash/flash_mmap.c:151

I (505) esp_image: segment 9: paddr=0x0010e138 vaddr=0x400c0000 size=0x00000 (     0) load
I (505) esp_image: segment 10: paddr=0x0010e140 vaddr=0x50000000 size=0x00000 (     0) load
I (512) esp_image: segment 11: paddr=0x0010e148 vaddr=0x50000000 size=0x00000 (     0) load
I (532) boot: Loaded app from partition at offset 0x10000
I (532) boot: Disabling RNG early entropy source...
I (533) cpu_start: Pro cpu up.
I (536) cpu_start: Application information:
I (541) cpu_start: Project name:     bt_ble_coex
I (546) cpu_start: App version:      4714e66-dirty
I (552) cpu_start: Compile time:     17:38:19
I (557) cpu_start: Compile date:     Mar 12 2019
I (562) cpu_start: ESP-IDF:          v3.3-beta1-210-gb173cf281-dirty
I (569) cpu_start: Starting app cpu, entry point is 0x40081248
0x40081248: call_start_cpu1 at /home/kyle/esp/esp-idf/components/esp32/cpu_start.c:265

I (0) cpu_start: App cpu up.
I (580) heap_init: Initializing. RAM available for dynamic allocation:
I (587) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (593) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (599) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (605) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (611) heap_init: At 3FFD2698 len 0000D968 (54 KiB): DRAM
I (617) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (623) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
[... 1203 lines cut ...]

D (134445) BT_BTM: btm_acl_report_role_change

D (134445) BT_BTM: btm_acl_update_busy_level

D (134455) BT_BTM: BTM_BLI_ACL_DOWN_EVT

D (134455) BT_BTM: acl hci_handle=129 transport=1 connectable_mode=0x0 link_role=1

D (134465) BT_BTM: before update p_dev_rec->sec_flags=0x40be

D (134475) BT_BTM: Bletooth link down

D (134475) BT_BTM: after update p_dev_rec->sec_flags=0x40b8

D (134485) BT_BTM: btm_acl_resubmit_page

D (134485) BT_BTM: btm_sec_disconnected before update sec_flags=0x40b8

D (134495) BT_BTM: btm_sec_disconnected after update sec_flags=0xb8

D (134495) BT_BTM: btm_get_acl_disc_reason_code

E (134505) BT_BTC: handle_rc_connect Connect failed with error code: 2
E (136505) BT_BTC: btc_initiate_av_open_tmr_hdlr No connected RC peers

blueMoodBHD, 2019-03-12:
@ChiShaoJun， Thanks for you quick reply. Do you have a high-priority task that keeps running, and if so, Please lower it or put it on another core, then test again.

wgcod, 2019-03-13:
I also had the same problem, when sending data with spp after HFP connection, but back to the previous version there was no problem.

blueMoodBHD, 2019-03-13:
@ChiShaoJun @wgcod, Could you please give me a demo to  reproduce this problem? Thanks.

ChiShaoJun, 2019-03-18:
@blueMoodBHD Thank you for your reply.Do you know how to solve this problem or provide a solution direction?

blueMoodBHD, 2019-03-19:
@ChiShaojun I find there is some problem in security simple pairing(SSP) with apple device. Disable SSP in menuconfig will not disconnect. And I'm try to fix this bugs ASAP.

closed 2019-04-04 by igrr

wasmac, 2020-09-14:
Hi, was this issue fixed? I am still getting this error on esp-idf v4.3-dev-771-gc77c4ccf6

expresspotato, 2021-07-28:
Hmm, I'm pretty sure this is related... I get strange errors that the client device doesn't support park or hold when using A2DP and the bluetooth stack never enters an intermittent transmission mode. This prevents light sleep (when using an external 32k crystal)...

For HFP things work as expected, when the call terminates the connection goes intermittent and light sleep starts, only to wake intermittently.