espressif/esp-idf #3141: GPIO 36, 39, 34, 35 are not configuring properly as UART Rx pin

PranavD10, 2019-03-07:
I am using ESP32 development board and want to use the any of the GPIO pins 36, 39, 34, 35 as a UART Rx pin. 
I use uart_select_example.c to check whether they work properly or not.
I found that when I configure pin 36 as Rx the pin 36 and 39 both are susceptible and causes reception.
Same thing happen with 34 and 35.
What could be the reason that the other pin also causes the reception...??

koobest, 2019-03-18:
Hi, @PranavD10
Has this problem been solved? I don't quite understand the issue you described. Can you please tell me how can I reproduce this issue and what is the phenomenon?

thanks!!

koobest, 2019-04-01:
Hi, @PranavD10
Can you please share your progress here :-).
thanks !!

Alvin1Zhang, 2019-05-16:
@PranavD10 Thanks for reporting the issue. Will close the issue for now due to lack of details. Feel free to reopen and please please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md). Thanks.

closed 2019-05-16 by Alvin1Zhang

KonstantinKondrashov, 2019-05-16:
I think this issue the same as [this](https://github.com/espressif/esp-idf/issues/3141). If you consider that it is not the same please give us more information about your issue. Thanks.

Mr-HaleYa, 2020-11-20:
I think this may be why. it looks like they are connected internally 
[image]