espressif/esp-idf #3142: Bootloader LoadProhibited

gregjesl, 2019-03-07:
## Environment
- IDF version: v3.3-beta2-21-gfa34fdddb
- Build System: Make
- Compiler version: 1.22.0-80-g6c4433a5
- Operating System: Windows
- Power Supply: USB

## Problem Description

The following is encountered on boot.  The device is stuck in a boot loop:
```
I (903) boot: Loaded app from partition at offset 0x10000
I (903) boot: Disabling RNG early entropy soR
7m#SI\v(:v!L+}npFatal exception (28): LoadProhibited
epc1=0x40008044, epc2=0x00000000, epc3=0x00000000, excvaddr=0xffffffff, depc=0x00000000
```

gregjesl, 2019-03-07:
I rolled back my IDF version and now I encounter the following error:
```
A fatal error occurred: Invalid segment count 24 (max 16). Usually this indicates a linker script problem.
```

gregjesl, 2019-03-08:
I suspect the error is sourced somewhere in here:
https://github.com/espressif/esp-idf/blob/1aa8e8d38ebe688c391359b5962707c1d135d07e/components/bootloader_support/src/bootloader_utility.c#L533-L555

I was not able to replicate the issue with the "Hello World" example.  I suspect the issue may be related to our large binary size (~1.3 MB).  

When I rolled back to a previous version of the IDF, I tried to recompile and flash without updating the Git submodules.  I received the following error:
```
A fatal error occurred: Invalid segment count 24 (max 16). Usually this indicates a linker script problem.
```
When I re-synced the Git submodules, the error went away but it seems to indicate that the way segments are handled has changed.

projectgus, 2019-03-14:
Hi @gregjesl ,

Are you able to share the bootloader and application .bin and .elf files with us, and also the project sdkconfig and custom partition table file? Can email these privately (angus at espressif dot com) if they're not suitable to post on here.

The bootloader crash suggests it's tried to dereference a pointer to (-1) inside the ROM ets_printf function. Which is a little unusual...

> I suspect the error is sourced somewhere in here:

May I ask what led you to this suspicion? Did you add some debugging output that narrowed down the crash point?

> When I rolled back to a previous version of the IDF, I tried to recompile and flash without updating the Git submodules. I received the following error:

Yes, IDF v3.3 adds some new un-linked sections in the WiFi libraries (for optimizing IRAM size), which are not supported by the older linker script. I don't think this is related to the issue you're seeing (the final .elf file and .bin file have the same segments, provided the correct linker script is used.)

gregjesl, 2019-03-14:
Hello @projectgus 

> Are you able to share the bootloader and application .bin and .elf files with us, and also the project sdkconfig and custom partition table file? 

I will double check with the team but I should be able to do so.  

> May I ask what led you to this suspicion? Did you add some debugging output that narrowed down the crash point?  

1. I commented out https://github.com/espressif/esp-idf/blob/1aa8e8d38ebe688c391359b5962707c1d135d07e/components/bootloader_support/src/bootloader_utility.c#L517 to see if the issue was in `bootloader_random_disable()`.  The issue persisted.  
2. If
https://github.com/espressif/esp-idf/blob/1aa8e8d38ebe688c391359b5962707c1d135d07e/components/bootloader_support/src/bootloader_utility.c#L533-L555
was successfully executed I would have expected to see `Mapping segment {} as {}` per the logging functions.  However, I recognize that the serial output simply may not have sent the log via the serial connection before the error was encountered.  
3. The section I posted has a number of array index references, which could lead to a LoadProhibited error if one of those references was null.

gregjesl, 2019-03-14:
@projectgus I've sent you an email with the requested files.

projectgus, 2019-03-14:
@gregjesl Got them, thanks. We have an engineer looking into this, will update you ASAP.

KonstantinKondrashov, 2019-03-15:
Hi, @gregjesl we have found the root cause and will fix it.

gregjesl, 2019-03-15:
@KonstantinKondrashov That's great to hear; thank you for the quick investigation!

KonstantinKondrashov, 2019-03-15:
Hi, @gregjes. If this issue blocks you then you could use this patch: [4520.zip](https://github.com/espressif/esp-idf/files/2970931/4520.zip).

We will fix it to master ASAP.

Thanks.

gregjesl, 2019-03-15:
@KonstantinKondrashov That fixed the issue.  Thanks you again for your quick response! 

Should I close this issue now or wait for the merge into the master branch?

projectgus, 2019-03-17:
Hi @gregjesl ,

Glad that solved the problem.

> Should I close this issue now or wait for the merge into the master branch?

Please leave it open, we'll close it when we merge the fix.

Angus

gregjesl, 2019-03-20:
@projectgus It looks like the fix was merged in commit https://github.com/espressif/esp-idf/commit/b7b5c3a8c75e8e805214c6ef83e72b1e5e37aa38

projectgus, 2019-03-20:
Hi @gregjesl ,

It was, sorry we neglected to put the right "Closes ...." line in the commit. Thanks for the heads-up.

Angus

closed 2019-03-20 by projectgus