espressif/esp-idf #3144: Build difficulty surrounding recent KonstantinKondrashov commits

cnlohr, 2019-03-07:
I've gone to checking out multiple branches inbetween c05a39de0280f896d7fcd7a8852d7e760e850393 and 509e1264b9895768424d587f493ae59278942674 on feb 27.

Everything after 509e1264b9895768424d587f493ae59278942674 seems to cause the following error.  Adding the CONFIG_EFUSE_MAX_BLK_LEN flag back to my sdkconfig seems not to have an impact.

I'm getting the following errors and not sure how to proceed:

```
CC build/bootloader/efuse/esp32/esp_efuse_table.o
/home/cnlohr/esp/esp-idf/components/efuse/esp32/esp_efuse_table.c:26:21: error: 'CONFIG_EFUSE_MAX_BLK_LEN' undeclared here (not in a function)
 #define MAX_BLK_LEN CONFIG_EFUSE_MAX_BLK_LEN
                     ^
/home/cnlohr/esp/esp-idf/components/efuse/esp32/esp_efuse_table.c:29:28: note: in expansion of macro 'MAX_BLK_LEN'
 #define LAST_FREE_BIT_BLK1 MAX_BLK_LEN
                            ^
/home/cnlohr/esp/esp-idf/components/efuse/esp32/esp_efuse_table.c:33:16: note: in expansion of macro 'LAST_FREE_BIT_BLK1'
 _Static_assert(LAST_FREE_BIT_BLK1 <= MAX_BLK_LEN, "The eFuse table does not match the coding scheme. Edit the table and restart the efus
                ^
/home/cnlohr/esp/esp-idf/components/efuse/esp32/esp_efuse_table.c:26:21: error: expression in static assertion is not an integer
 #define MAX_BLK_LEN CONFIG_EFUSE_MAX_BLK_LEN
                     ^
/home/cnlohr/esp/esp-idf/components/efuse/esp32/esp_efuse_table.c:29:28: note: in expansion of macro 'MAX_BLK_LEN'
 #define LAST_FREE_BIT_BLK1 MAX_BLK_LEN
                            ^
/home/cnlohr/esp/esp-idf/components/efuse/esp32/esp_efuse_table.c:33:16: note: in expansion of macro 'LAST_FREE_BIT_BLK1'
 _Static_assert(LAST_FREE_BIT_BLK1 <= MAX_BLK_LEN, "The eFuse table does not match the coding scheme. Edit the table and restart the efus
                ^
/home/cnlohr/esp/esp-idf/components/efuse/esp32/esp_efuse_table.c:26:21: error: expression in static assertion is not an integer
 #define MAX_BLK_LEN CONFIG_EFUSE_MAX_BLK_LEN
                     ^
/home/cnlohr/esp/esp-idf/components/efuse/esp32/esp_efuse_table.c:30:28: note: in expansion of macro 'MAX_BLK_LEN'
 #define LAST_FREE_BIT_BLK2 MAX_BLK_LEN
                            ^
/home/cnlohr/esp/esp-idf/components/efuse/esp32/esp_efuse_table.c:34:16: note: in expansion of macro 'LAST_FREE_BIT_BLK2'
 _Static_assert(LAST_FREE_BIT_BLK2 <= MAX_BLK_LEN, "The eFuse table does not match the coding scheme. Edit the table and restart the efus
                ^
/home/cnlohr/esp/esp-idf/components/efuse/esp32/esp_efuse_table.c:31:28: error: expression in static assertion is not an integer
 #define LAST_FREE_BIT_BLK3 192
                            ^
/home/cnlohr/esp/esp-idf/components/efuse/esp32/esp_efuse_table.c:35:16: note: in expansion of macro 'LAST_FREE_BIT_BLK3'
 _Static_assert(LAST_FREE_BIT_BLK3 <= MAX_BLK_LEN, "The eFuse table does not match the coding scheme. Edit the table and restart the efus
                ^
/home/cnlohr/esp/esp-idf/components/efuse/esp32/esp_efuse_table.c:63:5: warning: missing initializer for field 'bit_count' of 'esp_efuse_desc_t {aka const struct <anonymous>}' [-Wmissing-field-initializers]
     {EFUSE_BLK2, 0, MAX_BLK_LEN},   // Security boot. Key. (length = "None" - 256. "3/4" - 192. "REPEAT" - 128),
     ^
In file included from /home/cnlohr/esp/esp-idf/components/efuse/esp32/esp_efuse_table.c:16:0:
/home/cnlohr/esp/esp-idf/components/efuse/include/esp_efuse.h:57:25: note: 'bit_count' declared here
     uint16_t            bit_count;      /**< Length of bit field [1..-]*/
                         ^
/home/cnlohr/esp/esp-idf/components/efuse/esp32/esp_efuse_table.c:26:21: error: initializer element is not constant
 #define MAX_BLK_LEN CONFIG_EFUSE_MAX_BLK_LEN
                     ^
/home/cnlohr/esp/esp-idf/components/efuse/esp32/esp_efuse_table.c:71:21: note: in expansion of macro 'MAX_BLK_LEN'
     {EFUSE_BLK1, 0, MAX_BLK_LEN},   // Flash encrypt. Key. (length = "None" - 256. "3/4" - 192. "REPEAT" - 128),
                     ^
/home/cnlohr/esp/esp-idf/components/efuse/esp32/esp_efuse_table.c:26:21: note: (near initialization for 'ENCRYPT_FLASH_KEY[0].bit_count')
 #define MAX_BLK_LEN CONFIG_EFUSE_MAX_BLK_LEN
                     ^
/home/cnlohr/esp/esp-idf/components/efuse/esp32/esp_efuse_table.c:71:21: note: in expansion of macro 'MAX_BLK_LEN'
     {EFUSE_BLK1, 0, MAX_BLK_LEN},   // Flash encrypt. Key. (length = "None" - 256. "3/4" - 192. "REPEAT" - 128),
                     ^
/home/cnlohr/esp/esp-idf/components/efuse/esp32/esp_efuse_table.c:71:5: warning: missing initializer for field 'bit_count' of 'esp_efuse_desc_t {aka const struct <anonymous>}' [-Wmissing-field-initializers]
     {EFUSE_BLK1, 0, MAX_BLK_LEN},   // Flash encrypt. Key. (length = "None" - 256. "3/4" - 192. "REPEAT" - 128),
     ^
In file included from /home/cnlohr/esp/esp-idf/components/efuse/esp32/esp_efuse_table.c:16:0:
/home/cnlohr/esp/esp-idf/components/efuse/include/esp_efuse.h:57:25: note: 'bit_count' declared here
     uint16_t            bit_count;      /**< Length of bit field [1..-]*/
                         ^
/home/cnlohr/esp/esp-idf/make/component_wrapper.mk:289: recipe for target 'esp32/esp_efuse_table.o' failed
make[2]: *** [esp32/esp_efuse_table.o] Error 1

```

igrr, 2019-03-08:
Could you please attach your sdkconfig file?

cnlohr, 2019-03-08:
Gah.  Sorry about that.
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/2944888/sdkconfig.txt)

igrr, 2019-03-11:
Are you doing anything special in your project Makefile, or is this with one of the default example makefiles? When I copy your sdkconfig file into a `hello-world` project and start `make`, i get prompted about a large number of config options which haven't been defined, including some efuse related ones.

If you are removing/overriding some of the IDF components, please check that you still include the newly added `efuse` component.

cnlohr, 2019-03-11:
Ah! Yes. As a matter of fact, I am.  I removed a boatload of components for this current project.  I had not thought about that!  Thanks!  I can't look into it now, but that makes a lot of sense, and I'll give it a shot.

igrr, 2019-06-05:
Closing, assuming that did solve the issue. Please feel free to reopen if not.

closed 2019-06-05 by igrr