espressif/esp-idf #3146: bluetooth header can't be included a code in subdirectory

Nathaniel7687, 2019-03-08:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (DevKitC): [v4]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version (run ``git describe --tags`` to find it): 
    v3.3-beta2-20-gff020c3a1
- Build System:         [Make|CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    1.22.0-80-g6c4433a
- Operating System:     [macOS]
- Power Supply:         [USB]

## Problem Description

I'm trying to write a ble code with subdirectory, for example the source tree is below:
```
Project
├── CMakeLists.txt
├── Makefile
├── README.md
├── main
│   ├── CMakeLists.txt
│   ├── ble
│   │   ├── CMakeLists.txt
│   │   ├── ble.c
│   │   ├── ble.h
│   │   ├── component.mk
│   ├── component.mk
│   └── main.c
└── sdkconfig
```

If I try to build ble.c that include the ble headers (esp_gap_ble_api.h, esp_gatts_ble_api.h, esp_gattc_ble_api, etc.) in subdirectory `ble`, an error occurred.
However, if I write the main.c to include the ble headers or if I move the ble.c and ble.h files to `main` directory, anything is ok.

### Actual Behavior

When I build ble source codes which is  with cmake, an error occurred while processing "`Building C object esp-idf/ble/CMakeFiles/idf_component_ble.dir/ble.c.obj`."

The error message is below:
```{
[1/10] Building C object esp-idf/ble/CMakeFiles/idf_component_ble.dir/ble.c.obj
FAILED: esp-idf/ble/CMakeFiles/idf_component_ble.dir/ble.c.obj 
/Users/nate/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DESP_PLATFORM -DGCC_NOT_5_2_0=0 -DHAVE_CONFIG_H -DIDF_VER=\"v3.3-beta1-524-gff020c3a1\" -I../main/ble -Iconfig -I/Users/nate/esp/esp-idf/components/esp32/include -I/Users/nate/esp/esp-idf/components/driver/include -I/Users/nate/esp/esp-idf/components/esp_ringbuf/include -I/Users/nate/esp/esp-idf/components/tcpip_adapter/include -I/Users/nate/esp/esp-idf/components/lwip/include/apps -I/Users/nate/esp/esp-idf/components/lwip/lwip/src/include -I/Users/nate/esp/esp-idf/components/lwip/port/esp32/include -I/Users/nate/esp/esp-idf/components/lwip/port/esp32/include/arch -I/Users/nate/esp/esp-idf/components/lwip/include_compat -I/Users/nate/esp/esp-idf/components/vfs/include -I/Users/nate/esp/esp-idf/components/esp_event/include -I/Users/nate/esp/esp-idf/components/log/include -I/Users/nate/esp/esp-idf/components/efuse/include -I/Users/nate/esp/esp-idf/components/efuse/esp32/include -I/Users/nate/esp/esp-idf/components/newlib/platform_include -I/Users/nate/esp/esp-idf/components/newlib/include -I/Users/nate/esp/esp-idf/components/freertos/include -I/Users/nate/esp/esp-idf/components/app_trace/include -I/Users/nate/esp/esp-idf/components/heap/include -I/Users/nate/esp/esp-idf/components/soc/esp32/include -I/Users/nate/esp/esp-idf/components/soc/include -mlongcalls   -mlongcalls -Og -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -std=gnu99 -Wno-old-style-declaration -MD -MT esp-idf/ble/CMakeFiles/idf_component_ble.dir/ble.c.obj -MF esp-idf/ble/CMakeFiles/idf_component_ble.dir/ble.c.obj.d -o esp-idf/ble/CMakeFiles/idf_component_ble.dir/ble.c.obj   -c ../main/ble/ble.c
In file included from ../main/ble/ble.c:8:0:
../main/ble/ble.h:4:29: fatal error: esp_gap_ble_api.h: No such file or directory
compilation terminated.
[3/10] Performing build step for 'bootloader'
ninja: no work to do.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

Of course, I double checked that the bluetooth component is enabled by `idf.py menuconfig`.

### Code to reproduce this issue
CMakeLists.txt (in `main` directory)
```cpp
set(COMPONENT_SRCS "main.c")
set(COMPONENT_ADD_INCLUDEDIRS ".")

register_component()
```

main.c
```cpp
#include "ble/ble.h"

void app_main()
[... 699 lines cut ...]
CONFIG_SPIFFS_PAGE_SIZE=256
CONFIG_SPIFFS_OBJ_NAME_LEN=32
CONFIG_SPIFFS_USE_MAGIC=y
CONFIG_SPIFFS_USE_MAGIC_LENGTH=y
CONFIG_SPIFFS_META_LENGTH=4
CONFIG_SPIFFS_USE_MTIME=y
# CONFIG_SPIFFS_DBG is not set
# CONFIG_SPIFFS_API_DBG is not set
# CONFIG_SPIFFS_GC_DBG is not set
# CONFIG_SPIFFS_CACHE_DBG is not set
# CONFIG_SPIFFS_CHECK_DBG is not set
# CONFIG_SPIFFS_TEST_VISUALISATION is not set
CONFIG_IP_LOST_TIMER_INTERVAL=120
CONFIG_TCPIP_LWIP=y
CONFIG_UNITY_ENABLE_FLOAT=y
CONFIG_UNITY_ENABLE_DOUBLE=y
# CONFIG_UNITY_ENABLE_COLOR is not set
CONFIG_UNITY_ENABLE_IDF_TEST_RUNNER=y
# CONFIG_UNITY_ENABLE_FIXTURE is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
# CONFIG_WL_SECTOR_SIZE_512 is not set
CONFIG_WL_SECTOR_SIZE_4096=y
CONFIG_WL_SECTOR_SIZE=4096
```

igrr, 2019-03-08:
`ble` component should be placed into `components` directory of your project, not into the `main` directory (which is itself a component).

Please check https://github.com/espressif/esp-idf/tree/master/examples/system/console project as an example of adding extra components, and https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system.html#example-project for documentation.

Nathaniel7687, 2019-03-08:
@igrr Thank you for your reply.

I'm studying CMake and already fully read the document. But, I can't apply well and can't find any examples that are working about you mentioned.
Could you provide a simple example for this project?

Nathaniel7687, 2019-03-08:
@igrr Thank you!

closed 2019-03-08 by Nathaniel7687