espressif/esp-idf #3150: Hostname not sent via DHCP request

adydeac, 2019-03-10:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v1
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): arduino-esp32 1.0.2 RC1
- Build System:
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
- Operating System: Linux
- Power Supply: USB

## Problem Description
Hi,
I've noticed that somehow (I guess recently?) the hostname is not sent when making DHCP calls.

### Expected Behavior
It should send the hostname via DHCP request.

### Actual Behavior
It does not send any hostname via DHCP request.

### Steps to reproduce

### Code to reproduce this issue

```cpp
#include <WiFi.h>

void setup() {
  Serial.begin(115200);
  char hname[19]; // 5+12+1 - don't forget the \0
  snprintf(hname, 19, "ESP32-%012llX", ESP.getEfuseMac());  
  Serial.println(hname);
  WiFi.disconnect();
  WiFi.mode(WIFI_STA);
  WiFi.setHostname(hname);
  WiFi.begin("MiKesNET","bf70dd8a6c1c9");
  WiFi.waitForConnectResult();
  Serial.print("IP Address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  // put your main code here, to run repeatedly:

}

## Debug Logs

```
Console output:
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:10312
load:0x40080400,len:6460
entry 0x400806a4
ESP32-E03F14A4AE30
IP Address: 172.16.0.195

This is from the router:
# tcpdump -Anev -s 1024 -i br-lan port 67 or port 68 or port 69
tcpdump: listening on br-lan, link-type EN10MB (Ethernet), capture size 1024 bytes
01:22:43.011475 30:ae:a4:14:3f:e0 > ff:ff:ff:ff:ff:ff, ethertype IPv4 (0x0800), length 350: (tos 0x0, ttl 255, id 0, offset 0, flags [none], proto UDP (17), length 336)
0.0.0.0.68 > 255.255.255.255.67: BOOTP/DHCP, Request from 30:ae:a4:14:3f:e0, length 308, xid 0xaedfb4c4, Flags [none]
Client-Ethernet-Address 30:ae:a4:14:3f:e0
Vendor-rfc1048 Extensions
Magic Cookie 0x63825363
DHCP-Message Option 53, length 1: Request
MSZ Option 57, length 2: 576
Requested-IP Option 50, length 4: 172.16.0.195
Parameter-Request Option 55, length 12:
Subnet-Mask, Default-Gateway, BR, Domain-Name-Server
Domain-Name, Netbios-Name-Server, Netbios-Node, Netbios-Scope
Router-Discovery, Static-Route, Classless-Static-Route, Vendor-Option
E..P.................D.C.<.}............................0...?...........................................................................................................................................................................................................c.Sc5..9..@2.....7......,./.!y+.........................................
01:22:43.014082 e8￼27:6d:29:35 > 30:ae:a4:14:3f:e0, ethertype IPv4 (0x0800), length 342: (tos 0x0, ttl 64, id 7494, offset 0, flags [none], proto UDP (17), length 328)
172.16.0.1.67 > 172.16.0.195.68: BOOTP/DHCP, Reply, length 300, xid 0xaedfb4c4, Flags [none]
Your-IP 172.16.0.195
Server-IP 172.16.0.2
Client-Ethernet-Address 30:ae:a4:14:3f:e0
sname "tftp"
file "pxelinux.0"
Vendor-rfc1048 Extensions
Magic Cookie 0x63825363
DHCP-Message Option 53, length 1: ACK
Server-ID Option 54, length 4: 172.16.0.1
Lease-Time Option 51, length 4: 259200
RN Option 58, length 4: 129600
RB Option 59, length 4: 226800
Subnet-Mask Option 1, length 4: 255.255.255.0
BR Option 28, length 4: 172.16.0.255
Default-Gateway Option 3, length 4: 172.16.0.1
Domain-Name-Server Option 6, length 4: 172.16.0.1
Domain-Name Option 15, length 3: "lan"
E..H.F..@..{.........C.D.4Z*............................0...?...........tftp............................................................pxelinux.0......................................................................................................................c.Sc5..6.....3.....:....@;...u...........................lan....
```

I've opened the bug report initially on espressif/arduino-esp32, but got advised to move it here:
https://github.com/espressif/arduino-esp32/issues/2537

Thanks!
Adrian

espxiehang, 2019-03-14:
hi, Include the hostname field in DHCP Discover and DHCP Request .
The hostname is send when making DHCP calls.
 
Attachment is a screenshot of the capture package.
[image]

adydeac, 2019-03-14:
Hi!

Thank you for your answer. But how can I do that? I suppose this is on the client side? Or?

Thanks!

espxiehang, 2019-03-14:
Do you mean to set the hostname you want?

you try tcpip_adapter_set_hostname_api();
path: esp-idf/components/tcpip_adapter/tcpip_adapter_lwip.c

adydeac, 2019-03-14:
No. How do you include the Hostname field in the DHCP request/discover?

espxiehang, 2019-03-14:
ok,I understand.
The default configuration will have a hostname.
You can try to use the latest master branch to see if the issue still exists.
You can also check if the hostname configuration is in effect based on the following call relationships.
system_event_sta_start_handle_default（）--->  tcpip_adapter_sta_start( )  ---> tcpip_adapter_start( )
---> wlanif_init_sta ( ) ---> wlanif_init ( )

nightvisi0n, 2019-03-28:
I could successfully reproduce that this is a real bug and I created a PR in order to fix it: espressif/esp-lwip#6

@adydeac You can verify if that fixes your problem by erasing the complete flash of the esp32 (which in turn clears any previous dhcp lease, leading to a fresh dhcp lease and an updated hostname). This can be done e.g. with [esptool](https://github.com/espressif/esptool#erase-flash-erase_flash--erase-region) or [platformio](https://docs.platformio.org/en/latest/userguide/cmd_run.html#cmdoption-platformio-run-t) (`pio run -t erase`).

freakyxue, 2020-01-16:
@adydeac  @nightvisi0n   This is the right solution

freakyxue, 2020-02-19:
@adydeac @nightvisi0n  this bug has been fixed in 3.3.
later i will backport to 3.2 .3.1  
thanks

Alvin1Zhang, 2020-05-29:
Thanks for reporting, fix on release/3.2 is available at https://github.com/espressif/esp-idf/commit/db38a4e06826d052858e70b5464d0a07f9d902e5

closed 2020-10-22 by Alvin1Zhang