espressif/esp-idf #3151: ble_mesh_node nothing happens on switch in app

theobouwman, 2019-03-10:
I just flashed the `ble_mesh_node` example on my ESP32 and provisioned in via the Silicon Lab BLE Mesh app to my network and a group.

But when I change the switch in the app, when connected to the network, nothing happens. Even no log messages appear when viewing the monitor.

Campou, 2019-03-11:
The following is the log appeared on the monitor when i use Silicon Lab app to control the ble_mesh_node:
```
I (250676) ble_mesh_node: esp_ble_mesh_prov_cb, event = ESP_BLE_MESH_NODE_PROV_LINK_OPEN_EVT
I (250676) ble_mesh_node: ESP_BLE_MESH_NODE_PROV_LINK_OPEN_EVT, bearer PB-GATT
I (255346) ble_mesh_node: esp_ble_mesh_prov_cb, event = ESP_BLE_MESH_NODE_PROV_LINK_CLOSE_EVT
I (255356) ble_mesh_node: ESP_BLE_MESH_NODE_PROV_LINK_CLOSE_EVT, bearer PB-GATT
I (255356) ble_mesh_node: esp_ble_mesh_prov_cb, event = ESP_BLE_MESH_NODE_PROV_COMPLETE_EVT
I (255356) ble_mesh_node: net_idx: 0x0000, addr: 0x0012
I (273996) ble_mesh_node: gen_onoff_set_handler, addr 0x12 onoff 0x00
I (273996) ble_mesh_node: Publish previous 0x00 current 0x01
I (273996) ble_mesh_node: gen_onoff_set_handler, addr 0x12 onoff 0x01
W (273996) BOARD: led red is already on
I (274046) ble_mesh_node: gen_onoff_set_handler, addr 0x12 onoff 0x01
W (274046) BOARD: led red is already on
I (274126) ble_mesh_node: gen_onoff_set_handler, addr 0x12 onoff 0x01
I (274136) ble_mesh_node: Publish previous 0x01 current 0x00
I (274176) ble_mesh_node: gen_onoff_set_handler, addr 0x12 onoff 0x00
W (274176) BOARD: led red is already off
I (274216) ble_mesh_node: gen_onoff_set_handler, addr 0x12 onoff 0x00
W (274226) BOARD: led red is already off
I (274306) ble_mesh_node: gen_onoff_set_handler, addr 0x12 onoff 0x00
I (274316) ble_mesh_node: Publish previous 0x00 current 0x01
I (274356) ble_mesh_node: gen_onoff_set_handler, addr 0x12 onoff 0x01
W (274356) BOARD: led red is already on
I (274406) ble_mesh_node: gen_onoff_set_handler, addr 0x12 onoff 0x01
W (274406) BOARD: led red is already on
I (274406) ble_mesh_node: gen_onoff_set_handler, addr 0x12 onoff 0x01
I (274406) ble_mesh_node: Publish previous 0x01 current 0x00
I (274446) ble_mesh_node: gen_onoff_set_handler, addr 0x12 onoff 0x00
```
After provisioning is done, please config the node properly.

Alvin1Zhang, 2019-05-23:
@theobouwman Thanks for reporting the issue. Would you help share if any updates for the issue? Thanks.

Alvin1Zhang, 2019-10-11:
@theobouwman Thanks for reporting, will close this ticket due to short of feedback, feel free to reopen if you still have the issue, thanks.

closed 2019-10-11 by Alvin1Zhang