espressif/esp-idf #3153: error to build esp-idf 3.3 with visualgdb

arnold-b, 2019-03-11:
Hi, 
i freshly downloaded with git V3.3 release and try to compile it. looks one file is missing. i m not sure

FAILED: esp-idf/esp32/esp32.common.ld 

used sample v3.3\examples\protocols\http_server\file_serving
windows 10
visual gdb current version
esp32

thank you

------ Erstellen gestartet: Projekt: file_server, Konfiguration: Debug VisualGDB ------
VisualGDB: Run "C:\PROGRA~2\Sysprogs\VISUAL~1/ninja.exe flash" in directory "C:\VisualGdb\fileserver2\fileserver2/VisualGDB/Debug" on local computer
[1/819] Building C object esp-idf/soc/CMakeFiles/idf_component_soc.dir/esp32/gpio_periph.c.obj
[2/819] Building C object esp-idf/soc/CMakeFiles/idf_component_soc.dir/esp32/cpu_util.c.obj
[3/819] Building C object esp-idf/soc/CMakeFiles/idf_component_soc.dir/esp32/rtc_periph.c.obj
[4/819] Building C object esp-idf/soc/CMakeFiles/idf_component_soc.dir/esp32/rtc_init.c.obj
[5/819] Building C object esp-idf/soc/CMakeFiles/idf_component_soc.dir/esp32/rtc_clk_init.c.obj
[6/819] Building C object esp-idf/soc/CMakeFiles/idf_component_soc.dir/esp32/rtc_clk.c.obj
[7/819] Building C object esp-idf/soc/CMakeFiles/idf_component_soc.dir/esp32/sdmmc_periph.c.obj
[8/819] Building C object esp-idf/soc/CMakeFiles/idf_component_soc.dir/esp32/rtc_pm.c.obj
[9/819] Building C object esp-idf/soc/CMakeFiles/idf_component_soc.dir/esp32/rtc_time.c.obj
[10/819] Building C object esp-idf/soc/CMakeFiles/idf_component_soc.dir/esp32/sdio_slave_periph.c.obj
[11/819] Building C object esp-idf/soc/CMakeFiles/idf_component_soc.dir/esp32/spi_periph.c.obj
[12/819] Building C object esp-idf/soc/CMakeFiles/idf_component_soc.dir/esp32/rtc_sleep.c.obj
[13/819] Building C object esp-idf/soc/CMakeFiles/idf_component_soc.dir/esp32/soc_memory_layout.c.obj
[14/819] Building C object esp-idf/soc/CMakeFiles/idf_component_soc.dir/esp32/rtc_wdt.c.obj
[15/819] Building C object esp-idf/soc/CMakeFiles/idf_component_soc.dir/src/memory_layout_utils.c.obj
[16/819] Building C object esp-idf/log/CMakeFiles/idf_component_log.dir/log.c.obj
[17/819] Building C object esp-idf/heap/CMakeFiles/idf_component_heap.dir/heap_caps_init.c.obj
[18/819] Building C object esp-idf/heap/CMakeFiles/idf_component_heap.dir/heap_trace.c.obj
[19/819] Building C object esp-idf/heap/CMakeFiles/idf_component_heap.dir/heap_caps.c.obj
[20/819] Building C object esp-idf/xtensa-debug-module/CMakeFiles/idf_component_xtensa-debug-module.dir/eri.c.obj
[21/819] Linking C static library esp-idf\soc\libsoc.a
[22/819] Building C object esp-idf/heap/CMakeFiles/idf_component_heap.dir/multi_heap.c.obj
[23/819] Building C object esp-idf/xtensa-debug-module/CMakeFiles/idf_component_xtensa-debug-module.dir/trax.c.obj
[24/819] Linking C static library esp-idf\log\liblog.a
[25/819] Building C object esp-idf/app_trace/CMakeFiles/idf_component_app_trace.dir/app_trace.c.obj
[26/819] Linking C static library esp-idf\heap\libheap.a
[27/819] Generating idf_component_log.sections_info
[28/819] Generating idf_component_heap.sections_info
[29/819] Generating idf_component_soc.sections_info
[30/819] Linking C static library esp-idf\xtensa-debug-module\libxtensa-debug-module.a
[31/819] Building C object esp-idf/app_trace/CMakeFiles/idf_component_app_trace.dir/app_trace_util.c.obj
[32/819] Generating idf_component_xtensa-debug-module.sections_info
[33/819] Building C object esp-idf/app_trace/CMakeFiles/idf_component_app_trace.dir/gcov/gcov_rtio.c.obj
[34/819] Building C object esp-idf/app_trace/CMakeFiles/idf_component_app_trace.dir/host_file_io.c.obj
[35/819] Building C object esp-idf/freertos/CMakeFiles/idf_component_freertos.dir/list.c.obj
[36/819] Building C object esp-idf/freertos/CMakeFiles/idf_component_freertos.dir/port.c.obj
[37/819] Building C object esp-idf/freertos/CMakeFiles/idf_component_freertos.dir/FreeRTOS-openocd.c.obj
[38/819] Building C object esp-idf/freertos/CMakeFiles/idf_component_freertos.dir/event_groups.c.obj
[39/819] Building C object esp-idf/freertos/CMakeFiles/idf_component_freertos.dir/croutine.c.obj
[40/819] Linking C static library esp-idf\app_trace\libapp_trace.a
[41/819] Building ASM object esp-idf/freertos/CMakeFiles/idf_component_freertos.dir/portasm.S.obj
[42/819] Generating idf_component_app_trace.sections_info
[43/819] Building C object esp-idf/freertos/CMakeFiles/idf_component_freertos.dir/xtensa_init.c.obj
[44/819] Building ASM object esp-idf/freertos/CMakeFiles/idf_component_freertos.dir/xtensa_context.S.obj
[45/819] Building C object esp-idf/freertos/CMakeFiles/idf_component_freertos.dir/xtensa_intr.c.obj
[46/819] Building C object esp-idf/freertos/CMakeFiles/idf_component_freertos.dir/queue.c.obj
[47/819] Building ASM object esp-idf/freertos/CMakeFiles/idf_component_freertos.dir/xtensa_vectors.S.obj
[48/819] Building ASM object esp-idf/freertos/CMakeFiles/idf_component_freertos.dir/xtensa_vector_defaults.S.obj
[49/819] Building ASM object esp-idf/freertos/CMakeFiles/idf_component_freertos.dir/xtensa_intr_asm.S.obj
[50/819] Building C object esp-idf/freertos/CMakeFiles/idf_component_freertos.dir/timers.c.obj
[51/819] Building C object esp-idf/freertos/CMakeFiles/idf_component_freertos.dir/xtensa_overlay_os_hook.c.obj
[52/819] Building C object esp-idf/newlib/CMakeFiles/idf_component_newlib.dir/random.c.obj
[53/819] Building C object esp-idf/newlib/CMakeFiles/idf_component_newlib.dir/select.c.obj
[54/819] Building C object esp-idf/newlib/CMakeFiles/idf_component_newlib.dir/reent_init.c.obj
[55/819] Building C object esp-idf/newlib/CMakeFiles/idf_component_newlib.dir/pthread.c.obj
[56/819] Building C object esp-idf/freertos/CMakeFiles/idf_component_freertos.dir/tasks.c.obj
[... 834 lines cut ...]
[53/54] Linking C executable bootloader.elf
[54/54] Generating ../bootloader.bin
esptool.py v2.6
[808/819] No install step for 'bootloader'
[809/819] Linking C static library esp-idf\ulp\libulp.a
[810/819] Generating idf_component_ulp.sections_info
[811/819] Completed 'bootloader'
[812/819] Building C object esp-idf/spiffs/CMakeFiles/idf_component_spiffs.dir/spiffs/src/spiffs_nucleus.c.obj
[813/819] Linking C static library esp-idf\spiffs\libspiffs.a
[814/819] Generating idf_component_spiffs.sections_info
[815/819] Generating esp32.common.ld
FAILED: esp-idf/esp32/esp32.common.ld 
cmd.exe /C "cd /D C:\VisualGdb\fileserver2\fileserver2\VisualGDB\Debug\esp-idf\esp32 && C:\SysGCC\esp32\esp-idf\v3.3\tools\ldgen\ldgen.py --config C:/VisualGdb/fileserver2/fileserver2/sdkconfig --fragments C:/SysGCC/esp32/esp-idf/v3.3/components/soc/linker.lf	C:/SysGCC/esp32/esp-idf/v3.3/components/heap/linker.lf	C:/SysGCC/esp32/esp-idf/v3.3/components/xtensa-debug-module/linker.lf	C:/SysGCC/esp32/esp-idf/v3.3/components/app_trace/linker.lf	C:/SysGCC/esp32/esp-idf/v3.3/components/freertos/linker.lf	C:/SysGCC/esp32/esp-idf/v3.3/components/esp_ringbuf/linker.lf	C:/SysGCC/esp32/esp-idf/v3.3/components/spi_flash/linker.lf	C:/SysGCC/esp32/esp-idf/v3.3/components/lwip/linker.lf	C:/SysGCC/esp32/esp-idf/v3.3/components/esp32/linker.lf	C:/SysGCC/esp32/esp-idf/v3.3/components/esp32/ld/esp32_fragments.lf	C:/SysGCC/esp32/esp-idf/v3.3/components/espcoredump/linker.lf --input C:/SysGCC/esp32/esp-idf/v3.3/components/esp32/ld/esp32.common.ld.in --output C:/VisualGdb/fileserver2/fileserver2/VisualGDB/Debug/esp-idf/esp32/esp32.common.ld --sections C:/VisualGdb/fileserver2/fileserver2/VisualGDB/Debug/ldgen.section_infos --kconfig C:/SysGCC/esp32/esp-idf/v3.3/Kconfig --env "COMPONENT_KCONFIGS= C:/SysGCC/esp32/esp-idf/v3.3/components/app_trace/Kconfig C:/SysGCC/esp32/esp-idf/v3.3/components/aws_iot/Kconfig C:/SysGCC/esp32/esp-idf/v3.3/components/bt/Kconfig C:/SysGCC/esp32/esp-idf/v3.3/components/driver/Kconfig C:/SysGCC/esp32/esp-idf/v3.3/components/efuse/Kconfig C:/SysGCC/esp32/esp-idf/v3.3/components/esp32/Kconfig C:/SysGCC/esp32/esp-idf/v3.3/components/esp_adc_cal/Kconfig C:/SysGCC/esp32/esp-idf/v3.3/components/esp_event/Kconfig C:/SysGCC/esp32/esp-idf/v3.3/components/esp_http_client/Kconfig C:/SysGCC/esp32/esp-idf/v3.3/components/esp_http_server/Kconfig C:/SysGCC/esp32/esp-idf/v3.3/components/esp_https_ota/Kconfig C:/SysGCC/esp32/esp-idf/v3.3/components/espcoredump/Kconfig C:/SysGCC/esp32/esp-idf/v3.3/components/ethernet/Kconfig C:/SysGCC/esp32/esp-idf/v3.3/components/fatfs/Kconfig C:/SysGCC/esp32/esp-idf/v3.3/components/freemodbus/Kconfig C:/SysGCC/esp32/esp-idf/v3.3/components/freertos/Kconfig C:/SysGCC/esp32/esp-idf/v3.3/components/heap/Kconfig C:/SysGCC/esp32/esp-idf/v3.3/components/libsodium/Kconfig C:/SysGCC/esp32/esp-idf/v3.3/components/log/Kconfig C:/Sy [... cut ...]
  File "C:\SysGCC\esp32\esp-idf\v3.3\tools\ldgen\ldgen.py", line 22, in <module>
    from fragments import FragmentFileModel
ImportError: No module named fragments
ninja: build stopped: subcommand failed.
-------------------------------------------------------------
Command exited with code 1
Executable: C:\PROGRA~2\Sysprogs\VISUAL~1/ninja.exe
Arguments: flash
Directory: C:\VisualGdb\fileserver2\fileserver2/VisualGDB/Debug
VisualGDB: Error: Command-line action failed
========== Erstellen: 0 erfolgreich, 1 fehlerhaft, 0 aktuell, 0 übersprungen ==========

igrr, 2019-03-11:
Are you able to build this example outside of VisualGDB, from the command line? Since IDF v3.3 has not been released yet, perhaps something is missing on VisualGDB side to support it.

arnold-b, 2019-03-11:
Sorry, 

i forgot. just now i tried make flash and it works.

closed 2019-03-11 by arnold-b