espressif/esp-idf #3154: 'mbedtls/include' not found

arnold-b, 2019-03-11:
Hello,

if i try to build esp-idf v3.3-def i get this error:
'mbedtls/include' not found

Windows 10
visulagdb
v3.3-dev

------ Erstellen gestartet: Projekt: file_server, Konfiguration: Debug VisualGDB ------
VisualGDB: Run "C:\Users\braun\AppData\Local\VisualGDB\CMake\bin\cmake.exe ../.. -G "Ninja" -DCMAKE_BUILD_TYPE=DEBUG -DCMAKE_COLOR_MAKEFILE=OFF -DCMAKE_MAKE_PROGRAM="C:/Program Files (x86)/Sysprogs/VisualGDB/ninja.exe"" in directory "C:\VisualGdb\fileserver2\fileserver2/VisualGDB/Debug" on local computer
-- Checking Python dependencies...
Python requirements from C:/SysGCC/esp32/esp-idf/master\requirements.txt are satisfied.
-- Building for target esp32
-- Building empty aws_iot component due to configuration
-- Component names: soc log heap xtensa-debug-module app_trace freertos vfs newlib esp_ringbuf driver esp_event ethernet mbedtls micro-ecc efuse bootloader_support partition_table app_update spi_flash nvs_flash lwip tcpip_adapter pthread smartconfig_ack wpa_supplicant esp32 cxx asio jsmn aws_iot bootloader bt coap console nghttp esp-tls esp_adc_cal tcp_transport esp_http_client esp_http_server esp_https_ota openssl esp_https_server espcoredump esptool_py expat wear_levelling sdmmc fatfs freemodbus idf_test json libsodium mdns mqtt protobuf-c protocomm spiffs ulp unity wifi_provisioning main
-- Component paths: C:/SysGCC/esp32/esp-idf/master/components/soc;C:/SysGCC/esp32/esp-idf/master/components/log;C:/SysGCC/esp32/esp-idf/master/components/heap;C:/SysGCC/esp32/esp-idf/master/components/xtensa-debug-module;C:/SysGCC/esp32/esp-idf/master/components/app_trace;C:/SysGCC/esp32/esp-idf/master/components/freertos;C:/SysGCC/esp32/esp-idf/master/components/vfs;C:/SysGCC/esp32/esp-idf/master/components/newlib;C:/SysGCC/esp32/esp-idf/master/components/esp_ringbuf;C:/SysGCC/esp32/esp-idf/master/components/driver;C:/SysGCC/esp32/esp-idf/master/components/esp_event;C:/SysGCC/esp32/esp-idf/master/components/ethernet;C:/SysGCC/esp32/esp-idf/master/components/mbedtls;C:/SysGCC/esp32/esp-idf/master/components/micro-ecc;C:/SysGCC/esp32/esp-idf/master/components/efuse;C:/SysGCC/esp32/esp-idf/master/components/bootloader_support;C:/SysGCC/esp32/esp-idf/master/components/partition_table;C:/SysGCC/esp32/esp-idf/master/components/app_update;C:/SysGCC/esp32/esp-idf/master/components/spi_flash;C:/SysGCC/esp32/esp-idf/master/components/nvs_flash;C:/SysGCC/esp32/esp-idf/master/components/lwip;C:/SysGCC/esp32/esp-idf/master/components/tcpip_adapter;C:/SysGCC/esp32/esp-idf/master/components/pthread;C:/SysGCC/esp32/esp-idf/master/components/smartconfig_ack;C:/SysGCC/esp32/esp-idf/master/components/wpa_supplicant;C:/SysGCC/esp32/esp-idf/master/components/esp32;C:/SysGCC/esp32/esp-idf/master/components/cxx;C:/SysGCC/esp32/esp-idf/master/components/asio;C:/SysGCC/esp32/esp-idf/master/components/jsmn;C:/SysGCC/esp32/esp-idf/master/components/aws_iot;C:/SysGCC/esp32/esp-idf/master/components/bootloader;C:/SysGCC/esp32/esp-idf/master/components/bt;C:/SysGCC/esp32/esp-idf/master/components/coap;C:/SysGCC/esp32/esp-idf/master/components/console;C:/SysGCC/esp32/esp-idf/master/components/nghttp;C:/SysGCC/esp32/esp-idf/master/components/esp-tls;C:/SysGCC/esp32/esp-idf/master/components/esp_adc_cal;C:/SysGCC/esp32/esp-idf/master/components/tcp_transport;C:/SysGCC/esp32/esp-idf/master/components/esp_http_client;C:/SysGCC/esp32/esp-idf/master/components/esp_http_server;C:/SysGCC/esp32/esp-idf/master/components/esp_https_ota;C:/SysGCC/esp32/esp-idf/master/components/openssl;C:/SysGCC/esp32/esp-idf/master/components/esp_https_server;C:/SysGCC/esp32/esp-idf/master/components/espcoredump;C:/SysGCC/esp32/esp-idf/master/components/esptool_py;C:/SysGCC/esp32/esp-idf/master/components/expat;C:/SysGCC/esp32/esp-idf/master/components/wear_levelling;C:/SysGCC/esp32/esp-idf/master/components/sdmmc;C:/SysGCC/esp32/esp-idf/master/components/fatfs;C:/SysGCC/esp32/esp-idf/master/components/freemodbus;C:/SysGCC/esp32/esp-idf/master/components/idf_test;C:/SysGCC/esp32/esp-idf/master/components/json;C:/SysGCC/esp32/esp-idf/master/components/libsodium;C:/SysGCC/esp32/esp-idf/master/components/mdns;C:/SysGCC/esp32/esp-idf/master/components/mqtt;C:/SysGCC/esp32/esp-idf/master/components/protobuf-c;C:/SysGCC/esp32/esp-idf/master/components/protocomm;C:/SysGCC/esp32/esp-idf/master/components/spiffs;C:/SysGCC/esp32/esp-idf/master/components/ulp;C:/SysGCC/esp32/esp-idf/master/components/unity;C:/SysGCC/esp32/esp-idf/master/components/wifi_provisioning;C:/VisualGdb/fileserver2/fileserver2/main
Loading defaults file C:/VisualGdb/fileserver2/fileserver2/sdkconfig.defaults...
-- IDF_VER: v3.3-beta1-512-g89ae5908d
CMake Warning at C:/SysGCC/esp32/esp-idf/master/tools/cmake/git_submodules.cmake:44 (message):
  Git submodule components/bt/lib is out of date.  Run 'git submodule update
  --init --recursive' to fix.
Call Stack (most recent call first):
  C:/SysGCC/esp32/esp-idf/master/tools/cmake/idf_functions.cmake:233 (git_submodule_check)
  C:/SysGCC/esp32/esp-idf/master/CMakeLists.txt:78 (idf_get_git_revision)

CMake Warning at C:/SysGCC/esp32/esp-idf/master/tools/cmake/git_submodules.cmake:44 (message):
  Git submodule components/coap/libcoap is out of date.  Run 'git submodule
  update --init --recursive' to fix.
Call Stack (most recent call first):
  C:/SysGCC/esp32/esp-idf/master/tools/cmake/idf_functions.cmake:233 (git_submodule_check)
  C:/SysGCC/esp32/esp-idf/master/CMakeLists.txt:78 (idf_get_git_revision)

CMake Warning at C:/SysGCC/esp32/esp-idf/master/tools/cmake/git_submodules.cmake:44 (message):
  Git submodule components/esp32/lib is out of date.  Run 'git submodule
  update --init --recursive' to fix.
Call Stack (most recent call first):
  C:/SysGCC/esp32/esp-idf/master/tools/cmake/idf_functions.cmake:233 (git_submodule_check)
  C:/SysGCC/esp32/esp-idf/master/CMakeLists.txt:78 (idf_get_git_revision)

CMake Warning at C:/SysGCC/esp32/esp-idf/master/tools/cmake/git_submodules.cmake:44 (message):
  Git submodule components/expat/expat is out of date.  Run 'git submodule
  update --init --recursive' to fix.
Call Stack (most recent call first):
  C:/SysGCC/esp32/esp-idf/master/tools/cmake/idf_functions.cmake:233 (git_submodule_check)
  C:/SysGCC/esp32/esp-idf/master/CMakeLists.txt:78 (idf_get_git_revision)

CMake Warning at C:/SysGCC/esp32/esp-idf/master/tools/cmake/git_submodules.cmake:44 (message):
  Git submodule components/json/cJSON is out of date.  Run 'git submodule
  update --init --recursive' to fix.
Call Stack (most recent call first):
  C:/SysGCC/esp32/esp-idf/master/tools/cmake/idf_functions.cmake:233 (git_submodule_check)
  C:/SysGCC/esp32/esp-idf/master/CMakeLists.txt:78 (idf_get_git_revision)

CMake Warning at C:/SysGCC/esp32/esp-idf/master/tools/cmake/git_submodules.cmake:44 (message):
  Git submodule components/libsodium/libsodium is out of date.  Run 'git
  submodule update --init --recursive' to fix.
Call Stack (most recent call first):
  C:/SysGCC/esp32/esp-idf/master/tools/cmake/idf_functions.cmake:233 (git_submodule_check)
  C:/SysGCC/esp32/esp-idf/master/CMakeLists.txt:78 (idf_get_git_revision)

CMake Warning at C:/SysGCC/esp32/esp-idf/master/tools/cmake/git_submodules.cmake:44 (message):
  Git submodule components/mbedtls/mbedtls is out of date.  Run 'git
  submodule update --init --recursive' to fix.
Call Stack (most recent call first):
  C:/SysGCC/esp32/esp-idf/master/tools/cmake/idf_functions.cmake:233 (git_submodule_check)
  C:/SysGCC/esp32/esp-idf/master/CMakeLists.txt:78 (idf_get_git_revision)

CMake Warning at C:/SysGCC/esp32/esp-idf/master/tools/cmake/git_submodules.cmake:44 (message):
  Git submodule components/micro-ecc/micro-ecc is out of date.  Run 'git
[... 23 lines cut ...]
  C:/SysGCC/esp32/esp-idf/master/tools/cmake/idf_functions.cmake:233 (git_submodule_check)
  C:/SysGCC/esp32/esp-idf/master/CMakeLists.txt:78 (idf_get_git_revision)

-- Project is not inside a git repository,                         will not use 'git describe' to determine PROJECT_VER.
-- Project version: 1
C:\SysGCC\esp32\esp-idf\master\tools\cmake\components.cmake(88): error : C:/SysGCC/esp32/esp-idf/master/components/mbedtls/CMakeLists.txt:
  C:/SysGCC/esp32/esp-idf/master/components/mbedtls/CMakeLists.txt:
  COMPONENT_ADD_INCLUDEDIRS entry 'mbedtls/include' not found
Call Stack (most recent call first):
  C:/SysGCC/esp32/esp-idf/master/components/mbedtls/CMakeLists.txt:5 (register_component)

-- Configuring incomplete, errors occurred!
[... cut ...]
-------------------------------------------------------------
Command exited with code 1
Executable: C:\Users\braun\AppData\Local\VisualGDB\CMake\bin\cmake.exe
Arguments: ../.. -G "Ninja" -DCMAKE_BUILD_TYPE=DEBUG -DCMAKE_COLOR_MAKEFILE=OFF -DCMAKE_MAKE_PROGRAM="C:/Program Files (x86)/Sysprogs/VisualGDB/ninja.exe"
Directory: C:\VisualGdb\fileserver2\fileserver2/VisualGDB/Debug
VisualGDB: Error: Command-line action failed
========== Erstellen: 0 erfolgreich, 1 fehlerhaft, 0 aktuell, 0 übersprungen ==========

renzbagaporo, 2019-03-11:
Hi @arnold-b, can you verify if `mbedtls/include` is really in the mbedtls component directory (IDF_PATH/components/mbedtls)? There seems to be some warning for outdated git submodules.

arnold-b, 2019-03-13:
Sorry for that. You are right. i should learn to read the warnings.. 
It is not there.

closed 2019-03-13 by arnold-b

amrutnrp, 2020-04-28:
Well, did it reach to a solution?  I am facing the exact same problem in linux.

> Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory '/home/kanha/esp/esp-idf_master1/hello_world/build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
Running cmake in directory /home/kanha/esp/esp-idf_master1/hello_world/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DIDF_TARGET=esp32 -DCCACHE_ENABLE=0 /home/kanha/esp/esp-idf_master1/hello_world"...
Warn about uninitialized values.
-- Found Git: /usr/bin/git (found version "2.17.1") 
fatal: not a git repository (or any of the parent directories): .git
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/kanha/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/kanha/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/kanha/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/kanha/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /home/kanha/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
CMake Error at ../tools/cmake/component.cmake:302 (message):
  Include directory
  '/home/kanha/esp/esp-idf_master1/components/mbedtls/mbedtls/include' is not
  a directory.
Call Stack (most recent call first):
  ../tools/cmake/component.cmake:470 (__component_add_include_dirs)
  ../components/mbedtls/CMakeLists.txt:4 (idf_component_register)

> -- Configuring incomplete, errors occurred!
See also "/home/kanha/esp/esp-idf_master1/hello_world/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

A solution would be helpful.
Thanks

crgarcia12, 2020-09-14:
@amrutnrp Did you fix this? I am getting it now :(