espressif/esp-idf #3155: file_serving example error in creating ctrl socket (112)

arnold-b, 2019-03-11:
Hi, 

i get this error with fileserver example. but the example works.

I (4667) event: sta ip: 192.168.1.6, mask: 255.255.255.0, gw: 192.168.1.1
I (4667) TEST_WIFI: SYSTEM_EVENT_STA_GOT_IP
I (4667) TEST_WIFI: Got IP: '192.168.1.6'
E (4667) httpd: httpd_server_init: error in creating ctrl socket (112)

used sample v3.3\examples\protocols\http_server\file_serving
windows 10
visual gdb current version
esp32 wroom

anurag-kar, 2019-03-12:
@arnold-b Please clarify

> i get this error with fileserver example. but the example works.

What do you mean by example works? Is the file server accessible via a browser (eg. try opening  http://192.168.1.6/)? Also, it will help if you can paste the full log

Alvin1Zhang, 2019-03-13:
@arnold-b Hi, would you help provide any updates for the issue? Thanks.

arnold-b, 2019-03-13:
Sorry, 
i didnt see this. 
I just mean i get the error message in the log, but i still can access the files. later i will post the full log

anurag-kar, 2019-04-09:
@arnold-b  I'm going to close the issue for now. Please feel free to open it later.

closed 2019-04-09 by anurag-kar

htjulia-dev, 2025-05-12:
httpd_config_t ctrl_port already used  in httpd_server_init.  
change the ctrl port.
default is (#define ESP_HTTPD_DEF_CTRL_PORT         (32768) ).

httpd_handle_t httpd_server = NULL;
httpd_config_t httpd_config = HTTPD_DEFAULT_CONFIG();
httpd_config.ctrl_port = 32767; //example.
.....
httpd_start(&httpd_server, &httpd_config) == ESP_OK;