espressif/esp-idf #3157: how to open a file which contain simplified chinese in the file name?

ichenxing, 2019-03-11:


Alvin1Zhang, 2019-03-11:
@ichenxing Hi, would you help provide a screen shot of the said file? Thanks.

ichenxing, 2019-03-11:
@Alvin1Zhang 
[001安全第一我知道.zip](https://github.com/espressif/esp-idf/files/2951585/001.zip)

ichenxing, 2019-03-11:
@Alvin1Zhang 

test code:
```
void audio_play_for_list(void)
{
    static DIR* dir = NULL;
    if(dir == NULL){
        dir = opendir("/sdcard/music");
        if(dir == NULL){
            ESP_LOGE(TAG, "open file dir fail");
            return;
        }
    }

    struct dirent* de = readdir(dir);
    if (!de) {
        ESP_LOGE(TAG, "get file info fail");
        return;
    }
    // ESP_LOGI(TAG, "%s, %d", de->d_name, de->d_ino);

    // memcpy(de->d_name, "001安全第一我知道.mp3", strlen("001安全第一我知道.mp3"));

    for(int i=0; i<strlen(de->d_name); i++){
        printf("0x%02X ", de->d_name[i]);
    }
    printf("\r\n");

    char temp[128];
    sprintf(temp, "file://sdcard/music/%s", de->d_name);
    ESP_LOGI(TAG, "play url: %s", temp);

    audio_play_from_url(temp);
}
```
print log:
```
0x30 0x30 0x31 0xB0 0xB2 0xC8 0xAB 0xB5 0xDA 0xD2 0xBB 0xCE 0xD2 0xD6 0xAA 0xB5 0xC0 0x2E 0x6D 0x70 0x33
I (28431) FILE_SCAN: play url: file://sdcard/music/001▒▒ȫ▒▒һ▒▒֪▒▒.mp3
I (28441) ESP_AUDIO_CTRL: Func:media_ctrl_play, Line:301, MEM Total:4265584 Bytes, Inter:165684 Bytes, Dram:150768 Bytes

E (28451) ESP_AUDIO_CTRL: URI Parse error : 1
```