espressif/esp-idf #3158: New WROVER modules are missing from HW documentation

gmos, 2019-03-11:
## Environment
Documentation https://docs.espressif.com/projects/esp-idf/en/stable/hw-reference/modules-and-boards.html#wroom-and-wrover-modules

## Problem Description
WROVER modules with new external RAM and Flash sizes are missing from table.

### Expected Behavior
8MB RAM and 8/16GB Flash models added.

### Actual Behavior
n/a
### Steps to reproduce
n/a
### Code to reproduce this issue
n/a
## Debug Logs
n/a

igrr, 2019-03-11:
Thanks for raising this issue. I'm going to close it as the issue is not related to ESP-IDF project. Please use the "Contact us" section of the website to report such issues, or post them on the esp32.com forum in Documentation requests/feedback topic: https://esp32.com/viewtopic.php?f=5&t=905.

krzychb, 2019-07-19:
Resolved in 752c7c32d460a92903adb50a60b33db729bfc28a.

Thank for reporting @gmos!

closed 2019-07-19 by krzychb