espressif/esp-idf #3159: xEventGroupSetBits and xEventGroupClearBits are used in wifi_event_handler.

nopnop2002, 2019-03-11:
xEventGroupSetBits and xEventGroupClearBits are used in wifi_event_handler.
Aren't they not xEventGroupSetBits and xEventGroupClearBits, but xEventGroupSetBitsFromISR and xEventGroupClearBitsFromISR?

	static esp_err_t wifi_event_handler(void *ctx, system_event_t *event)
	{
	    switch (event->event_id) {
	        case SYSTEM_EVENT_STA_START:
	            esp_wifi_connect();
	            break;
	        case SYSTEM_EVENT_STA_GOT_IP:
	            xEventGroupSetBits(wifi_event_group, CONNECTED_BIT);

	            break;
	        case SYSTEM_EVENT_STA_DISCONNECTED:
	            esp_wifi_connect();
	            xEventGroupClearBits(wifi_event_group, CONNECTED_BIT);
	            break;
	        default:
	            break;
	    }
	    return ESP_OK;
	}

negativekelvin, 2019-03-11:
wifi_event_handler doesn't run in isr context

nopnop2002, 2019-03-12:
Thank you for answer.
OK. 
Please close this issues.

closed 2019-03-12 by Alvin1Zhang