espressif/esp-idf #3161: Http content type

arnold-b, 2019-03-11:
Hi, 

is there a easy way to get the http content type back from the request ?
its written with httpd_resp_set_type(req, "text/css");

and now i want to read it again. (like resp_get_type )
if not. how can i include esp_httpd_priv.h

esp-idv v3.3 released 
thanks

anurag-kar, 2019-03-12:
@arnold-b No, presently there is no API for getting back the content type. Though it's puzzling why you need this functionality? To clarify things, `httpd_resp_set_type()` is meant to be called only from inside a request handler (a function which is executed when HTTP request is received by the server) for the purpose of creating an HTTP response. Note that the default content type (before setting a new type with `httpd_resp_set_type()`) is 'text/html'. Now, since `httpd_resp_set_type()` can only be called from a limited scope, the content type parameter, if set, will be known to the user inside the scope of the handler, therefore having an API like `httpd_resp_get_type()` is an overkill.

Since this is more of a question than an issue I suggest that you start a discussion at https://esp32.com/

closed 2019-03-12 by Alvin1Zhang

arnold-b, 2019-03-13:
Why its closed without any comment. thats strange

arnold-b, 2019-03-13:
Its makes it just more easy. Because i set it once, i know. but in every function i just use req for handling the requests.