espressif/esp-idf #3162: SPI slave broken by commit 58955a

dralves, 2019-03-12:
## Environment

- Development Kit:      [ESP32-DevKitC|]

- Using VSPI with IOMUX pins as described in https://github.com/espressif/esp-idf/blob/master/docs/en/api-reference/peripherals/spi_slave.rst
- No dma

- Init code:
`    spi_bus_config_t buscfg = {
      .mosi_io_num = config_.gpio_mosi,
      .miso_io_num = config_.gpio_miso,
      .sclk_io_num = config_.gpio_sclk,
      .quadwp_io_num = -1,
      .quadhd_io_num = -1
    };
    buscfg.max_transfer_sz = OBCI_SPI_PACKET_SIZE_BYTES;

    spi_slave_interface_config_t slvcfg = {
      .spics_io_num = config_.gpio_ss,
      .flags = 0,
      .queue_size = 1,
      .mode = 0
    };

    ...
    RETURN_NOT_OK(spi_slave_initialize(host_, &buscfg, &slvcfg, 0));`

## Problem Description

Using ESP32 as an SPI slave, running against a non-esp MCU master. As found by git bisect, before commit 58955a data was transmitted reliably, after that commit started getting errors on the master (see below for the types of errors). The anatomy of the error (1 or 2 bit shifts) seems to hint at timing problems.

Reverting the changes to spi_slave.c on top of master as of a few days ago (ebdcbe8c) makes the problem go away.

### Expected Behavior

Data transmitted by the slave arrives at the master as sent.

### Actual Behavior

Data is usually bit shifted by 1 bit (e.g. an xFE byte sent by the ESP32 slave becomes xFF at the master), sometimes (rarely) arrives correctly, sometimes it's bit shifted by 2 bits.

igrr, 2019-03-14:
Sorry could you please double check the commit id — 58955a you've mentioned doesn't seem to be a commit in IDF repo?

dralves, 2019-03-14:
Hi @igrr 

  I was missing one more letter to be uniquely identifiable :)
  I changed the title/links to the right one, but here's a more direct one: https://github.com/espressif/esp-idf/commit/58955a79a27d3c7331eaec6e464878df42615a36
  I'm talking specifically about the changes to spi_slave.c. I haven't tested or reverted the other changes.

ginkgm, 2019-03-14:
@dralves Can you provide the MCU you are using, as well as the spi clock speed?

The problem is that, ESP32 slave has a delay (quite large!!!!!) on the MISO line after the SPI clock launch edge. When the GPIO matrix is used, it's 62.5ns, and if IOMUX isused, it's 37.5ns. Which means, it cannot meet the timing requirements when the SPI clock is above 8MHz (GPIO matrix), or 13MHz (IOMUX) (See the [programming guide](https://espressif-docs.readthedocs-hosted.com/projects/esp-idf/en/latest/api-reference/peripherals/spi_slave.html#sclk-frequency-requirement)).

In the code before, we mistakenly shifted the timing of both launch and latch edge by half a spi clock ahead. This solves some DMA issues, and made the timing performance better in some high frequency cases.

But basicly it's incorrect, and complained by someone else in #1346 and #2393 . So we fixed it.

If you prefer the timing configurations, you can:
1. If you are using mode 1/3, set the mode of slave to mode 0/2, it's half a clock ahead then mode 1/3. But the DMA should be disabled.
2. If you are using mode 0/2, turn on the DMA. The workaround in 58955a7 will help you shift the edges half a clock ahead.

dralves, 2019-03-14:
Hi @ginkgm 

  The MCU/SPI master is a PIC32, mode 0, the master sets the speed at 10Mhz.
  I read the docs you wrote :), so even though at first I was using the GPIO matrix I changed to IOMUX. Then I started reducing the speed I went all the way down to 5MHz and stlll the comms were unreliable. That's when I finally tried looking into reverting the changes. SPI had been super-reliable before.

dralves, 2019-03-14:
Oh, btw, not sure if relevant this exact same code & mode & frequency on the master works flawlessly against an ESP8266.

dralves, 2019-03-14:
Another note I did see the comma becoming slightly better as I reduced speed. But at 5 MHz they were still very unreliable and that’s the lowest I could go in terms of speed

dralves, 2019-03-14:
Yet another note:
The table in #1346 states:
Registers | mode0 | mode1 | mode2 | mode3
-- | -- | -- | -- | --
SPI_CK_IDLE_EDGE | 0 | 0 | 1 | 1
SPI_CK_I_EDGE | 0 | 1 | 1 | 0
SPI_MISO_DELAY_MODE | 0 | 0 | 0 | 0
SPI_MISO_DELAY_NUM | 0 | 0 | 0 | 0
SPI_MOSI_DELAY_MODE | 2 | 1 | 1 | 2
SPI_MOSI_DELAY_NUM | 0 | 0 | 0 | 0

but the currently checked in code is:

```
    if (mode == 0) {
        //The timing needs to be fixed to meet the requirements of DMA
        spihost[host]->hw->pin.ck_idle_edge = 1;
        spihost[host]->hw->user.ck_i_edge = 0;
        spihost[host]->hw->ctrl2.miso_delay_mode = 0;
        spihost[host]->hw->ctrl2.miso_delay_num = 0;
        spihost[host]->hw->ctrl2.mosi_delay_mode = 2;
        spihost[host]->hw->ctrl2.mosi_delay_num = 2;
```
If I understand things correctly, this doesn't match the table. Not sure whether the table is accurate, but according to the table this should be:
```
    if (mode == 0) {
        //The timing needs to be fixed to meet the requirements of DMA
        spihost[host]->hw->pin.ck_idle_edge = 0;
        spihost[host]->hw->user.ck_i_edge = 0;
        spihost[host]->hw->ctrl2.miso_delay_mode = 0;
        spihost[host]->hw->ctrl2.miso_delay_num = 0;
        spihost[host]->hw->ctrl2.mosi_delay_mode = 2;
        spihost[host]->hw->ctrl2.mosi_delay_num = 0;
```

ginkgm, 2019-03-14:
@dralves 
The TRM is already updated: https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf
[image]

dralves, 2019-03-14:
@ginkgm might it be hardware rev dependent? I have quite a few esp32 boards, but mainly use adafruits huzzah32 with the ESP-WROOM-32 (not sure which rev, but think it's pretty old). I say this because the current implementation is supposed to work well at low frequencies, (<7MHz even if I got the GPIO/IOMUX thing wrong) and that's not what I observed.

dralves, 2019-03-14:
@ginkgm in any case if this is only a problem for me, I already have my own fork of esp-idf (where I include arduino as a component) so I can just maintain my fork with this additional change. Feel free to close this if you think it's not relevant/not a problem for other folks.
I just raised the issue in case others run into the same problem where their boards are working fine and suddenly stop working after an update.

ginkgm, 2019-03-14:
Some suggestions on using PIC32: 
I assume you are using the model as this [spec](http://ww1.microchip.com/downloads/en/DeviceDoc/PIC32MX1XX2XX-28-36-44-PIN-DS60001168K.pdf)

[image]
Mode 0 correspond to CKP=0, CKE=1. 
[image]
SP40=15ns, 1/(15ns+62.5ns)/2=6.5MHz. I think the slack is small, and maybe you are still via the GPIO matrix? Maybe you can enable the debug message in the spi_common.c to see whether you're using the IOMUX or not.

Finally, you can set SMP=1 to delay the master sample time, or use the trick I mentioned above to advance the slave launch time.

ginkgm, 2019-03-14:
thanks @dralves

dralves, 2019-03-18:
@ginkgm 
[image]

ginkgm, 2019-03-21:
@dralves thanks alot!