espressif/esp-idf #3163: stdatomic.h 64-bit atomic access support

johnnyxwan, 2019-03-12:
## Environment

- Development Kit: Custom board
- Kit version (for WroverKit/PicoKit/DevKitC): N/A
- Module or chip used: ESP32
- IDF version: v3.1.3
- Build System: CMake (Tried Make as well)
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System: Windows (WSL Ubuntu)
- Power Supply: USB

## Problem Description

**stdatomic.h 64-bit atomic access support**
I am not using atomic library directly, instead I am importing a 3rd party library which depends on it.

### Expected Behavior

If I compile the code below (32-bit unsigned int), it can be compiled successfully.
`#include "stdatomic.h"`
`atomic_uint_fast32_t* test(0);`
`atomic_fetch_add_explicit(test, 1, __ATOMIC_RELAXED);`

### Actual Behavior

When compiling the code below (64-bit unsigned int), linker shows: undefined reference to '__atomic_fetch_add_8' and failed to compile.
`#include "stdatomic.h"`
`atomic_uint_fast64_t* test(0);`
`atomic_fetch_add_explicit(test, 1, __ATOMIC_RELAXED);`

### Summary

Seems that the compiler didn't generate the __atomic_xxxxxx_N version for 64-bit.
Is 64-bit supported? Or I just I did something wrongly?

For reference, I am trying to use this library:
https://github.com/wqking/eventpp

projectgus, 2019-03-12:
Hi @johnnyxwan ,

Unfortunately, ESP32 only has hardware support for 32-bit atomic operations. There is no built-in support for updating memory which is wider than 32 bits, atomically.

Angus

johnnyxwan, 2019-03-12:
But I would expect the compiler would generate a "locked" version (software simulated version) of that function, maybe with compilation warning, for the ease of code porting from other libraries.

dsptech, 2019-03-15:
Hi, @johnnyxwan 
If you are stucked in this issue, you could try to write your own __atomic_xxxxxx_8 and placing it in any C sourcecode. 
Some time ago, I used a similar way on esp8266 to avoid changes in a library.
Regards.

johnnyxwan, 2019-03-15:
@dsptech,
Thank you for your suggestion, quite a good one, should not be very hard with some critical sessions I guess. But instead, I have modded the library to fit in the 32-bit environment at this moment. Really look forward to see esp-idf officially support that.

closed 2021-05-17 by espressif-bot