espressif/esp-idf #3165: Null data frame (or non QOS) can't generate CSI data

MyDog3Legs, 2019-03-12:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it): 
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows|Linux|macOS]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to repropduce

1. step1
2. ...

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

MyDog3Legs, 2019-03-12:
According to the docs, disconnected frames are always transmitted at 1mbs. OFDM is not used until data rate hits 6mbs, so there can be no CSI info unless the node is connected. CSI is useful for spectral scanning and software beam forming, but not so useful if the connection is already made. The ability to send a null frame at least 6mbs would be very useful.

TimXia, 2019-03-18:
@MyDog3Legs Sorry, what is the problem? If the connection is already made, the station does send a null frame at 6mbs very 20s to keep connection.

MyDog3Legs, 2019-03-25:
Yes, it works fine when the connection is made, but if there is no connection, everything goes at 1mbs, too slow for OFDM to cut in so there is no CSI. If I send a null frame to a disconnected node, the ACK will come back using CCK. It is very useful to be able to transmit a NULL frame and receive the ACK at a minimum of 6mbs, in order to enable OFDM and get csi info from clients in range, not just the ones connected

Sent with [ProtonMail](https://protonmail.com) Secure Email.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

MaxwellAlan, 2021-03-01:
hi @MyDog3Legs , we have supported disable tx 11b rate in internal esp-idf master branch, and this feature will be synced to github esp-idf master some days, you can have a try then.
Thanks.

NiyamPoudel, 2021-07-23:
[image]
Hello @MaxwellAlan , @TimXia , @MyDog3Legs . I am quite new to this whole thing. I am trying to understand the CSI data I am getting based on the above chart. Could you please help me understand this or point me towards a useful resource. Thanks. 

[112 -128 6 0 0 0 0 0 0 0 0 0 3 9 2 10 1 12 0 14 1 14 1 16 2 16 3 16 3 16 4 15 5 15 5 14 5 13 4 12 2 11 1 11 0 11 -1 11 -2 12 -3 13 -3 14 -3 15 -2 16 -1 17 0 17 1 17 0 0 2 15 2 14 2 14 1 12 0 12 -1 12 -2 12 -4 12 -4 13 -4 15 -4 16 -4 17 -3 18 -3 19 -1 19 0 19 1 18 2 17 1 16 2 15 0 14 -1 14 -2 13 -3 14 -4 15 -6 16 0 0 0 0 0 0 0 0 0 0 ]

Here is a sample of CSI Matrix that I am getting.

MaxwellAlan, 2021-07-27:
Hi @NiyamPoudel , maybe you refer to this link: https://dhalperi.github.io/linux-80211n-csitool/
Thanks.

closed 2021-09-14 by Alvin1Zhang