espressif/esp-idf #3170: ESP32 use_apll feature does not work

srini1948, 2019-03-12:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it): 
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows|Linux|macOS]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to repropduce

1. step1
2. ...

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

srini1948, 2019-03-12:
When I enable use_apll feature in the I2S config to get accurate audio sampling rate the data generated seems to show a much lower sampling rate!

projectgus, 2019-03-13:
@srini1948 Please edit the initial post of this Issue and fill in all of the requested information, so we can debug this issue.

srini1948, 2019-03-13:
Environment
Development Kit: [Wemos Lolin32]
Kit version: V2
Module or chip used:  ESP32-WROOM-32 ( rev. 1)
IDF : v3.2
Build System: [Make|
Compiler version:
Operating System: Windows
Power Supply: [USB|
Problem Description
//Detailed problem description goes here.
I am using I2S to capture data from the built in ADC. In order to get a more accurate sampling rate as suggested in the IDF instruction manual,  I decided to use the "use_apll" feature in the configuration. My sampling rate was 32000 samples per second. Upon capturing 256 samples of a 1 kHz input sine wave I should see 8 cycles of the waveform. Instead I see just one cycle - meaning that the actual sampling rate was 32*8 = 256. The message during the run shows that the requested rate and real rate are identical.

With "use_apll" set to false the capture is perfect.

koobest, 2019-04-12:
Hi, @srini1948 
Have you solved this issue? I think if you specify the parameter  `fixed_mclk` of `i2s_config_t` , this issue can be solved. you also need to change the `i2s_channel_fmt_t`.  the following is my test code, you can have a try. The output of  GPIO21 should be same with the GPIO19, and you can see the ADC sample rate on GPIO18.

```
#include <stdio.h>
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_spi_flash.h"
#include "esp_err.h"
#include "esp_log.h"
#include "driver/i2s.h"
#include "driver/adc.h"
#include "driver/ledc.h"

#define EXAMPLE_I2S_NUM           (0)
#define EXAMPLE_I2S_SAMPLE_RATE   (32000)
#define EXAMPLE_I2S_SAMPLE_BITS   (16)
#define EXAMPLE_I2S_FORMAT        (I2S_CHANNEL_FMT_ONLY_RIGHT)
#define I2S_ADC_UNIT              (ADC_UNIT_1)
#define I2S_ADC_CHANNEL           (ADC1_CHANNEL_0)
#define I2S_ADC_SAMPLE_CNT        (256)

//Out put WS signal from gpio18(only for debug mode)
void i2s_adc_check_clk(void)
{
    PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[18], PIN_FUNC_GPIO);
    gpio_set_direction(18, GPIO_MODE_DEF_OUTPUT);
    gpio_matrix_out(18, I2S0I_WS_OUT_IDX, 0, 0);

    PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[21], PIN_FUNC_GPIO);
    gpio_set_direction(21, GPIO_MODE_DEF_OUTPUT);
    gpio_set_level(21, 0);
}

void ledc_init(void)
{
    ledc_timer_config_t ledc_timer = {
        .duty_resolution = LEDC_TIMER_13_BIT, // resolution of PWM duty
        .freq_hz = 1000,                      // frequency of PWM signal
        .speed_mode = LEDC_HIGH_SPEED_MODE,           // timer mode
        .timer_num = LEDC_TIMER_0            // timer index
    };
    // Set configuration of timer0 for high speed channels
    ledc_timer_config(&ledc_timer);
    ledc_channel_config_t ledc_channel = {
        .channel    = LEDC_CHANNEL_0,
        .duty       = 4096,
        .gpio_num   = 19,
        .speed_mode = LEDC_HIGH_SPEED_MODE,
        .hpoint     = 0,
        .timer_sel  = LEDC_TIMER_0,
    };
    ledc_channel_config(&ledc_channel);
}

void app_main(void)
{
    int i2s_num = EXAMPLE_I2S_NUM;
    i2s_config_t i2s_config = {
        .mode = I2S_MODE_MASTER | I2S_MODE_RX | I2S_MODE_ADC_BUILT_IN,
        .sample_rate =  EXAMPLE_I2S_SAMPLE_RATE,
        .bits_per_sample = EXAMPLE_I2S_SAMPLE_BITS,
        .communication_format = I2S_COMM_FORMAT_I2S_MSB,
        .channel_format = EXAMPLE_I2S_FORMAT,
        .intr_alloc_flags = 0,
        .dma_buf_count = 2,
        .dma_buf_len = 1024,
        .use_apll = true,
        .fixed_mclk = 2000000, //When use-apll is enabled, we should  specify the fixwd_mclk. 
    };
    //install and start i2s driver
    i2s_driver_install(i2s_num, &i2s_config, 0, NULL);
    //init ADC pad
    i2s_set_adc_mode(I2S_ADC_UNIT, I2S_ADC_CHANNEL);
    const size_t buf_len = sizeof(uint16_t) * I2S_ADC_SAMPLE_CNT;
    uint16_t *rd_buf = (uint16_t *)malloc(buf_len);
    size_t rd_size = 0;
    i2s_adc_check_clk();
    ledc_init();
    while(1) {
        i2s_read(i2s_num, (void*) rd_buf, buf_len, &rd_size, portMAX_DELAY);
        if(rd_size == buf_len) {
            for(int i = 0; i < I2S_ADC_SAMPLE_CNT; i += 2) {
                gpio_set_level(21, (rd_buf[i+1] & 0xfff) > 512);
                ets_delay_us(31);
                gpio_set_level(21, (rd_buf[i] & 0xfff) > 512);
                ets_delay_us(31);
            }
        }
    }
}
```
thanks !!

srini1948, 2019-04-12:
I will try it. Thanks

srini1948, 2019-04-12:
Any reason why you set mclk to 2000000. The clock is now good - but still the converted values have noise as well as non-linearity.

Unless Espressif improves the ADC it is highly unlikely anyone will use this module for applications requiring high performance.

closed 2020-03-25 by Alvin1Zhang