espressif/esp-idf #3171: some packet loss when using HFP

wgcod, 2019-03-13:
There are some packet loss when using HFP. The phenomenon is that 120 zero have been received, which should be normal audio data.
It's similar to the jamming problem, but there's no jamming problem here, because there's no problem with other Bluetooth devices. 
Packet loss is not a continuous loss, but a random occurrence, but there will be.
[image]

blueMoodBHD, 2019-03-14:
I'm working on packet loss concealment（PLC）for HFP，which can fix single packet loss. And we plan to add mSBC for HFP. Please wait patient. Thanks.

allex1978, 2019-04-17:
> I'm working on packet loss concealment（PLC）for HFP，which can fix single packet loss. And we plan to add mSBC for HFP. Please wait patient. Thanks.

Hi, can you share HFP example? How you do this? I`m trying to rebuild a2dp example to HFP but can`t connect HFP AG. I`m using VoHCI setting.

allex1978, 2019-04-28:
> I'm working on packet loss concealment（PLC）for HFP，which can fix single packet loss. And we plan to add mSBC for HFP. Please wait patient. Thanks.

Hi, any progress with mSBC for HFP codec?

blueMoodBHD, 2019-07-03:
Hi, sorry for later reply,  this have been merged, refer to commit: da13efc17a73e773d8da6cd95c076fadabefc2cc. Please update and test again, thanks.

Alvin1Zhang, 2019-07-03:
@wgcod Thanks for reporting the issue, would you please help verify on the commit: da13efc, and help provide feedback? Thanks.

wgcod, 2019-07-04:
I've been using it for a few days, now with mSBC and PLC, the sound quality has been greatly improved, and no problem has been found yet.Thanks.

Alvin1Zhang, 2019-07-19:
@wgcod Thanks for reporting the issue, feel free to reopen the issue if it still exists. Thanks.