espressif/esp-idf #3173: problem with Linker Fragment Files / Mapping Entries: Type I

harryzz, 2019-03-13:
for example when trying to build lwip 
with option LWIP_IRAM_OPTIMIZATION = y 
build failed.
Traceback (most recent call last):
  File "C:/msys32/home/xxxx/esp/esp-idf/tools/ldgen/ldgen.py", line 113, in <module>
    main()
  File "C:/msys32/home/xxxx/esp/esp-idf/tools/ldgen/ldgen.py", line 97, in main
    mapping_rules = generation_model.generate_rules(sdkconfig, sections_infos)
  File "C:/msys32/home/xxxx/esp/esp-idf/tools/ldgen/generation.py", line 370, in generate_rules
    self._create_exclusions(mapping_rules, default_rules, sections_infos)
  File "C:/msys32/home/xxxx/esp/esp-idf/tools/ldgen/generation.py", line 465, in _create_exclusions
    general_rule.add_exclusion(specific_rule, sections_info)
  File "C:/msys32/home/xxxx/esp/esp-idf/tools/ldgen/generation.py", line 99, in add_exclusion
    expansions = do_section_expansion(self, section)
  File "C:/msys32/home/xxxx/esp/esp-idf/tools/ldgen/generation.py", line 81, in do_section_expansion
    sections_in_obj = sections_infos.get_obj_sections(rule.archive, rule.obj)
  File "C:/msys32/home/xxxx/esp/esp-idf/tools/ldgen/generation.py", line 634, in get_obj_sections
    stored = self.sections[archive]
KeyError: 'liblwip.a'
make: *** [/home/xxxx/esp/esp-idf/components/esp32/Makefile.projbuild:51: /home/......../esp32/esp32.common.ld] Error 1
The terminal process terminated with exit code: 2 

br
Zahari

projectgus, 2019-03-20:
Hi @harryzz ,

Thanks for reporting this, there was an issue with path management on Windows. A fix is coming shortly.

Angus

javqui, 2019-03-21:
I'm in the same boat

closed 2019-03-26 by igrr