espressif/esp-idf #3174: [HID] HID over GATT : How to force virtual keyboard to be displayed on iphone

Marc-Aurele, 2019-03-13:
Hello,

When i pair a HID device with an iphone, my device is idendified as a physical keyboard. As a consequence, the virtual keyboard disappear on iphone 5 and 8 and i don't know how to enable it again. Indeed, i want the user to be able to send sms or stuff like that. I found that i can send an EJECT (0xB8) event to force the virtual keyboard to be displayed, but the view needs to be focused on an edit box.

Actually, i'm searching for a solution to automatically set the virtual keyboard on  when my device is connected to the iphone.

For now, i'm sending the following consumer control event at connection : 
 - AC Search
 - EJECT
 - AC Home
But this is not fully reliable ...
If anyone has an idea how to do it, don't hesitate.

Best regards,

Aurélien

Weijian-Espressif, 2019-03-22:
@Marc-Aurele, I don't know how to solve it.

EasyRiderr, 2019-03-22:
@Weijian-Espressif Did you already heard about any tricks that could allow to work around this issue as it will be difficult to fix the bug on the iPhone side ?

Many thanks in advance,

Yoan

Weijian-Espressif, 2019-04-25:
@EasyRiderr, I have no idea how to solve it.

Alvin1Zhang, 2019-07-05:
Thanks for reporting the issue. Feel free to reopen it if your issue still exists. Thanks.

closed 2019-07-05 by Alvin1Zhang

chegewara, 2020-02-03:
Hi @Marc-Aurele 
i remember you asked about it and today i found just scrap of info, but you may try to dig deeper(page 4):
https://cdn.sparkfun.com/datasheets/Wireless/Bluetooth/RN-HID-User-Guide-v1.0r.pdf

There is no described how to do it, but obviously it is possible.

Marc-Aurele, 2020-02-03:
Hi @chegewara 
Thank you very much for the information :)
Indeed, you are right, it is possible for sure, but unfortunatly now, the product is available for sale but i won't hesitate to take a look at it again if i need to make some upgrade.
Thx again ;)