espressif/esp-idf #3176: RTC XTAL current fix reverted?

JonathanWitthoeft, 2019-03-13:
Regarding the extra current fix using the touch pad for external 32k XTAL, why was this commit reverted? I am implementing this fix as it has solved the issue of my boards getting into a boot-loop, and I am wondering if there is anything wrong with this implementation.

https://github.com/espressif/esp-idf/commit/dc133f9fc4d33e103aedf3704b40ef276486a348#diff-514610947bd67f035310298823b9e4ab

Best Regards
Jonathan Witthoeft

igrr, 2019-03-14:
The extra current feature was reverted because it has caused a slight regression in the default use case (no extra current). An updated version is in review now and should be merged soon.

JonathanWitthoeft, 2019-03-14:
So this will not cause unseen error for my use case where the extra current is needed? I am only asking as I re-implemented it in the mean time.

igrr, 2019-04-10:
Sorry for the delay, has been merged again as https://github.com/espressif/esp-idf/commit/ae585b6615419399e31e83a89ee710c6ca7f1a09. Backported to release/v3.3 and will be backported to release/v3.2, release/v3.1 branches as well.

closed 2019-04-10 by igrr