espressif/esp-idf #3179: esp_bt_controller_mem_release() crash  with v3.3-beta2

kvbulusu, 2019-03-14:
==================

## Environment

- Development Kit:  ESP32-LyraT
- Module or chip used: ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it):   v3.3-beta2-60-g86c9b6354 )
- Build System:     Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 5.20
- Operating System:   Linux
- Power Supply:      USB

## Problem Description

 ble_ibeacon example under esp-idf/examples/bluetooth ,always crashes when it tries to release the controller memory ( esp_bt_controller_mem_release() )  . The ibeacon demo example works standalone . However it doesnt with when we need to turn off wifi , initialize ble/bt, scan for beacon.

### Expected Behavior

### Actual Behavior

### Steps to repropduce

1. Initialize wifi and connect to station
2.  Stop Wifi
3.  Use  ibeacon_demo.c 

### Code to reproduce this issue

```cpp
===Code snippet=========================
    ESP_ERROR_CHECK(esp_wifi_stop() );
    ESP_ERROR_CHECK(esp_wifi_deinit() );
    ESP_ERROR_CHECK(nvs_flash_init());
    ESP_ERROR_CHECK(esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT));
    esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
    esp_bt_controller_init(&bt_cfg);
    esp_bt_controller_enable(ESP_BT_MODE_BLE);
    ble_ibeacon_init();

    /* set scan parameters */
#if (IBEACON_MODE == IBEACON_RECEIVER)
    ESP_LOGW(TAG, "IBEACON_MODE: IBEACON_RECEIVER  \n");
    esp_ble_gap_set_scan_params(&ble_scan_params);

    ESP_LOGW(TAG, "Initialize Wifi again \n");
    wifi_init_sta();

```

## Debug Logs
abort() was called at PC 0x40108acb on core 0
0x40108acb: app_main at 19Mar12/esp-avs-sdk/examples/ble_ibeacon/main/ibeacon_demo.c:975 (discriminator 2)

ELF file SHA256: 318f883c12abc674faabce3dc22693803511dbb428187f3fb4bf9992338a5923

Backtrace: 0x4009094b:0x3ffd32f0 0x40090ad1:0x3ffd3310 0x40108acb:0x3ffd3330 0x400d0ad8:0x3ffd33d0
0x4009094b: invoke_abort at 19Mar12/esp-idf/components/esp32/panic.c:136

0x40090ad1: abort at 19Mar12/esp-idf/components/esp32/panic.c:171

0x40108acb: app_main at 19Mar12/esp-avs-sdk/examples/ble_ibeacon/main/ibeacon_demo.c:975 (discriminator 2)

0x400d0ad8: main_task at 19Mar12/esp-idf/components/esp32/cpu_start.c:530

## Other items if possible

Weijian-Espressif, 2019-03-14:
@kvbulusu,  please disconnect WIFI before calling `esp_wifi_stop()`, then you can deinit WIFI.

Weijian-Espressif, 2019-03-14:
[ibeacon_demo.c.zip](https://github.com/espressif/esp-idf/files/2965603/ibeacon_demo.c.zip)

kvbulusu, 2019-03-21:
@Weijian-Espressif  Thank you!! It works. I was using esp-avs-sdk repo and its using 
 provisioning and released mem using conn_mgr_prov_mem_release(). So when I added esp_bt_controller_mem_release() , its crashing.  But your example was very good and helped a lot.

closed 2019-03-21 by kvbulusu