espressif/esp-idf #3180: esp_http_client socket error

keesus, 2019-03-14:
- Development Kit:  ESP32-Wrover
- Module or chip used: ESP32-WROVER
- IDF version : v3.2-dev-1148-g96cd3b75c
- Build System: CMake
- Compiler version : 1.22.0-80-g6c4433a
- Operating System:  mac
- Power Supply: externa 5V
-----------------------------------------------

Hi. With esp_http_client.c,
When I try to make http connections over 10, I always see following error.

-
E (51355) TRANS_TCP: Error create socket
E (51365) HTTP_CLIENT: Connection failed, sock < 0
E (51365) HTTP_CLIENT: HTTP POST request failed: ESP_ERR_HTTP_CONNECT
-

I tried esp_http_client_cleanup() but nothing changed. Also I'm aware of that there is an related option in sdkconfig to increase maximum socket number but not perfect solution due to my application requirements.

what am i supposed to do ?

Thanks.

keesus, 2019-03-18:
anybody please help me out I can't even find related issue so I have no idea.

jitin17, 2019-03-18:
@keesus What is the maximum number of connections you expect to invoke in your application? 
By default, the maximum number of open sockets is 10, and it can be configured under menuconfig to be 16(max val). As you might know, sockets consumes certain amout of memory, which is why there is a limitation on maximum number of open sockets. Yet, it would be insightful if you can give more information about how many open connections you expect to have in your application.

keesus, 2019-03-18:
@jitin17 

Hi.
I expect those HTTP connections would be closed and cleaned up on every usage.
In other words, I want to use HTTP connections without a limitation.

*After few hard works, esp_http_client_cleanup does not working I guess.
*esp-idf/examples/protocols/esp_http_client.c works fine. but once I copy example codes to my project, the error occurs.

keesus, 2019-03-19:
problem resolved.

I defined function named "close()" on another module and It caused a overlap with ESP-IDF function "close()".

closed 2019-03-19 by keesus

Mauricio-Alencar, 2022-02-09:
Keesus, 
Can you please explain what you did to solve the problem?

CQC2016, 2023-06-19:
@keesus Hi Keesus, is just a "close()" redefined ? 
I am facing the same problem, any updates ?