espressif/esp-idf #3181: Bootloader as Submodule

gregjesl, 2019-03-14:
## Problem

We are preparing to send devices containing an ESP32 to customers.  Once we send the units out, the only way of updating firmware will be via OTA updates; this means that the bootloader is locked in at the time of shipment.  

We would like to continue updating our IDF once we deploy units but pulling updates from the IDF results in pulling changes to the bootloader code as well.  During testing we would like to use the same bootloader that is deployed to the units.  I recognize that we could always just use `app-flash` instead of `flash` during testing, but from a configuration management standpoint this is not ideal.  

## Solution

Would it be possible to make the bootloader a submodule?  This would allow us to lock in the submodule on our fork of the IDF while continuing to pull updates from the rest of the IDF.

negativekelvin, 2019-03-14:
You could probably put the bootloader binary in your project's own bootloader component and use a component makefile that copies it to the build directory.

gregjesl, 2019-03-15:
@negativekelvin That would work for us.  Thank you for the idea.

closed 2019-03-15 by gregjesl