espressif/esp-idf #3183: esp_http_server calls accept() too much, runs out of sockets

jimparis, 2019-03-14:
Running esp-idf recent master (ff020c3a18788d13f9dc8e853424d0067e79a357)

If a client opens too many connections to the server, it starts reporting accept() error 23 (ENFILE, too many open files) -- this is because we start to hit `CONFIG_LWIP_MAX_SOCKETS`.  From the client point of view, the sockets connect and immediately disconnect, causing all sorts of unexpected errors.  The problem is that `httpd_server()` will continue to call `accept()` on new connections regardless of how many are open.  If `lru_purge_enable` is false (meaning I don't want to force close old connections), it would be better to just not call `accept()` unless we can actually handle a new session.  This patch seems to do the trick for me:

```patch
diff --git a/components/esp_http_server/src/httpd_main.c b/components/esp_http_server/src/httpd_main.c
index 52228a97..cbfcbf0a 100644
--- a/components/esp_http_server/src/httpd_main.c
+++ b/components/esp_http_server/src/httpd_main.c
@@ -156,7 +156,8 @@ static esp_err_t httpd_server(struct httpd_data *hd)
 {
     fd_set read_set;
     FD_ZERO(&read_set);
-    FD_SET(hd->listen_fd, &read_set);
+    if (hd->config.lru_purge_enable || httpd_is_sess_available(hd))
+        FD_SET(hd->listen_fd, &read_set);
     FD_SET(hd->ctrl_fd, &read_set);
 
     int tmp_max_fd;
```
With this patch and a config like this:

```c
  httpd_config_t config = HTTPD_DEFAULT_CONFIG();
  config.backlog_conn     = 0;
  config.max_open_sockets = 8;
  config.lru_purge_enable = false;
```
then a client's attempted connection will not complete until the server is able to handle it, which is what one would expect.

anurag-kar, 2019-03-17:
@jimparis This looks great!

> If a client opens too many connections to the server, it starts reporting accept() error 23 (ENFILE, too many open files) -- this is because we start to hit CONFIG_LWIP_MAX_SOCKETS

Actually this behavior is intentional. Even if you were to increase the value of CONFIG_LWIP_MAX_SOCKETS the server would accept the connection (this time without error) and close it immediately (`httpd_is_sess_available()` returns false when limit is reached on config.max_open_sockets).

Though, to rethink about the appropriate and desirable behavior, what you suggest seems the right way to go about. Would it be possible for you to raise a pull request with your modifications? Then we can merge it after our internal review process.

anurag-kar, 2019-04-01:
@jimparis I will move ahead and fix the issue from our side using your patch. Thanks

closed 2019-04-04 by igrr

ronnye-roger, 2020-11-24:
Hello,
I'm working with ESP-WROOM-02 module and I'm basing my solution on ESP8266 RTOS SDK, that is a IDF-style SDK for ESP8266.
I found out the same problem discovered by **jimparis**, so I corrected my httpd_main.c file of the library with the same patch he suggested and, voilà, the problem seems to disappear!

Is it possible to update the ESP8266_RTOS_SDK with this adjustment? I'd like to work with code officially tested and released by Espressif. Furthermore, I am not sure whether other collateral points have to be adjusted after having introduced this change, perhaps something that have been already found out for ESP32.

Thank you.

jimparis, 2020-12-12:
> Furthermore, I am not sure whether other collateral points have to be adjusted after having introduced this change, perhaps something that have been already found out for ESP32.

I'm not familiar with the esp8266 version of the code, but this is a pretty straightforward change from one intended behavior to another, so I wouldn't expect any additional changes to be necessary.