espressif/esp-idf #3184: esp32-wrover-b smartconfig unable to work. Here's the log

kingofalpha, 2019-03-15:
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4017cb73  PS      : 0x00060630  A0      : 0x800e5ec4  A1      : 0x3ffe16b0  
A2      : 0x3ffe7ec4  A3      : 0x00000001  A4      : 0x3ffbde80  A5      : 0x3ffd2be0  
A6      : 0x3ffd2bd8  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffe16a0  
A10     : 0x3ffe7ec4  A11     : 0x3ffbfbaa  A12     : 0x00000006  A13     : 0x3ffe7eca  
A14     : 0x0000ffff  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000002c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

ELF file SHA256: ce9eac84d782201d2422585d9ef8723d8674e32462c3363b03f82355ee16a756

Backtrace: 0x4017cb73:0x3ffe16b0 0x400e5ec1:0x3ffe1700 0x400e61e6:0x3ffe1770 0x400e6248:0x3ffe17b0 0x400ea2cc:0x3ffe17e0 0x400ea3ee:0x3ffe1800 0x40092a00:0x3ffe1820 0x400958c9:0x3ffe1860

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6456
load:0x40078000,len:11308
load:0x40080400,len:6732
entry 0x40080764
[0;32mI (29) boot: ESP-IDF v3.3-beta2-53-g79ca95ddf-dirty 2nd stage bootloader[0m
[0;32mI (29) boot: compile time 11:52:23[0m
[0;32mI (30) boot: Enabling RNG early entropy source...[0m
[0;32mI (36) boot: SPI Speed      : 40MHz[0m
[0;32mI (40) boot: SPI Mode       : DIO[0m
[0;32mI (44) boot: SPI Flash Size : 4MB[0m
[0;32mI (48) boot: Partition Table:[0m
[0;32mI (51) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (74) boot:  2 factory          factory app      00 00 00010000 00200000[0m
[0;32mI (81) boot:  3 storage          Unknown data     01 82 00210000 000f0000[0m
[0;32mI (89) boot: End of partition table[0m
[0;32mI (93) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x3a7dc (239580) map[0m
[0;32mI (186) esp_image: segment 1: paddr=0x0004a804 vaddr=0x3ffbdb60 size=0x032dc ( 13020) load[0m
[0;32mI (191) esp_image: segment 2: paddr=0x0004dae8 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (193) esp_image: segment 3: paddr=0x0004def0 vaddr=0x40080400 size=0x02120 (  8480) load[0m
[0;32mI (205) esp_image: segment 4: paddr=0x00050018 vaddr=0x400d0018 size=0xedbb4 (973748) map[0m
[0;32mI (551) esp_image: segment 5: paddr=0x0013dbd4 vaddr=0x40082520 size=0x14c60 ( 85088) load[0m
[0;32mI (600) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (600) boot: Disabling RNG early entropy source...[0m
[0;32mI (601) cpu_start: Pro cpu up.[0m
[0;32mI (604) cpu_start: Application information:[0m
[0;32mI (609) cpu_start: Project name:     hello-world[0m
[0;32mI (614) cpu_start: App version:      v3.3-beta2-53-g79ca95ddf-dirty[0m
[0;32mI (621) cpu_start: Compile time:     Mar 15 2019 11:52:11[0m
[0;32mI (627) cpu_start: ELF file SHA256:  ce9eac84d782201d...[0m
[0;32mI (633) cpu_start: ESP-IDF:          v3.3-beta2-53-g79ca95ddf-dirty[0m
[0;32mI (640) cpu_start: Starting app cpu, entry point is 0x40081114[0m
[0;32mI (633) cpu_start: App cpu up.[0m
[0;32mI (651) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (658) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM[0m
[0;32mI (664) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM[0m
[0;32mI (670) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM[0m
[0;32mI (676) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM[0m
[0;32mI (682) heap_init: At 3FFD3F50 len 0000C0B0 (48 KiB): DRAM[0m
[0;32mI (688) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (695) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (701) heap_init: At 40097180 len 00008E80 (35 KiB): IRAM[0m
[0;32mI (707) cpu_start: Pro cpu start user code[0m
[0;32mI (54) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (98) BTDM_INIT: BT controller compile version [4cccca0][0m
[0;32mI (98) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (188) phy: phy_version: 4100, 6fa5e27, Jan 25 2019, 17:02:06, 0, 0[0m
[0;32mI (628) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;31mE (5628) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:[0m
[0;31mE (5628) task_wdt:  - IDLE0 (CPU 0)[0m
[0;31mE (5628) task_wdt: Tasks currently running:[0m
[0;31mE (5628) task_wdt: CPU 0: main[0m
[0;31mE (5628) task_wdt: CPU 1: IDLE1[0m
I (5748) wifi: wifi driver task: 3ffdfe8c, prio:23, stack:3584, core=0
I (5748) wifi: wifi firmware version: 53ea8b1
I (5748) wifi: config NVS flash: enabled
I (5748) wifi: config nano formating: disabled
[0;32mI (5758) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (5768) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
I (5788) wifi: Init dynamic tx buffer num: 32
I (5788) wifi: Init data frame dynamic rx buffer num: 32
I (5788) wifi: Init management frame dynamic rx buffer num: 32
I (5798) wifi: Init management short buffer num: 32
I (5798) wifi: Init static rx buffer size: 1600
I (5808) wifi: Init static rx buffer num: 10
I (5808) wifi: Init dynamic rx buffer num: 32
I (5818) wifi: mode : sta (30:ae:a4:c6:fe:74)
I (5898) smartconfig: SC version: V2.8.0

TimXia, 2019-03-18:
@kingofalpha Please provide the brand of the router and cellphone. Does the smartconfig example of IDF work?

Alvin1Zhang, 2019-05-16:
@kingofalpha Thanks for reporting the issue. Would you please any updates of the issue? Has your issue been fixed? Thanks.

Alvin1Zhang, 2019-06-05:
@kingofalpha Thanks for reporting the issue. Will close this issue due to lack of more information. Feel free to reopen and please help provide more details. Thanks.

closed 2019-06-05 by Alvin1Zhang