espressif/esp-idf #3185: ASSERT_ERR(evt->evt.conn.counter == evt->evt.conn.instant), in lld_evt.c at line 866

gilhossi, 2019-03-15:
I got the following error message. (ESP-IDF version : 3.1.1)

ASSERT_ERR(evt->evt.conn.counter == evt->evt.conn.instant), in lld_evt.c at line 866

After the message was printed on the terminal, BLE connection with a device was disconnected.
(server: esp32 / client: smart-phone(android))
I cannot get the information about this message because lld_evt.c is in libbtdm_app.a.
I'd like to know about this error situation.

Weijian-Espressif, 2019-03-15:
@gilhossi,  
* Please provide a detailed test log
* IDF commit 
* how to reproduce the issue

gilhossi, 2019-03-15:
@Weijian-Espressif 

I am using 'tag v3.1.1'.
And an error message is only one line. (ASSERT_ERR(evt->evt.conn.counter == evt->evt.conn.instant), in lld_evt.c at line 866)
Actually I don't know how to reproduce the issue.
I am just waiting to see the error message again.
I would like to know the meaning of the message only. 

Thanks.

Kython89, 2019-03-27:
We received the following error after being connected for 16 hours.
IDF: v3.2-dev-1385-g129d32772
```
ASSERT_ERR(evt->evt.conn.counter == evt->evt.conn.instant), in lld_evt.c at line 866
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x40084685  PS      : 0x00060f34  A0      : 0x80086ec8  A1      : 0x3ffbe1f0
0x40084685: r_assert_err at ??:?

A2      : 0x00000001  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x3ffbdc8c
A6      : 0x3ffb93cc  A7      : 0x3ffb8360  A8      : 0x80084680  A9      : 0x3ffbe1d0
A10     : 0x00000055  A11     : 0x00000055  A12     : 0x00000010  A13     : 0xffffffff
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x40084629  LEND    : 0x40084630  LCOUNT  : 0x00000000
0x40084629: r_assert_err at ??:?

0x40084630: r_assert_err at ??:?

Core 0 was running in ISR context:
EPC1    : 0x40009203  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40084685
0x40084685: r_assert_err at ??:?

Backtrace: 0x40084685:0x3ffbe1f0 0x40086ec5:0x3ffbe210 0x400865ae:0x3ffbe240 0x40087152:0x3ffbe260 0x40
08809a:0x3ffbe280 0x400891cb:0x3ffbe2a0 0x40082551:0x3ffbe2c0 0x40190663:0x00000000
0x40084685: r_assert_err at ??:?

0x40086ec5: r_lld_evt_restart at ??:?

0x400865ae: r_lld_evt_end at ??:?

0x40087152: r_lld_evt_end_isr at ??:?

0x4008809a: r_rwble_isr at ??:?

0x400891cb: r_rwbtdm_isr_wrapper at intc.c:?

0x40082551: _xt_lowint1 at C:/Users/btate/repos/dash/msys32/home/btate/esp/esp-idf/components/freertos/
xtensa_vectors.S:1105

0x40190663: esp_pm_impl_waiti at C:/Users/btate/repos/dash/msys32/home/btate/esp/esp-idf/components/esp
32/pm_esp32.c:474

Core 1 register dump:
PC      : 0x40190666  PS      : 0x00060a34  A0      : 0x800d34f5  A1      : 0x3ffbc330
0x40190666: esp_pm_impl_waiti at C:/Users/btate/repos/dash/msys32/home/btate/esp/esp-idf/components/esp
32/pm_esp32.c:474

A2      : 0x00000000  A3      : 0x00000001  A4      : 0x00000001  A5      : 0x00000001
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x80090b34  A9      : 0x3ffbc310
A10     : 0x3ffbd2e8  A11     : 0x40082e90  A12     : 0x00060020  A13     : 0x00000001
0x40082e90: _free_r at C:/Users/btate/repos/dash/msys32/home/btate/esp/esp-idf/components/newlib/syscal
ls.c:41

A14     : 0x00060020  A15     : 0x3ffbd250  SAR     : 0x00000000  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x40190666:0x3ffbc330 0x400d34f2:0x3ffbc350 0x40091731:0x3ffbc370 0x40092309:0x3ffbc390
0x40190666: esp_pm_impl_waiti at C:/Users/btate/repos/dash/msys32/home/btate/esp/esp-idf/components/esp
32/pm_esp32.c:474

0x400d34f2: esp_vApplicationIdleHook at C:/Users/btate/repos/dash/msys32/home/btate/esp/esp-idf/compone
nts/esp32/freertos_hooks.c:86

0x40091731: prvIdleTask at C:/Users/btate/repos/dash/msys32/home/btate/esp/esp-idf/components/freertos/
tasks.c:5056

0x40092309: vPortTaskWrapper at C:/Users/btate/repos/dash/msys32/home/btate/esp/esp-idf/components/free
rtos/port.c:403

Rebooting...
```

Weijian-Espressif, 2019-04-01:
@Kython89, Please test with v3.2 commit 46fa5654a72726d61dc5dd6e3b3680f814fe802f
@gilhossi,  Please test with v3.1 commit 03527e840c6d04c6b8312351adc1501ad3536b76

Alvin1Zhang, 2019-07-05:
@gilhossi Thanks for reporting the issue, feel free to reopen the issue if it still exists. Thanks.

closed 2019-07-05 by Alvin1Zhang

RoryMotion, 2020-08-04:
Hello,
This doesn't appear to be fixed, unfortunately. We're using v4.0.1 and we've seen: -

ASSERT_ERR(evt->evt.conn.counter == evt->evt.conn.instant), in lld_evt.c at line 1017

...just a different line number this time (with v3.1 it was on line 858).

Thanks in anticipation,
Rory

WCCWCC, 2020-08-04:
Hi @RoryMotion ，
You can re-update the release/v4.0 code, your problem should be fixed.

The repaired commit is shown below:

[image]

RoryMotion, 2020-08-04:
Thanks for the quick response @WCCWCC,

We'll give that a try.

Thanks,
Rory

RoryMotion, 2020-08-04:
Hi again,

Could you please tell us if this change has been verified. As this issue is difficult to reproduce any information will help for our commercial product.

Incidentally, when I used esp-idf-tools-setup-2.3.exe, I ended up with v4.0.1. This seemed good as that version has some changes to the WiFi stack, which are of use to get accreditation. Can you tell me if the changes made to v4.0 will be applied to v4.0.1?

Thanks in anticipation,
Rory

WCCWCC, 2020-08-05:
You update the code to the latest release/v4.0, then execute git submodule update --init, and then you use git log to see if the commit on the picture is included.

RoryMotion, 2020-08-05:
Thanks again @WCCWCC,

We see that, on start-up, the ESP reports "v4.0.1-378-g26d7fb774" which matches the commit made on 02 Aug 2020 and I see that that branch was merged when I execute git log.

What we also need is a formal report proving that the fix is robust. Does Espressif provide such a record? Some kind of test report, for example. (I think I know what the answer is, but I have to ask)

Renewed thanks,
Rory

WCCWCC, 2020-08-05:
Hi, RoryMotion

This is a problem reported by customers before. 

This problem only occurs when the instant value of the **channel map packet** is 65535. 

After the bug was fixed, the customer verified that the problem no longer exists

RoryMotion, 2020-08-07:
Hi @WCCWCC,

Sorry about the delay -- I was waiting to see if we had any follow-up questions.

Many thanks for your help and prompt responses. I am much obliged to you.

Kind regards,
Rory

RoryMotion, 2020-09-30:
Hello again,

We now have another problem whereby the ESP32 module cannot be found by an app we have developed. A BLE analyser app confirms that it doesn't advertise itself (even when we re-enable advertising) but we have found that stopping and restarting the Bluetooth stack fixes the problem.

Does this sound familiar to anyone, please? As always, any advice will be appreciated!

Thanks in anticipation,
Rory

WCCWCC, 2020-10-13:
@RoryMotion 
You mean that ESP32 advertising packets cannot be found by the app.    
Confirm through other apps that esp32 does not send advertising packets.
So your question is why esp32 will stop sending advertising packets？

Is there any abnormal log when this problem occurs，Or can you help us provide a way to reproduce?

Alvin1Zhang, 2020-10-13:
@RoryMotion Thanks for reporting and updates, for the new issue, it will be much appreciated if could file a new ticket and update details required by the issue template, so that we could keep good track of. Thanks for understanding.

RoryMotion, 2020-10-22:
Hello @WCCWCC and @Alvin1Zhang,
Sorry; I was hoping to obtain more information shortly after I posted my question about advertising. Unfortunately this wasn't possible (the problem occurred 'in the field' and we were unable to capture any debug, nor ascertain what led up to the problem -- e.g. whether the operator did something we hadn't anticipated).
If/when the problem reoccurs and I have more information, I'll create a new ticket (unless, of course, it's something we can fix) as you requested, @Alvin1Zhang.
Thank you