espressif/esp-idf #3186: Send empty mqtt message to delete retained message.

Zeni241, 2019-03-15:
How can I send empty mqtt message to broker to delete some retained message?
I tried following but it is not working:
esp_mqtt_client_publish(mqttclient, "mytopic"," ",0, 0, 0);
esp_mqtt_client_publish(mqttclient, "mytopic",NULL,0, 0, 0);
esp_mqtt_client_publish(mqttclient, "mytopic",' ',0, 0, 0);

benjaminaigner, 2019-03-15:
Short search:
according to http://www.steves-internet-guide.com/mqtt-retained-messages-example/
and https://www.hivemq.com/blog/mqtt-essentials-part-8-retained-messages/

You need to set the retain flag to TRUE and send an empty message.

Zeni241, 2019-03-15:
Thanks a lot benjaminaigner, I was missing setting retained flag!
Now following deleted retained message on broker:
esp_mqtt_client_publish(mqttclient, "mytopic"," ",0, 0, **1**)

benjaminaigner, 2019-04-23:
@Zeni241 

I'm glad my hint works :-)
If you are satisfied, please close this issue.

closed 2019-05-02 by Zeni241