espressif/esp-idf #3188: How do i send ALT+TAB (or.. how do modifiers work?)

Webunity, 2019-03-15:
I can't seem to be able to send ALT+TAB;
According to this page: https://github.com/jpbrucker/BLE_HID/blob/master/doc/HID.md

> For an uppercase 'A', the report will also need to contain a 'Left Shift'
modifier. The modifier byte is actually a bitmap, which means that each bit
corresponds to one key:
>
> - bit 0: left control
> - bit 1: left shift
> - bit 2: left alt
> - bit 3: left GUI (Win/Apple/Meta key)
> - bit 4: right control
> - bit 5: right shift
> - bit 6: right alt
> - bit 7: right GUI
>
> With left shift pressed, out report will look like that:
>
>    'A' report:     [2, 0, 4, 0, 0, 0, 0, 0]

Question: Why is there a '2' and not a 1 ? And how do i combine 2 bits?

Weijian-Espressif, 2019-03-22:
@Webunity,  2 is 0b0010, bit 1 is left shift. for example, if you want to send shift+alt, the value is (0b00000010 | 0b00000100) = 0b00000110 = 6.

Weijian-Espressif, 2019-03-22:
@Webunity, [4, 0, 43] is `ALT+TAB`, I do not know why it not works.
May be different keyboard types, SPEC define that NOTE: A general note on Usages and languages: Due to the variation of keyboards from language to language, it is not feasible to specify exact key mappings for every language. Where this list is not specific for a key function in a language, the closest equivalent key position should be used, so that a keyboard may be modified for a different language by simply printing different keycaps. One example is the Y key on a North American keyboard. In Germany this is typically Z. Rather than changing the keyboard firmware to put the Z Usage into that place in the descriptor list, the vendor should use the Y Usage on both the North American and German keyboards. This continues to be the existing practice in the industry, in order to minimize the number of changes to the electronics to accommodate other languages.

Alvin1Zhang, 2019-07-05:
@Webunity Thanks for reporting the issue, feel free to reopen the issue if it still exists. Thanks.

closed 2019-07-05 by Alvin1Zhang