espressif/esp-idf #3191: MQTT TCP example problem with IP adress

aronfeher, 2019-03-15:
Hi everyone, my name is Aron.
As the title suggests I found an issue with the MQTT/TCP setup - or maybe I expect something which was not meant to be used...

I wanted to use the MQTT/TCP project provided here. I've created a node-red dashboard, and the mosquitto broker is running locally. So instead of the given mqtt address i used my pc-s IP address as URI - for which the ESP32 gave a parsing error.  

- Development Kit:   ESP32-DevKitC
- Kit version DevKitC v1 and v3
- IDE: Platform IO with VS Code

I've tried tho achieve the same thing in Arduino environment, where the program ran smoothly, and the IP was accepted.

Either way the debug log is here:
```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5800
load:0x40078000,len:7924
load:0x40080000,len:5916
entry 0x40080314
␛[0;32mI (29) boot: ESP-IDF 3.30103.190221 2nd stage bootloader␛[0m
␛[0;32mI (29) boot: compile time 18:35:44␛[0m
␛[0;32mI (29) boot: Enabling RNG early entropy source...␛[0m
␛[0;32mI (34) boot: SPI Speed      : 40MHz␛[0m
␛[0;32mI (38) boot: SPI Mode       : DIO␛[0m
␛[0;32mI (42) boot: SPI Flash Size : 4MB␛[0m
␛[0;32mI (46) boot: Partition Table:␛[0m
␛[0;32mI (50) boot: ## Label            Usage          Type ST Offset   Length␛[0m
␛[0;32mI (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000␛[0m
␛[0;32mI (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000␛[0m
␛[0;32mI (72) boot:  2 factory          factory app      00 00 00010000 00100000␛[0m
␛[0;32mI (80) boot: End of partition table␛[0m
␛[0;32mI (84) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x18858 (100440) map␛[0m
␛[0;32mI (128) esp_image: segment 1: paddr=0x00028880 vaddr=0x3ffc0000 size=0x02e6c ( 11884) load␛[0m
␛[0;32mI (132) esp_image: segment 2: paddr=0x0002b6f4 vaddr=0x40080000 size=0x00400 (  1024) load␛[0m
␛[0;32mI (135) esp_image: segment 3: paddr=0x0002bafc vaddr=0x40080400 size=0x04514 ( 17684) load␛[0m
␛[0;32mI (150) esp_image: segment 4: paddr=0x00030018 vaddr=0x400d0018 size=0x7a9c4 (502212) map␛[0m
␛[0;32mI (326) esp_image: segment 5: paddr=0x000aa9e4 vaddr=0x40084914 size=0x0b1ec ( 45548) load␛[0m
␛[0;32mI (354) boot: Loaded app from partition at offset 0x10000␛[0m
␛[0;32mI (354) boot: Disabling RNG early entropy source...␛[0m
␛[0;32mI (354) cpu_start: Pro cpu up.␛[0m
␛[0;32mI (358) cpu_start: Starting app cpu, entry point is 0x40081000␛[0m
␛[0;32mI (350) cpu_start: App cpu up.␛[0m
␛[0;32mI (368) heap_init: Initializing. RAM available for dynamic allocation:␛[0m
␛[0;32mI (375) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM␛[0m
␛[0;32mI (381) heap_init: At 3FFC90B0 len 00016F50 (91 KiB): DRAM␛[0m
␛[0;32mI (387) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM␛[0m
␛[0;32mI (394) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM␛[0m
␛[0;32mI (400) heap_init: At 4008FB00 len 00010500 (65 KiB): IRAM␛[0m
␛[0;32mI (406) cpu_start: Pro cpu start user code␛[0m
␛[0;32mI (89) cpu_start: Starting scheduler on PRO CPU.␛[0m
␛[0;32mI (0) cpu_start: Starting scheduler on APP CPU.␛[0m
␛[0;32mI (90) MQTT_EXAMPLE: [APP] Startup..␛[0m
␛[0;32mI (90) MQTT_EXAMPLE: [APP] Free memory: 201912 bytes␛[0m
␛[0;32mI (90) MQTT_EXAMPLE: [APP] IDF version: 3.30103.190221␛[0m
I (150) wifi: wifi driver task: 3ffd20c8, prio:23, stack:3584, core=0
I (150) wifi: wifi firmware version: 71d6725
I (150) wifi: config NVS flash: enabled
I (150) wifi: config nano formating: disabled
␛[0;32mI (160) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE␛[0m
␛[0;32mI (170) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE␛[0m
I (190) wifi: Init dynamic tx buffer num: 32
I (190) wifi: Init data frame dynamic rx buffer num: 32
I (190) wifi: Init management frame dynamic rx buffer num: 32
I (200) wifi: Init static rx buffer size: 1600
I (200) wifi: Init static rx buffer num: 10
I (200) wifi: Init dynamic rx buffer num: 32
␛[0;32mI (210) MQTT_EXAMPLE: start the WIFI SSID:[Gothic]␛[0m
␛[0;32mI (280) phy: phy_version: 4008, c9ae59f, Jan 25 2019, 16:54:06, 0, 0␛[0m
I (280) wifi: mode : sta (30:ae:a4:0b:07:f0)
␛[0;32mI (280) MQTT_EXAMPLE: Waiting for wifi␛[0m
I (1010) wifi: n:6 2, o:1 0, ap:255 255, sta:6 2, prof:1
I (1850) wifi: state: init -> auth (b0)
I (1860) wifi: state: auth -> assoc (0)
I (1870) wifi: state: assoc -> run (10)
I (1880) wifi: connected with Gothic, channel 6
I (1880) wifi: pm start, type: 1

␛[0;32mI (2650) event: sta ip: 192.168.69.103, mask: 255.255.255.0, gw: 192.168.69.69␛[0m
␛[0;32mI (2650) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE␛[0m
␛[0;31mE (2650) MQTT_CLIENT: Error parse uri = 192.168.69.104:1883␛[0m
C:\users\csurleny\.platformio\packages\framework-espidf\components\freertos\event_groups.c:350 (xEventGroupWaitBits)- assert failed!
abort() was called at PC 0x4008a978 on core 0

Backtrace: 0x4008d7db:0x3ffcc540 0x4008d933:0x3ffcc560 0x4008a978:0x3ffcc580 0x400d291a:0x3ffcc5b0 0x400d292d:0x3ffcc5d0 0x400d2ebd:0x3ffcc5f0 0x400d2098:0x3ffcc630 0x400d4a98:0x3ffcc6b0

Rebooting...
ets Jun  8 2016 00:22:57
```

The code is exactly as in the given example, with my WiFi SSID/PASS, and the MQTT_URI being "192.168.69.104:1883".

Any help/idea would be rally appreciated.

david-cermak, 2019-03-16:
Hi Aron,

You're missing a scheme part of URI. Please try "mqtt://192.168.69.104:1883".

negativekelvin, 2019-03-16:
Shouldn't crash like that anyway, bug in mqtt_client.c doesn't check for null event group handle

aronfeher, 2019-03-17:
Thank you for the replies. Yes, the problem was the "mqtt://" tag missing from the URI. Now it works as intended - still it is a bit strange for me, but I won't complain.

On a side note There are still times when the nvm_flash_init() won't do it's job, so the wifi_init is complaining, but that can be solved by erasing the ESP32 memory and re-flashing.

Thanks again for the help.

closed 2019-03-17 by aronfeher

HectorCarreno, 2020-07-16:
> Thank you for the replies. Yes, the problem was the "mqtt://" tag missing from the URI. Now it works as intended - still it is a bit strange for me, but I won't complain.
>
> On a side note There are [...]

Yes, mqtt:// is a important part of the URI on ESP-MQTT. I have had problems with establish connection with AWS IoT Core server. The issue presented in the MCU response is:

I (5079) ppp_protocol: ~~~~~~~~~~~~~~
I (5079) ppp_protocol: GOT ip event!!!
D (5079) event: running post IP_EVENT:5 with handler 0x400dbccc and context 0x3ffb47c0 on loop 0x3ffb8efc
D (5089) esp_netif_handlers: esp_netif action connected with netif0x3ffb9eb0 from event_id=5
D (5099) esp_netif_lwip: check: remote, if=0x3ffb9eb0 fn=0x400dc8e8

D (5109) esp_netif_lwip: esp_netif_up_api esp_netif:0x3ffb9eb0
I (5109) system_api: Base MAC address is not set
D (5109) esp_netif_lwip: check: local, if=0x3ffb9eb0 fn=0x400dcf1c

I (5129) system_api: read default base MAC address from EFUSE
D (5129) efuse: coding scheme 0
D (5129) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffb9eb0
D (5139) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 8 bit
D (5139) esp_netif_lwip: call api in lwip: ret=0x0, give sD (5159) efuse: coding scheme 0
em
D (5159) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 0 bit
D (5169) efuse: coding scheme 0
D (5169) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 24 bit
D (5179) efuse: coding scheme 0
D (5179) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 16 bit
D (5189) efuse: coding scheme 0
D (5189) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 8 bit
D (5199) efuse: coding scheme 0
D (5199) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 0 bit
D (5209) efuse: coding scheme 0
D (5209) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 16 bit
D (5219) MQTT_CLIENT: MQTT client_id=ESP32_158D1C
D (5219) event: created event loop 0x3ffbd070
D (5229) MQTT_CLIENT: Core selection disabled
I (5229) ppp_protocol: MQTT other event id: 7
D (5239) esp-tls: host:a1eaun1ie4ulmg-ats.iot.sa-east-1.amazonaws.com: strlen 46
E (18859) esp-tls: couldn't get hostname for :a1eaun1ie4ulmg-ats.iot.sa-east-1.amazonaws.com:
E (18859) esp-tls: Failed to open new connection
E (18859) TRANS_SSL: Failed to open a new connection
E (18859) MQTT_CLIENT: Error transport connect
I (18869) ppp_protocol: MQTT_EVENT_ERROR
D (18869) MQTT_CLIENT: Reconnect after 10000 ms

It happens when I use mqtts://xxxxxx.brokerendpoint. When I use the broker IP address I got this:

I (5989) system_api: Base MAC address is not set
D (5989) esp_netif_lwip: check: local, if=0x3ffb9eb0 fn=0x400dcf1c

I (6009) system_api: read default base MAC address from EFUSE
D (6009) efuse: coding scheme 0
D (6009) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffb9eb0
D (6019) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 8 bit
D (6019) esp_netif_lwip: call api in lwip: ret=0x0, give sD (6039) efuse: coding scheme 0
em
D (6039) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 0 bit
D (6049) efuse: coding scheme 0
D (6049) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 24 bit
D (6059) efuse: coding scheme 0
D (6059) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 16 bit
D (6069) efuse: coding scheme 0
D (6069) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 8 bit
D (6079) efuse: coding scheme 0
D (6079) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 0 bit
D (6089) efuse: coding scheme 0
D (6089) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 16 bit
D (6099) MQTT_CLIENT: MQTT client_id=ESP32_158D1C
D (6099) event: created event loop 0x3ffbd01c
D (6109) MQTT_CLIENT: Core selection disabled
I (6109) ppp_protocol: MQTT other event id: 7
D (6119) esp-tls: host:54.207.57.8: strlen 11
E (7199) esp-tls-mbedtls: mbedtls_x509_crt_parse returned -0x2180
E (7199) esp-tls-mbedtls: Failed to set client pki context
E (7199) esp-tls-mbedtls: Failed to set client configurations
E (7209) esp-tls: create_ssl_handle failed
E (7209) esp-tls: Failed to open new connection
E (7219) TRANS_SSL: Failed to open a new connection
E (7219) MQTT_CLIENT: Error transport connect
I (7229) ppp_protocol: MQTT_EVENT_ERROR
D (7229) MQTT_CLIENT: Reconnect after 10000 ms
I (7239) ppp_protocol: MQTT_EVENT_DISCONNECTED

What happen here? both cases. The second case I believe that i can to be making an error with the certs or something, but the first, I haven't idea. Who Can help me?... I will appreciate any support. Thanks.